openrazer/openrazer #1350: persistence.conf does not save DPI informations

JulienDlq, 2020-11-26:
### Distribution

Operating System: Gentoo Linux
Kernel Version: 5.9.11-gentoo-x86_64
OS Type: 64-bit

### Support

- [X] Have you tried rebooting?
- [X] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Each time I restart OpenRazer service, the DPI of my DeathAdder 2013 is reset to 1800 DPI instead of restore my 500 DPI.
I do not have any entry about DPI in persistence.conf file.

Here are the entries in my persistence.conf file in the PMXXXXXXXXXXXXX section
```
poll_rate = 500
logo_active = True
logo_brightness = 75.0
logo_effect = spectrum
logo_colors = 0 255 0 0 255 255 0 0 255
logo_speed = 1
logo_wave_dir = 1
scroll_active = True
scroll_brightness = 75.0
scroll_effect = spectrum
scroll_colors = 0 255 0 0 255 255 0 0 255
scroll_speed = 1
scroll_wave_dir = 1
```

JulienDlq, 2020-11-26:
Furthermore, I don't know if it will help, but in dmesg I have this kind of line anytime I change DPI settings in Polychromatic Controller.
```
[11128.589418] razermouse: Command failed. Start Marker: 03 id: ff Num Params: 03 Reserved: 04 Command: 01 Params: 0f0f0000000000000000000000000000 .
```

lah7, 2020-11-28:
Persistence was missing in the daemon for this type of device, but it's supported now. Please give the latest `master` branch (or ff7c2688f8fbe15236a34d5c07c814693fb72d1b) a try.

For the `dmesg` error, does the mouse successfully retrieve and set the DPI?

closed 2020-11-28 by lah7

JulienDlq, 2020-11-28:
> Persistence was missing in the daemon for this type of device, but it's supported now. Please give the latest `master` branch (or [ff7c268](https://github.com/openrazer/openrazer/commit/ff7c2688f8fbe15236a34d5c07c814693fb72d1b)) a try.

I will give a try. :wink: 

> For the `dmesg` error, does the mouse successfully retrieve and set the DPI?

The DPI was correctly set, it seems, and the mouse reacts with a corresponding speed.

JulienDlq, 2020-11-28:
> > Persistence was missing in the daemon for this type of device, but it's supported now. Please give the latest `master` branch (or [ff7c268](https://github.com/openrazer/openrazer/commit/ff7c2688f8fbe15236a34d5c07c814693fb72d1b)) a try.
> 
> I will give a try. wink
> 

And it works!! Thanks. :grin:

commit ff7c2688 (2020-11-28): daemon: Persist DPI values for 'nagahex' method

Includes DeathAdder 2013. Fixes #1350.