openrazer/openrazer #1353: Support for razer Nari (lsusb

AlessandroMedici, 2020-11-27:
### Device Info

* Name: <See under device>
* Product Number: <See under device>
* Model Number: <See under device>

### Device Information
```
Please post the output of
    lsusb -d 1532: -v
```

```
Please post the output of
    ls -lah /dev/input/by-id/
```

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on Razer Synapse>

AlessandroMedici, 2020-11-27:
lsusb -d 1532: -v

Bus 001 Device 005: ID 1532:051c Razer USA, Ltd Nari (Wireless)
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x051c Nari (Wireless)
  bcdDevice            1.00
  iManufacturer           4 Razer
  iProduct                5 Razer Nari
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x01bd
    bNumInterfaces          6
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol      0 
      iInterface              1 
      AudioControl Interface Descriptor:
        bLength                10
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               1.00
        wTotalLength       0x0046
        bInCollection           2
        baInterfaceNr(0)        1
        baInterfaceNr(1)        2
      AudioControl Interface Descriptor:
        bLength                12
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID             6
        wTerminalType      0x0402 Headset
        bAssocTerminal         80
        bNrChannels             1
        wChannelConfig     0x0000
        iChannelNames           0 
        iTerminal               0 
      AudioControl Interface Descriptor:
        bLength                 9
        bDescriptorType        36
        bDescriptorSubtype      6 (FEATURE_UNIT)
        bUnitID                 3
        bSourceID               6
        bControlSize            1
        bmaControls(0)       0x03
          Mute Control
          Volume Control
        bmaControls(1)       0x00
[... 343 lines cut ...]
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              3 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     426
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               1

AlessandroMedici, 2020-11-27:
ls -lah /dev/input/by-id/
totale 0
drwxr-xr-x 2 root root 160 27 nov 13.27 .
drwxr-xr-x 4 root root 600 27 nov 13.27 ..
lrwxrwxrwx 1 root root   9 27 nov 13.07 usb-HAILUCK_CO._LTD_USB_KEYBOARD-event-if01 -> ../event7
lrwxrwxrwx 1 root root   9 27 nov 13.07 usb-HAILUCK_CO._LTD_USB_KEYBOARD-event-kbd -> ../event5
lrwxrwxrwx 1 root root   9 27 nov 13.07 usb-HAILUCK_CO._LTD_USB_KEYBOARD-if01-event-mouse -> ../event6
lrwxrwxrwx 1 root root   9 27 nov 13.07 usb-HAILUCK_CO._LTD_USB_KEYBOARD-if01-mouse -> ../mouse0
lrwxrwxrwx 1 root root   9 27 nov 13.23 usb-Razer_Razer_Nari-event-if05 -> ../event2
lrwxrwxrwx 1 root root  10 27 nov 13.07 usb-Sonix_Technology_Co.__Ltd._USB_2.0_Camera_SN0001-event-if00 -> ../event13

AlessandroMedici, 2020-11-27:
Synapse Screenshots:
[image]
[image]
[image]
[image]
[image]
[image]
[image]

z3ntu, 2020-11-29:
Duplicate of #724

closed 2020-11-29 by z3ntu