openrazer/openrazer #1354: Daemon ValueError: Invalid Literal for int () with base 10: '\x02'

chadwl, 2020-11-28:
When trying to run polychromatic I recieve an error stating the 'Daemon not responding' followed by 'Poly-chromatic could not initialise the daemon. This could be caused by a bug with the daemon process. Exception: org.freedsetkop.DBus.Error.ServiceUnknown: The name :1.381 was not provided by any .service files' 

I then when into the console and ran the daemon in verbose mode to see the output of it and it presented me with the following information.

`Starting daemon.
2020-11-28 06:15:52 | razer                          | INFO     | Initialising Daemon (v2.9.0). Pid: 29522
2020-11-28 06:15:52 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2020-11-28 06:15:52 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2020-11-28 06:15:52 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus1800 ------------------------ (1532:0020)
2020-11-28 06:15:52 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus2000 ------------------------ (1532:005E)
2020-11-28 06:15:52 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusEliteDVaEdition ------------- (1532:006A)
2020-11-28 06:15:52 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusEssential ------------------- (1532:006B)
2020-11-28 06:15:52 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusV2 -------------------------- (1532:005B)
2020-11-28 06:15:52 | razer                          | DEBUG    | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)
2020-11-28 06:15:52 | razer                          | DEBUG    | Loaded device specification: RazerAtherisReceiver -------------------- (1532:0062)
2020-11-28 06:15:52 | razer                          | DEBUG    | Loaded device specification: RazerBaseStationChroma ------------------ (1532:0F08)
2020-11-28 06:15:52 | razer                          | DEBUG    | Loaded device specification: RazerBasilisk --------------------------- (1532:0064)
2020-11-28 06:15:52 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskXHyperSpeed ---------------- (1532:0083)
2020-11-28 06:15:52 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2019 --------------------- (1532:0241)
2020-11-28 06:15:52 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
2020-11-28 06:15:52 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaOverwatch ---------- (1532:0211)
2020-11-28 06:15:52 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
2020-11-28 06:15:52 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaV2 ----------------- (1532:0221)
2020-11-28 06:15:52 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowElite -------------------- (1532:0228)
2020-11-28 06:15:52 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowEssential ---------------- (1532:0237)
2020-11-28 06:15:52 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowLite --------------------- (1532:0235)
2020-11-28 06:15:52 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowStealth ------------------ (1532:011B)
2020-11-28 06:15:52 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowStealthEdition ----------- (1532:010E)
2020-11-28 06:15:52 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2012 ------------- (1532:010D)
2020-11-28 06:15:52 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2013 ------------- (1532:011A)
2020-11-28 06:15:52 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2016 ------------- (1532:0214)
2020-11-28 06:15:52 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChroma ------------------ (1532:0216)
2020-11-28 06:15:52 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXTournamentEditionChroma - (1532:021A)
2020-11-28 06:15:52 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXUltimate ---------------- (1532:0217)
2020-11-28 06:15:52 | razer                          | DEBUG    | Loaded device specification: RazerBlade15Advanced2020 ---------------- (1532:0253)
2020-11-28 06:15:52 | razer                          | DEBUG    | Loaded device specification: RazerBlade2018 -------------------------- (1532:0233)
2020-11-28 06:15:52 | razer                          | DEBUG    | Loaded device specification: RazerBlade2018Base ---------------------- (1532:023B)
2020-11-28 06:15:52 | razer                          | DEBUG    | Loaded device specification: RazerBlade2018Mercury ------------------- (1532:0240)
2020-11-28 06:15:52 | razer                          | DEBUG    | Loaded device specification: RazerBlade2019Adv ----------------------- (1532:023A)
2020-11-28 06:15:52 | razer                          | DEBUG    | Loaded device specification: RazerBlade2019Base ---------------------- (1532:0246)
2020-11-28 06:15:52 | razer                          | DEBUG    | Loaded device specification: RazerBlade2019StudioEdition ------------- (1532:024D)
2020-11-28 06:15:52 | razer                          | DEBUG    | Loaded device specification: RazerBladeEarly2020Base ----------------- (1532:0255)
2020-11-28 06:15:52 | razer                          | DEBUG    | Loaded device specification: RazerBladeLate2016 ---------------------- (1532:0224)
2020-11-28 06:15:52 | razer                          | DEBUG    | Loaded device specification: RazerBladeMid2019Mercury ---------------- (1532:0245)
2020-11-28 06:15:52 | razer                          | DEBUG    | Loaded device specification: RazerBladePro2017 ----------------------- (1532:0225)
2020-11-28 06:15:52 | razer                          | DEBUG    | Loaded device specification: RazerBladePro2017FullHD ----------------- (1532:022F)
2020-11-28 06:15:52 | razer                          | DEBUG    | Loaded device specification: RazerBladePro2019 ----------------------- (1532:0234)
2020-11-28 06:15:52 | razer                          | DEBUG    | Loaded device specification: RazerBladeProLate2016 ------------------- (1532:0210)
2020-11-28 06:15:52 | razer                          | DEBUG    | Loaded device specification: RazerBladeProLate2019 ------------------- (1532:024C)
2020-11-28 06:15:52 | razer                          | DEBUG    | Loaded device specification: RazerBladeQHD --------------------------- (1532:020F)
2020-11-28 06:15:52 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealth ----------------------- (1532:0205)
2020-11-28 06:15:52 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealth2019 ------------------- (1532:0239)
2020-11-28 06:15:52 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthEarly2020 -------------- (1532:0252)
2020-11-28 06:15:52 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2016 --------------- (1532:0220)
2020-11-28 06:15:52 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2017 --------------- (1532:0232)
2020-11-28 06:15:52 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2019 --------------- (1532:024A)
2020-11-28 06:15:52 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthMid2017 ---------------- (1532:022D)
2020-11-28 06:15:52 | razer                          | DEBUG    | Loaded device specification: RazerChromaHDK -------------------------- (1532:0F09)
2020-11-28 06:15:52 | razer                          | DEBUG    | Loaded device specification: RazerChromaMugHolder -------------------- (1532:0F07)
[... cut ...]
  File "/usr/lib/python3/dist-packages/openrazer_daemon/hardware/device_base.py", line 271, in __init__
    self.zone[i]["speed"] = int(self.persistence[self.storage_name][i + '_speed'])
ValueError: invalid literal for int() with base 10: '\x02'

Stopping daemon.
`

I've tried re-installing and whatnot however it's provided no resolve. The strange thing is it did work (polychromatic) at one point. It appears since a possible update this issue has been present. 

Is anybody able to provide some insight. It would be most helpful and welcomed. Thankyou in advance.

brett-smith, 2020-11-28:
This looks the same as the issue I raised - https://github.com/openrazer/openrazer/issues/1351. 

You can get the daemon starting again by editing `.config/openrazer/persistence.conf` and finding the line that might look something like `backlight_speed = \x02` and changing that to a proper number, e.g.  `backlight_speed = 1`

lah7, 2020-11-28:
Fixed! The daemon was been writing _bad_ data to `persistence.conf` for the starlight effect and some brightnesses, which caused the daemon to crash on its next launch.

You could [patch it using this commit](https://github.com/openrazer/openrazer/commit/604f1cab3a899a1877e787e0e6e42b58c289f3ed.patch) or install the `-git` or daily packages (if on Ubuntu) for the fix.

closed 2020-11-28 by lah7

chadwl, 2020-11-29:
Thank you all for your feedback. Issue is resolved.

commit 604f1cab (2020-11-28): daemon: Fix wrong data type being written for persistence

The daemon would throw an exception upon next run as it was reading a
bytes when it expected an integer.

The starlight effect and some brightnesses were affected.

Fixes #1351.
Fixes #1354.