openrazer/openrazer #1357: Support for Razer Addressable (ARGB) LED Controller

CalcProgrammer1, 2020-11-29:
### Device Info

* Name: Razer Addressable LED controller
* Product Number: 
* Model Number: 

### Device Information
```
Please post the output of
    lsusb -d 1532: -v
```

```
Please post the output of
    ls -lah /dev/input/by-id/
```

### Packet Captures

Device is on its way, will get captures when I get it

### Synapse Screenshots

Device is on its way

I just ordered this controller and would like to get it added to OpenRazer/OpenRGB.  My guess is that it's going to have a command to resize each channel in addition to the modes/direct control messages.

CalcProgrammer1, 2020-12-01:
I started integrating putting it in razeraccessory, using the same code paths as the Chroma HDK, and it is partially working.  Static effects work fine, but it seems to have a larger 320 byte HID packet for direct control.

CalcProgrammer1, 2020-12-02:
The brightness control seems to be writing to 1A-1F in the LED slot.  The function `razer_chroma_extended_matrix_brightness` should work for brightness.

z3ntu, 2020-12-03:
Can you try if changing brightness works with the ZERO_LED? Also I'd be nice if you could share the pcaps you took.

CalcProgrammer1, 2020-12-07:
I tried using the same code path as the other devices (just stuck the ARGB controller device ID in every case that the Chroma HDK was also in).  This would use the ZERO_LED.  Unfortunately, it didn't work, the brightness stayed full bright on the Chroma ARGB controller while all of my other Razer gear changed brightness appropriately.  With the snippet above, the Chroma ARGB also changed brightness appropriately.  I haven't had time to continue investigating this but I will provide more pcaps when I get back to it.

CalcProgrammer1, 2020-12-27:
Function to send the 320-byte ARGB packet:

```
int razer_send_argb_msg(struct usb_device* usb_dev, unsigned char channel, unsigned char size, void const* data)
{
    uint request = HID_REQ_SET_REPORT; // 0x09
    uint request_type = USB_TYPE_CLASS | USB_RECIP_INTERFACE | USB_DIR_OUT; // 0x21
    uint value = 0x300;
    int len;

    struct razer_argb_report report;

    printk(KERN_WARNING "razer_send_argb_msg: Channel: %d Size: %d.", channel, size);

    if(channel < 5)
    {
        report.report_id = 0x04;
    }
    else
    {
        report.report_id = 0x84;
    }

    report.channel_1 = channel;
    report.channel_2 = channel;

    report.pad = 0;

    report.last_idx = size - 1;

    memcpy(report.color_data, data, size * 3);

    char * buf = kmemdup(&report, sizeof(report), GFP_KERNEL);

    // Send usb control message
    len = usb_control_msg(usb_dev, usb_sndctrlpipe(usb_dev, 0),
        request,            // Request
        request_type,       // RequestType
        value,              // Value
        0x01,               // Index
        buf,                // Data
        sizeof(report),     // Length
        USB_CTRL_SET_TIMEOUT);

    // Wait
    //usleep_range(wait_min, wait_max);

    if (len != sizeof(report))
        printk(KERN_WARNING "razer driver: Device data transfer failed. len = %d", len);

    return ((len < 0) ? len : ((len != size) ? -EIO : 0));
}
```

paxos, 2022-01-04:
Is there anything I can do to help moving this forward?

z3ntu, 2022-11-20:
Fixed by #1929

closed 2022-11-20 by z3ntu