openrazer/openrazer #1359: Razer Ornata V2 not supported

FIvanO, 2020-12-01:
Razer Ornata V2 not supported. Tried on Linux 20.04

FIvanO, 2020-12-01:
openrazer-daemon can't recognize Razer Ornata V2 keyboard. Does someone have any tips to recognize it?
(Screen from razer-genie app)
[image]

Thanks

nightsky30, 2020-12-01:
Support for this device is tracked in #1187

FIvanO, 2020-12-01:
Did not found there at least something useful for resolving the issue.

nightsky30, 2020-12-01:
The linked issue is to track adding support for the Ornata V2. It is not supported by the daemon at this time. There is nothing that can be done regarding your issue other than actually adding and testing the code to support the device.

z3ntu, 2020-12-03:
Closing as duplicate of #1187 , thanks @nightsky30

closed 2020-12-03 by z3ntu