openrazer/openrazer #1362: Error after update to python-3.9.0-1

ErayOzcanPXL, 2020-12-03:
### Distribution

Arch Linux 5.9.11-arch2-1

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

/usr/bin/polychromatic-controller throws exception at line 52 and 58. I rolled back afterwards so I don't remember the exact message I got, but it was something like: "couldn't find pylib" (same for "polychromatic"). I didn't get the expected print messages (see line 63-65). There is basically a problem with the "pylib" and "polychromatic" module when using python-3.9.0-1, they are not recognized.

z3ntu, 2020-12-03:
You need to rebuild all AUR packages that contain Python with such upgrades in Arch. 
You can list the affected packages with the command `pacman -Qo /usr/lib/python3.8/`

ErayOzcanPXL, 2020-12-03:
Will try soon, I am currently in class :). I will let you know after rebuilding, thanks for replying so fast btw.

ErayOzcanPXL, 2020-12-03:
@z3ntu rebuilding worked, thanks for the tip! I will close the issue.

closed 2020-12-03 by ErayOzcanPXL

Trollwut, 2021-02-03:
Sorry to do the grave-digging, but just for the sake of completeness, you can update all affected packaghe with this command (using the `paru` AUR helper in this example:

    pacman -Qoq /usr/lib/python3.8/ | paru -S --rebuild -

This command could be modified for future updated, if necessary.

CloudyWhale, 2022-01-29:
Lots of apps coded in python really hate me

CloudyWhale, 2022-01-29:
Whenever I try the solution I get this error:
No package owns /usr/lib/python3.8

lah7, 2022-01-29:
@CloudyWhale Try again but reinstall packages that were installed for `3.9`. Python 3.10 is the latest now (`pacman -Q python`). In Arch, whenever there's a major version of Python, all Python packages that came from the AUR need to be rebuilt.

    pacman -Qoq /usr/lib/python3.9/ | paru -S --rebuild -

CloudyWhale, 2022-01-29:
I did earlier and it worked.