openrazer/openrazer #1364: Missing kernel headers, kernel unable to find razerkbd module

Salazarsly, 2020-12-06:
### Distribution
RHEL 

RHEL 8
kernel version: 4.18.0-144.el8.x86_64
### Support

- [yes ] Have you tried rebooting?
- [yes ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Folks, I'm trying to install polychromatic/razergenie by following instructions [https://github.com/moozhub/yum-repo-mooz] 
I am only able to download packages (including dependencies) : openrazer-meta, polychromatic
However, it also gives me an error as below:
~~~
DKMS: add completed.
Error! echo
Your kernel headers for kernel 4.18.0-144.el8.x86_64 cannot be found at
/lib/modules/4.18.0-144.el8.x86_64/build or /lib/modules/4.18.0-144.el8.x86_64/source.
You can use the --kernelsourcedir option to tell DKMS where it's located.
warning: %post(openrazer-kernel-modules-dkms-2.9.0-1.1.noarch) scriptlet failed, exit status 1
~~~

--> I have already installed "kernel-devel" package, but it still fails to get the kernel headers. 
Here are some diagnostics cmd output:
~~~
# rpm -qa | grep razer
openrazer-meta-2.9.0-1.1.noarch
python3-openrazer-2.9.0-1.1.noarch
openrazer-kernel-modules-dkms-2.9.0-1.1.noarch
openrazer-daemon-2.9.0-1.1.noarch
=====================================
# openrazer-daemon -Fv
Traceback (most recent call last):
  File "/usr/bin/openrazer-daemon", line 11, in <module>
    from openrazer_daemon.daemon import RazerDaemon, __version__
ModuleNotFoundError: No module named 'openrazer_daemon'
=====================================
# sudo journalctl -b | grep -i razer_mount
Dec 07 01:45:40  razer_mount[1222]: Modprobing razerkbd
Dec 07 01:45:40  systemd-udevd[1182]: Process 'razer_mount razerkbd 1-3:1.0' failed with exit code 1.
Dec 07 01:45:40  systemd-udevd[1178]: Process 'razer_mount razerkbd 1-3:1.1' failed with exit code 1.
Dec 07 01:45:40  systemd-udevd[1192]: Process 'razer_mount razerkbd 1-3:1.2' failed with exit code 1.
Dec 07 01:45:40  razer_mount[1231]: Driver razerkbd
Dec 07 01:45:40  razer_mount[1232]: Device_ID 0003:1532:0216.0003
Dec 07 01:45:40  razer_mount[1230]: Driver razerkbd
Dec 07 01:45:40  systemd-udevd[1192]: Process 'razer_mount razerkbd 0003:1532:0216.0003' failed with exit code 1.
Dec 07 01:45:40  razer_mount[1238]: Modprobing razerkbd
Dec 07 01:45:40  systemd-udevd[1178]: Process 'razer_mount razerkbd 0003:1532:0216.0002' failed with exit code 1.
=========================================
# modprobe razerkbd
modprobe: FATAL: Module razerkbd not found in directory /lib/modules/4.18.0-144.el8.x86_64
~~~

z3ntu, 2020-12-06:
Generally we (= me) don't support RHEL so you're mostly on your own, but I'll still try to help you where I can.

Kernel headers must match up with the exact version of your running kernel, it looks like it might not in your case, check the version strings of `kernel` and `kernel-devel` if they match up (`4.18.0-144.el8.x86_64`) - maybe you just need to reboot to activate the new kernel.

Regarding the daemon/python problem:
Can you check if in the directory printed by `python3 -c "from distutils.sysconfig import get_python_lib; print(get_python_lib())"` exists a folder called openrazer_daemon? If not check where the `openrazer-daemon` package has placed its files - maybe into e.g. python3.6 instead of python3.7 directory?

Salazarsly, 2020-12-07:
@z3ntu Appreciate your prompt help. 
- I'm not able to find any kernel-devel package, with the same version strings as of my kernel package. No repos hold the same kernel-devel package, seems like this is issue. :(
~~~
# rpm -qa | grep kernel   [Redacted o/p]
kernel-4.18.0-144.el8.x86_64
kernel-devel-4.18.0-240.1.1.el8_3.x86_64
=====
# uname -a
Linux xyz 4.18.0-144.el8.x86_64 #1 SMP Thu Sep 12 14:30:19 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
~~~

- Also, the `operazer-daemon` placed the files not in python3.6 rather in python3.8 .
~~~
# python3 -c "from distutils.sysconfig import get_python_lib; print(get_python_lib())"
/usr/lib/python3.6/site-packages

[root@ychoukse ~]# ls /usr/lib/python
python2.7/ python3.6/ python3.8/ 

[root@ychoukse ~]# ls /usr/lib/python3.8/site-packages/
easy_install.py                        openrazer_daemon-2.9.0-py3.8.egg-info/ polychromatic/
openrazer/                             pip/                                   __pycache__/
openrazer-2.9.0-py3.8.egg-info/        pip-19.3.1.dist-info/                  setuptools/
openrazer_daemon/                      pkg_resources/                         setuptools-41.6.0.dist-info/
[root@ychoukse ~]# ls /usr/lib/python3.8/site-packages/openrazer-2.9.0-py3.8.egg-info/
dependency_links.txt  PKG-INFO              SOURCES.txt           top_level.txt         
[root@ychoukse ~]# ls /usr/lib/python3.8/site-packages/openrazer-2.9.0-py3.8.egg-info/
dependency_links.txt  PKG-INFO              SOURCES.txt           top_level.txt         
[root@ychoukse ~]# ls /usr/lib/python3.8/site-packages/openrazer-2.9.0-py3.8.egg-info/^C
[root@ychoukse ~]# python
python2     python2.7   python3     python3.6   python3.6m  python3.8   
~~~

z3ntu, 2020-12-07:
1. You need to fix that somehow - otherwise dkms won't be able to build the kernel modules

2. I'm guessing you got the binary packages from somewhere - either from a wrong distro version or the binary packages just need to be rebuilt. It's probably easiest for you to build the packages from source so everything matches up.

Salazarsly, 2020-12-07:
@z3ntu I have somehow sidestepped the kernel-devel thing, and dkms has built the correct modules now. I can see the info:
~~~
# dkms install openrazer-driver/2.9.0
Module openrazer-driver/2.9.0 already installed on kernel 4.18.0-240.1.1.el8_3.x86_64/x86_64
~~~

However, I'm not good with building packages from source and all(have basic knowledge). I'm looking my way on google to do so ;) 
I might try to redo the steps again, let see how it goes. 
Can you share some steps or best practices?

Salazarsly, 2020-12-07:
@z3ntu Hi, finally its working now. *Cheers*
So, since I'm on rhel 8 (fedora equivalent 28), I was using this repo [https://download.opensuse.org/repositories/hardware:/razer/Fedora_28/hardware:razer.repo] to get openrazer-meta package. But the content is unavailable. 
Moving on, I tried `Fedora_32` repo, that worked deployed openrazer-meta package. 

Still, the packages went to python3.8 site (might be because I used fedora_32 repo,idk), so I moved them to python3.6/site-packages. And, voila, daemon was started. 

Still I'm not able to use polychromatic(it never opens), however, razergenie is working like a charm and I'm satisfied! :) 

Thanks a lot for your help, let me know if there is something I can help considering you guys doing a tremendous work for the open-source community. \m/

closed 2020-12-10 by Salazarsly

LeGAlbert, 2021-10-30:
> 1. You need to fix that somehow - otherwise dkms won't be able to build the kernel modules
>
>     2. I'm guessing you got the binary packages from somewhere - either from a wrong distro version or the [...]

Hello, i have pretty much the same issue, it doesn't find the build file, when i do `rpm -qa | grep kernel` it shows :
```
> rpm -qa | grep kernel

kernel-srpm-macros-1.0-4.fc34.noarch
libreport-plugin-kerneloops-2.15.2-2.fc34.x86_64
kernel-core-5.14.9-200.fc34.x86_64
kernel-modules-5.14.9-200.fc34.x86_64
kernel-5.14.9-200.fc34.x86_64
kernel-modules-extra-5.14.9-200.fc34.x86_64
kernel-headers-5.14.9-200.fc34.x86_64
abrt-addon-kerneloops-2.14.6-4.fc34.x86_64
kernel-core-5.14.13-200.fc34.x86_64
kernel-modules-5.14.13-200.fc34.x86_64
kernel-5.14.13-200.fc34.x86_64
kernel-modules-extra-5.14.13-200.fc34.x86_64
kernel-devel-5.14.14-200.fc34.x86_64
openrazer-kernel-modules-dkms-3.1.0-1.1.noarch
kernel-core-5.14.14-200.fc34.x86_64
kernel-modules-5.14.14-200.fc34.x86_64
kernel-5.14.14-200.fc34.x86_64
kernel-modules-extra-5.14.14-200.fc34.x86_64
======================================
> uname -a

Linux fedora 5.14.13-
200.fc34.x86_64 #1 SMP Mon Oct 18 12:39:31 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
```
Do i have to remove all the unmatching files ?