openrazer/openrazer #1365: Can't get it work on Pop!OS 20.04

Somamint, 2020-12-08:
once i try to install openrazer on POP!OS 20.04 i got this error:

the following packages have unmet dependencies:

polychromatic : Depends on: python3-openrazer (>= 2.0.0) but should not be installed

Depends on: imagemagick is not installable

Depends on: openrazer-daemon (>= 2.0.0) should not be installed

E: Problems cannot be corrected, you have retained defective packages.

lah7, 2020-12-08:
Your package database seems to be broken (apt/dpkg). You'll need to fix this before you can upgrade/install packages properly. Pop!_OS is a variant of Ubuntu (which also inherits Debian) so any guides out there will help Pop!_OS too.

Please try running these commands and paste the output here:

```
sudo apt --fix-broken install
sudo dpkg --configure -a
sudo apt update
```

Somamint, 2020-12-08:
Still the same : 

```
Sudo apt --fix-broken install

Packet lists are read... Done
Dependency tree is built.       
Status information is read.... Ready
The following packages were installed automatically and are no longer needed:
  gnome-shell-extension-desktop-icons libnvidia-cfg1-440 libnvidia-common-440
  libnvidia-compute-440 libnvidia-compute-440:i386 libnvidia-decode-440
  libnvidia-decode-440:i386 libnvidia-encode-440 libnvidia-encode-440:i386
  libnvidia-extra-440 libnvidia-fbc1-440 libnvidia-fbc1-440:i386
  libnvidia-gl-440 libnvidia-ifr1-440 libnvidia-ifr1-440:i386
  linux-headers-5.4.0-7642 linux-headers-5.4.0-7642-generic
  linux-image-5.4.0-7642-generic linux-modules-5.4.0-7642-generic
  linux-modules-extra-5.4.0-7642-generic mousetweaks nvidia-compute-utils-440
  nvidia-dkms-440 nvidia-kernel-common-440 nvidia-kernel-source-440
  nvidia-utils-440 xserver-xorg-video-nvidia-440
Use "sudo apt autoremove" to remove them.
0 updated, 0 reinstalled, 0 to remove and 29 not updated.
somavakien@pop-os:~$ sudo dpkg --configure -a
somavakien@pop-os:~$ sudo apt update
OK:1 http://apt.pop-os.org/proprietary focal InRelease
OK:2 http://download.opensuse.org/repositories/hardware:/razer/xUbuntu_20.04 InRelease
Get:3 http://ppa.launchpad.net/deadsnakes/ppa/ubuntu focal InRelease [18,1 kB]
OK:4 http://ppa.launchpad.net/deki/firejail/ubuntu focal InRelease             
OK:5 http://ppa.launchpad.net/openrazer/daily/ubuntu focal InRelease 
OK:6 http://ppa.launchpad.net/openrazer/stable/ubuntu focal InRelease
OK:7 http://ppa.launchpad.net/polychromatic/daily/ubuntu focal InRelease
OK:8 http://ppa.launchpad.net/polychromatic/stable/ubuntu focal InRelease
OK:9 http://ppa.launchpad.net/system76-dev/stable/ubuntu focal InRelease
OK:10 http://ppa.launchpad.net/system76/pop/ubuntu focal InRelease             
Get:11 http://ppa.launchpad.net/deadsnakes/ppa/ubuntu focal/main i386 Packages [7.260 B]
Get:12 http://ppa.launchpad.net/deadsnakes/ppa/ubuntu focal/main amd64 Packages [17,7 kB]
OK:13 https://download.mono-project.com/repo/ubuntu stable-focal InRelease     
43.0 kB were fetched in 15 s (2,807 fps).                                  
Packet lists are read... Done
Dependency tree is built.       
Status information is read.... Ready
Update available for 28 packages. Run "apt list --upgradable" to view them.
N: The loading of the configured file "main/binary-i386/Packages" is skipped because the repository "https://download.mono-project.com/repo/ubuntu stable-focal InRelease" does not support the "i386" architecture.
somavakien@pop-os:~$ sudo apt install openrazer-meta
Packet lists are read... Done
Dependency tree is built.       
Status information is read.... Ready
Some packages could not be installed. This can mean that
you have requested an impossible situation or, if you have
unstable distribution, that some required packages are still available
were not created or have not yet left Incoming.
The following information may help you to resolve the situation:

The following packages have unmet dependencies:
 openrazer-meta : Depends on: openrazer-daemon (= 2.9.0-1~585~ubuntu20.04.1) but should not be installed
                  Depends on: python3-openrazer (= 2.9.0-1~585~ubuntu20.04.1) but should not be installed
E: Problems cannot be corrected, you have retained defective packages.
```
`sudo apt install software-properties-gtk
Packet lists are read... Done
Dependency tree is built.       
Status information is read.... Ready
E: Package software-properties-gtk cannot be found.
somavakien@pop-os:~$ sudo add-apt-repository ppa:openrazer/stable
 
OpenRazer is a driver and daemon that enables control of lighting and other functionality for supported Razer devices.

This PPA contains stable releases of the driver and daemon.

 More information: https://launchpad.net/~openrazer/+archive/ubuntu/stable
Press [ENTER] to continue or Ctrl-c to cancel adding.
[... 22 lines cut ...]
OK:8 http://ppa.launchpad.net/polychromatic/daily/ubuntu focal InRelease
OK:9 http://ppa.launchpad.net/polychromatic/stable/ubuntu focal InRelease
OK:10 http://ppa.launchpad.net/system76-dev/stable/ubuntu focal InRelease
OK:11 http://ppa.launchpad.net/system76/pop/ubuntu focal InRelease
Packet lists are read... Done
Dependency tree is built.       
Status information is read.... Ready
Update available for 28 packages. Run "apt list --upgradable" to view them.
N: The loading of the configured file "main/binary-i386/Packages" is skipped because the repository "https://download.mono-project.com/repo/ubuntu stable-focal InRelease" does not support the "i386" architecture.
somavakien@pop-os:~$ sudo apt install openrazer-meta
Packet lists are read... Done
Dependency tree is built.       
Status information is read.... Ready
Some packages could not be installed. This can mean that
you have requested an impossible situation or, if you have
unstable distribution, that some required packages are still available
were not created or have not yet left Incoming.
The following information may help you to resolve the situation:

The following packages have unmet dependencies:
 openrazer-meta : Depends on: openrazer-daemon (= 2.9.0-1~585~ubuntu20.04.1) but should not be installed
                  Depends on: python3-openrazer (= 2.9.0-1~585~ubuntu20.04.1) but should not be installed
E: Problems cannot be corrected, you have retained defective packages.

Translated with www.DeepL.com/Translator (free version)`

lah7, 2020-12-08:
From the output of `sudo apt update`, it doesn't look like you have the main, universe or multiverse repositories enabled, so none of the packages required for this software can be installed.

Please run `software-properties-gtk` and :ballot_box_with_check:  check all the repositories in the **Pop Software** tab. Then, go to the **Other Software** tab and **remove** the duplicate OpenRazer/Polychromatic `daily` entries. When closing, make sure to **reload** the package cache.

You have some updates - if any are for the kernel (`linux-generic`), you should update the system first otherwise OpenRazer won't be able to install the correct version properly.

Finally, try `sudo apt update` and `sudo apt install openrazer-meta` again. :crossed_fingers:

Somamint, 2020-12-08:
It worked.. installed everything nicely! But how to open it?

```
sudo apt install openrazer-meta
Package lists are being read ... Done
Dependency tree is built.
Status information is read in .... Done
The following packages were installed automatically and are no longer required:
  gnome-shell-extension-desktop-icons libnvidia-cfg1-440 libnvidia-common-440
  libnvidia-compute-440 libnvidia-compute-440: i386 libnvidia-decode-440
  libnvidia-decode-440: i386 libnvidia-encode-440 libnvidia-encode-440: i386
  libnvidia-extra-440 libnvidia-fbc1-440 libnvidia-fbc1-440: i386
  libnvidia-gl-440 libnvidia-ifr1-440 libnvidia-ifr1-440: i386
  linux-headers-5.4.0-7642 linux-headers-5.4.0-7642-generic
  linux-image-5.4.0-7642-generic linux-modules-5.4.0-7642-generic
  linux-modules-extra-5.4.0-7642-generic mousetweaks nvidia-compute-utils-440
  nvidia-dkms-440 nvidia-kernel-common-440 nvidia-kernel-source-440
  nvidia-utils-440 xserver-xorg-video-nvidia-440
Use "sudo apt autoremove" to remove them.
The following additional packages are installed:
  libblas3 libgfortran5 liblapack3 openrazer-daemon openrazer-driver-dkms
  python3-daemonize python3-notify2 python3-numpy python3-openrazer
  python3-pyudev xautomation
Suggested packages:
  openrazer-doc python-daemonize-doc gfortran python-numpy-doc python3-pytest
  python3-numpy-dbg
The following NEW packages will be installed:
  libblas3 libgfortran5 liblapack3 openrazer-daemon openrazer-driver-dkms
  openrazer-meta python3-daemonize python3-notify2 python3-numpy
  python3-openrazer python3-pyudev xautomation
0 updated, 12 reinstalled, 0 removed, and 134 not updated.
5,994 kB of archives have to be downloaded.
After this operation, an additional 26.6 MB of disk space will be used.
Do you want to continue? [Y / n] y
Get: 1 http://archive.ubuntu.com/ubuntu focal / main amd64 libblas3 amd64 3.9.0-1build1 [142 kB]
Get: 2 http://archive.ubuntu.com/ubuntu focal / main amd64 liblapack3 amd64 3.9.0-1build1 [2.154 kB]
Get: 3 http://archive.ubuntu.com/ubuntu focal / main amd64 python3-pyudev all 0.21.0-3ubuntu1 [37.1 kB]
Get: 4 http://archive.ubuntu.com/ubuntu focal / universe amd64 python3-notify2 all 0.3-4 [11.0 kB]
Get: 5 http://archive.ubuntu.com/ubuntu focal / universe amd64 python3-daemonize all 2.4.7-4 [6.520 B]
Get: 6 http://archive.ubuntu.com/ubuntu focal / universe amd64 xautomation amd64 1.09-2 [37.6 kB]
Get: 7 http://archive.ubuntu.com/ubuntu focal / main amd64 python3-numpy amd64 1: 1.17.4-5ubuntu3 [2.725 kB]
Get: 8 http://security.ubuntu.com/ubuntu focal-security / main amd64 libgfortran5 amd64 10.2.0-5ubuntu1 ~ 20.04 [736 kB]
Get: 9 http://ppa.launchpad.net/openrazer/stable/ubuntu focal / main amd64 openrazer-driver-dkms all 2.9.0 ~ ubuntu20.04.1 [50.3 kB]
Get: 10 http://ppa.launchpad.net/openrazer/stable/ubuntu focal / main amd64 openrazer-daemon all 2.9.0 ~ ubuntu20.04.1 [57.4 kB]
Get: 11 http://ppa.launchpad.net/openrazer/stable/ubuntu focal / main amd64 python3-openrazer all 2.9.0 ~ ubuntu20.04.1 [24.8 kB]
Get: 12 http://ppa.launchpad.net/openrazer/stable/ubuntu focal / main amd64 openrazer-meta all 2.9.0 ~ ubuntu20.04.1 [13.6 kB]
5,994 kB were fetched in 12 s (512 kB / s).
Previously unselected package libblas3: amd64 is selected.
(Read database ... 367591 files and directories are currently installed.)
Preparation for unpacking ... / 00-libblas3_3.9.0-1build1_amd64.deb ...
Unpacking libblas3: amd64 (3.9.0-1build1) ...
Previously unselected package libgfortran5: amd64 is selected.
Preparation for unpacking ... / 01-libgfortran5_10.2.0-5ubuntu1 ~ 20.04_amd64.d
eb ...
Unpacking libgfortran5: amd64 (10.2.0-5ubuntu1 ~ 20.04) ...
Previously unselected package liblapack3: amd64 is selected.
Preparation for unpacking ... / 02-liblapack3_3.9.0-1build1_amd64.deb ...
Unpacking liblapack3: amd64 (3.9.0-1build1) ...
Previously unselected package openrazer-driver-dkms is selected.
Preparation for unpacking ... / 03-openrazer-driver-dkms_2.9.0 ~ ubuntu20.04.1_
all.deb ...
Extracting openrazer-driver-dkms (2.9.0 ~ ubuntu20.04.1) ...
Previously unselected package python3-pyudev is selected.
Preparation for unpacking ... / 04-python3-pyudev_0.21.0-3ubuntu1_all.deb ...
Extracting python3-pyudev (0.21.0-3ubuntu1) ...
Previously unselected package python3-notify2 is selected.
Preparation for unpacking ... / 05-python3-notify2_0.3-4_all.deb ...
Extracting python3-notify2 (0.3-4) ...
Previously unselected package python3-daemonize is selected.
Preparation for unpacking ... / 06-python3-daemonize_2.4.7-4_all.deb ...
[... 56 lines cut ...]
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - installation
   - Installing to /lib/modules/5.8.0-7630-generic/kernel/drivers/hid/

depmod ....

DKMS: install completed.

ADD YOUR USER TO THE PLUGDEV GROUP: 'sudo gpasswd -a $ USER plugdev'
python3-daemonize (2.4.7-4) is set up ...
liblapack3: amd64 (3.9.0-1build1) is being set up ...
update-alternatives: /usr/lib/x86_64-linux-gnu/lapack/liblapack.so.3 will be used
det to /usr/lib/x86_64-linux-gnu/liblapack.so.3 (liblapack.so.3-x86_64-linux-gn
u) to provide in automatic mode
setting up openrazer-daemon (2.9.0 ~ ubuntu20.04.1) ...
python3-numpy (1: 1.17.4-5ubuntu3) is set up ...
python3-openrazer (2.9.0 ~ ubuntu20.04.1) is set up ...
openrazer-meta (2.9.0 ~ ubuntu20.04.1) is being set up ...
Processing triggers for man-db (2.9.1-1) ...
Processing triggers for libc-bin (2.31-0ubuntu9.1) ...
somavakien @ pop-os: ~ $ openrazer-meta
openrazer-meta: command not found.
```

lah7, 2020-12-08:
Good to hear! OpenRazer on its own is just a driver and daemon. You'll need to install a frontend (like [Polychromatic](https://polychromatic.app)) [or any other](https://github.com/openrazer/openrazer#applications) of your choice.

:point_up:  Make sure you do run that plugdev group command and reboot, otherwise frontends won't work either.

    sudo gpasswd -a $ USER plugdev

closed 2020-12-08 by lah7