openrazer/openrazer #1373: openrazer-daemon no init

adrianro9224, 2020-12-16:
### Distribution
Ubuntu 18.04.5 LTS

v2.9.0
### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

The openrazer-daemon no init from the last update.

the result  of openrazer-daemon -Fv is:
[image]

lah7, 2020-12-17:
Looks similar to #1354 (and #1351). Some invalid data got written to persistence when the certain effects were selected. This should have been fixed in 604f1cab3a899a1877e787e0e6e42b58c289f3ed.

You can either:

* Delete the file at `~/.config/openrazer/persistence.conf`,  which will reset effects to spectrum. This error would happen again upon choosing ripple and restarting the daemon.
* Edit the file and replace `\x02` with `2`.

Make sure you've got the latest build from master containing the fix:

    sudo add-apt-repository ppa:openrazer/daily
    sudo apt install openrazer-daemon

adrianro9224, 2020-12-17:
Thanks guys the option: Delete the file at ~/.config/openrazer/persistence.conf works for me.

closed 2020-12-17 by adrianro9224