openrazer/openrazer #1378: Daemon does not detect my Cynosa Chroma on Debian Unstable.

sanacl, 2020-12-19:
When I execute the razergenie tool I get the message "The daemon didn't detect a device that is connected".

I'm using Debian unstable and the following Debian packages:
```
ii  openrazer-daemon                              2.9.0+dfsg-1                        all          OpenRazer peripheral drivers (daemon)
ii  openrazer-driver-dkms                         2.9.0+dfsg-1                        all          OpenRazer peripheral drivers (DKMS)
ii  openrazer-meta                                2.9.0+dfsg-1                        all          OpenRazer peripheral drivers (metapackage)
ii  python3-openrazer                             2.9.0+dfsg-1                        all          OpenRazer peripheral drivers (Python 3)
ii  razergenie                                    0.9.0-0                             amd64        Qt application for configuring your Razer devices
```
The device I'm trying to customize is the Cynosa Chroma.

I've executed the commands you detailed in the troubleshooting section and this is the output:
```
┬─[luis@gluten:~]─[19:04:17]
╰─>$ set versionor "2.9.0"

╰─>$ sudo dkms install openrazer-driver/$versionor
Module openrazer-driver/2.9.0 already installed on kernel 5.8.0-1-amd64/x86_64

╰─>$ openrazer-daemon -Fv
Unable to lock on the pidfile.
```

lah7, 2020-12-19:
There's not enough info to go by here - please could you print the last lines of the log?

    tail ~/.local/share/openrazer/logs/razer.log

If it says your user is not part of the `plugdev` group, run this command, log out and log back in:

    sudo gpasswd -a $USER plugdev

If the last line says `Serving DBus`, and you have a UEFI system - secure boot may be preventing the driver from loading, in which probing would output `Operation not permitted` or similar:

    sudo modprobe razerkbd

sanacl, 2020-12-21:
@lah7 

The last line says `Serving DBus` yep.

```
╰─>$ tail ~/.local/share/openrazer/logs/razer.log
2020-12-19 18:34:12 | razer                          | INFO     | Initialising Daemon (v2.9.0). Pid: 1395028
2020-12-19 18:34:12 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2020-12-19 18:34:12 | razer                          | INFO     | Serving DBus
```
.. but modprobe did not solve it :( 
```
╰─>$ lsmod|grep razerkbd
razerkbd               81920  0
usbcore               323584  11 xhci_hcd,ehci_pci,usbhid,usblp,razerkbd,usb_storage,uvcvideo,ehci_hcd,btusb,xhci_pci,uas
hid                   147456  6 i2c_hid,usbhid,hid_multitouch,razerkbd,hid_generic,hid_plantronics
```

lah7, 2020-12-21:
It looks like it should be working (driver and daemon loaded)... unless...

* Is your Razer Cynosa Chroma definitely a `1532:022A`?  

      lsusb | grep 1532

* Does the running kernel version match the one that the driver was installed to?

      uname -r

    Your output earlier suggests this should be along the lines of `5.8.0-1-amd64`. If it's different, you'll likely need to use `dkms` to install the driver to the right kernel, or remove the old kernels.

sanacl, 2020-12-21:
@lah7 It is a `1532:022A`, is it bad news? :-/

```
╰─>$ lsusb | grep 1532
Bus 002 Device 083: ID 1532:022a Razer USA, Ltd Cynosa Chroma
```
```
╰─>$ uname -r
5.8.0-1-amd64
```

Thanks for your help :)

lah7, 2020-12-21:
You're welcome! No bad news with `1532:022A` :slightly_smiling_face:  Just double checking in the odd chance Razer had a different model with the same name.

I'm a bit stumped as we've exhausted the usual culprits from the [troubleshooting guide](https://github.com/openrazer/openrazer/wiki/Troubleshooting). Let us know the output of these logs for clues - it's very likely to be a driver or Debian-specific issue.

    sudo journalctl -b | grep -i razer_mount
    dmesg

The daemon would normally see the device with a properly loaded driver by now. You could also check the entire daemon output again (ignoring `Loaded device specification`) in case there's anything in there:

    openrazer-daemon -s
    openrazer-daemon -Fv

Maybe also try re-plugging in the keyboard as the kernel might (finally?) bind `razerkbd` to the keyboard.

sanacl, 2020-12-21:
Bingo @lah7 , it is an issue with the permissions.

```
2020-12-21 16:59:27 | razer                          | INFO     | Found device.0: 0003:1532:022A.051D
2020-12-21 16:59:27 | razer                          | CRITICAL | Could not access /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2.3/2-2.3:1.2/0003:1532:022A.051D/device_type, file is not owned by plugdev
2020-12-21 16:59:27 | razer                          | DEBUG    | Adding razer.devices.enableTurnOffOnScreensaver method to DBus

```

After adding my username to the plugdev group as mentioned [here](https://github.com/openrazer/openrazer/wiki/Troubleshooting#daemon) it started to work

Thanks for your time. Appreciated!

closed 2020-12-21 by sanacl