openrazer/openrazer #1381: Support for Razer Mouse Bungee V3 Chroma

oxhak, 2020-12-21:
### Device Info

* Name: Razer Mouse Bungee V3 Chroma
* Product Number: R-R-RAZ-RC21-0152
* Model Number: RC21-0152

### Device Information
```
Bus 002 Device 013: ID 1532:0f1d Razer USA, Ltd Razer Mouse Bungee V3 Chroma
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0f1d 
  bcdDevice            2.00
  iManufacturer           1 RAZER
  iProduct                2 Razer Mouse Bungee V3 Chroma
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x003b
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               2.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
```

```
lrwxrwxrwx 1 root root  10 21 déc.  10:36 usb-RAZER_Razer_Mouse_Bungee_V3_Chroma-event-if01 -> ../event25
lrwxrwxrwx 1 root root  10 21 déc.  10:36 usb-RAZER_Razer_Mouse_Bungee_V3_Chroma-event-mouse -> ../event21
lrwxrwxrwx 1 root root  10 21 déc.  10:36 usb-RAZER_Razer_Mouse_Bungee_V3_Chroma-if01-event-kbd -> ../event22
lrwxrwxrwx 1 root root   9 21 déc.  10:36 usb-RAZER_Razer_Mouse_Bungee_V3_Chroma-mouse -> ../mouse1
```

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on Razer Synapse>

z3ntu, 2020-12-21:
Implemented in #1370 , please test that if possible.

oxhak, 2020-12-21:
> Implemented in #1370 , please test that if possible.
 
 Switched to git version on Archlinux, but still cannot get it detected, this is what I get when I run polychromatic-controller, I also tried razercommander but no devices are detected, any idea please ?
 
 Daemon is running with no issue: 
`2020-12-21 12:06:46 | razer                          | INFO     | Initialising Daemon (v2.9.0). Pid: 26024
2020-12-21 12:06:46 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2020-12-21 12:06:46 | razer                          | INFO     | Serving DBus
`
[image]

z3ntu, 2020-12-21:
See https://github.com/openrazer/openrazer/wiki/Building-a-package#build-packages-from-a-different-github-repository

oxhak, 2020-12-21:
> See https://github.com/openrazer/openrazer/wiki/Building-a-package#build-packages-from-a-different-github-repository

Sorry I did not notice that it was another repository, I had to switch to a kernel other than the lts to be able to install it, in the end I have no change, same result as on the screenshot.

oxhak, 2020-12-21:
Sorry i was not on the right branch, i will try again.

oxhak, 2020-12-21:
Ok it works now and the parameters are persistent after a reboot, thank you very much for your help and your time, the wave effect is a bit fast though.

[image]

nightsky30, 2020-12-21:
@z3ntu @oxhak I could add another slower wave specifically for this device. It's using the same speed value as other accessories 0x28. Maybe because this device only has 8 LEDs vs 15, it cycles faster?? And thank you for testing as well :smile:

oxhak, 2020-12-21:
> @z3ntu @oxhak I could add another slower wave specifically for this device. It's using the same wave speed value as other accessories 0x28. Maybe because this device only has 8 LEDs vs 15, it cycles through faster?? And thank you for testing as well smile

Yes, it would be nice, thank you!

nightsky30, 2020-12-21:
@oxhak I just doubled the argument value before the report is sent since it has about half the LEDs. Now it's 0x4e. I also tried multiplying the hex value by 2 in code, but that gave us 0x50 which was IMO a bit too slow. Though doubling like that would be easier to maintain if the default accessory value were changed in the future. For speed comparison I'm using a Firefly.

@z3ntu Are you ok with these changes? Any concerns?

oxhak, 2020-12-21:
@nightsky30 Speed is better but it looks like some "frames" are missing, it's not smooth as it is on windows.

Windows:

https://user-images.githubusercontent.com/2832152/102819855-919c4f00-43d4-11eb-8f0f-3201bad0c8d4.mp4

Your build:

https://user-images.githubusercontent.com/2832152/102820014-d3c59080-43d4-11eb-9234-ba1f3ee0ec86.mp4

oxhak, 2020-12-21:
@nightsky30 After comparison, the speed is too slow now, and before it seemed to be very little faster compared to the windows version, not x2 anyway.

nightsky30, 2020-12-21:
@oxhak I booted up my VM and took a video. Then reverted the changes and rebooted to use the default speed value under openrazer and took a second video. Running both videos side by side I honestly don't see a difference in speed as far as number of cycles is concerned. I think 0x28 is the correct speed value if not something very close to it.

> it looks like some "frames" are missing, it's not smooth as it is on windows.

Yes, and I believe the difference is how Synapse is creating the effect vs openrazer.

When I boot into my VM and set wave as the quick effect in Synapse, it does look a bit different. And I've also noticed that even with a VM given 8GB memory, there are times when the wave effect on the mouse bungee slows, lags, or stutters as if to pause for a moment when the system is under load.

Synapse is sending individual packets for the wave quick effect (and the others) as we've seen via wireshark even though there are hardware based effects built into the devices. This causes the lag under load while the device waits for frames to be sent. And it accounts for the difference in pattern as well (what you described as lost frames). Whatever pattern Synapse is sending is slightly different than what the hardware based effect is doing.

Openrazer, on the other hand, is using the hardware based effects. It sends 1 command report, and the device pretty much does the rest. There's no way in openrazer to recreate the exact pattern Synapse is sending out unless openrazer were to implement packet spamming and just copy exactly what Synapse does.

I hope this clarifies :smile: I think the best we can do with the current daemon is approximate.

oxhak, 2020-12-22:
@nightsky30 Yes I think it's ok on the old commit, I reverted to it, I will continue to follow the progress, thanks again!

z3ntu, 2020-12-22:
Let's keep this open until #1370 is merged.

closed 2020-12-27 by z3ntu