openrazer/openrazer #1383: 🖱 Support for Razer Lancehead Tournament Mercury Edition

TGotwig, 2020-12-22:
### Device Info

* Name: Razer Lancehead Tournament Mercury Edition
* Product Number: RZ01-02130200
* Model Number: RZ01-0213

### Device Information
```
Please post the output of
    lsusb -d 1532: -v

Bus 001 Device 006: ID 1532:0060 Razer USA, Ltd Razer BlackWidow Lite
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0060 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
[... 183 lines cut ...]

total 0
drwxr-xr-x 2 root root 280 Dez 22 19:59 ./
drwxr-xr-x 4 root root 680 Dez 22 19:59 ../
lrwxrwxrwx 1 root root   9 Dez 22 19:13 usb-Azurewave_USB2.0_HD_IR_UVC_WebCam_200901010001-event-ir -> ../event5
lrwxrwxrwx 1 root root   9 Dez 22 19:13 usb-Azurewave_USB2.0_HD_IR_UVC_WebCam_200901010001-if02-event-ir -> ../event6
lrwxrwxrwx 1 root root  10 Dez 22 19:13 usb-Razer_Razer_BlackWidow_Lite-event-if01 -> ../event11
lrwxrwxrwx 1 root root   9 Dez 22 19:13 usb-Razer_Razer_BlackWidow_Lite-event-kbd -> ../event7
lrwxrwxrwx 1 root root   9 Dez 22 19:13 usb-Razer_Razer_BlackWidow_Lite-if01-event-kbd -> ../event8
lrwxrwxrwx 1 root root  10 Dez 22 19:13 usb-Razer_Razer_BlackWidow_Lite-if02-event-mouse -> ../event12
lrwxrwxrwx 1 root root   9 Dez 22 19:13 usb-Razer_Razer_BlackWidow_Lite-if02-mouse -> ../mouse2
lrwxrwxrwx 1 root root  10 Dez 22 19:59 usb-Razer_Razer_Lancehead_Tournament_Edition-event-if01 -> ../event17
lrwxrwxrwx 1 root root  10 Dez 22 19:59 usb-Razer_Razer_Lancehead_Tournament_Edition-event-mouse -> ../event13
lrwxrwxrwx 1 root root  10 Dez 22 19:59 usb-Razer_Razer_Lancehead_Tournament_Edition-if01-event-kbd -> ../event14
lrwxrwxrwx 1 root root  10 Dez 22 19:59 usb-Razer_Razer_Lancehead_Tournament_Edition-if02-event-kbd -> ../event18
lrwxrwxrwx 1 root root   9 Dez 22 19:59 usb-Razer_Razer_Lancehead_Tournament_Edition-mouse -> ../mouse3
```

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on Razer Synapse>

z3ntu, 2020-12-22:
This device is already supported, see readme:

> Razer Lancehead Tournament Edition | 1532:0060

closed 2020-12-22 by z3ntu

TGotwig, 2020-12-22:
@z3ntu but it can't be recognised? 🤔

[image]

TGotwig, 2020-12-22:
Ahh probably because of this 🧐
```fish
❯ sudo modprobe razerkbd
modprobe: ERROR: could not insert 'razerkbd': Operation not permitted
```

TGotwig, 2020-12-22:
@z3ntu now it works!
Following https://github.com/openrazer/openrazer/wiki/Troubleshooting helped 😊
Sorry for the duplicate 🙈

But I can't set the colors for my side-buttons, is this a known issue?

[image]

z3ntu, 2020-12-22:
Known limitation of the stable polychromatic version, try the latest code from git.