openrazer/openrazer #1386: Razer Blade Pro (Late 2016, [1532:0210]) keyboard not recognized

jackz314, 2020-12-24:
### Distribution

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

My Razer Blade Pro (Late 2016, [1532:0210])'s keyboard is not recognized by the driver, it does show up in lsusb:

> Bus 001 Device 005: ID 1532:0210 Razer USA, Ltd Razer Blade Pro
> Bus 001 Device 003: ID 1532:0059 Razer USA, Ltd Razer Lancehead
> Bus 001 Device 007: ID 1532:0228 Razer USA, Ltd Razer BlackWidow Elite

My other Razer devices show up in `/sys/bus/hid/drivers/razerkbd` and `/sys/bus/hid/devices/`, but not the RBP keyboard.

I'm not sure if this is a linux/ubuntu issue or an OpenRazer issue, but I thought I'd ask here to see what are some potential problems.

z3ntu, 2020-12-27:
Any useful output in `sudo journalctl -b | grep -i razer_mount` ?

jackz314, 2020-12-29:
Not really, It just wasn't being picked up at all. Anyway, I think it's a kernel issue because after I upgraded to Ubuntu 20.10 from 20.04 (and the kernel versions as well), this problem went away.

closed 2020-12-29 by jackz314