openrazer/openrazer #1387: could not find a distribution template for Pop/groovy

Tommimon, 2020-12-25:
OS: Pop!_OS 20.10 x86_64 
Kernel: 5.8.0-7630-generic

I can't add ppa repository, when I run:
`sudo add-apt-repository ppa:openrazer/stable`

I get this error:
```
Traceback (most recent call last):
  File "/usr/bin/add-apt-repository", line 330, in <module>
    addaptrepo = AddAptRepository()
  File "/usr/bin/add-apt-repository", line 35, in __init__
    self.distro.get_sources(self.sourceslist)
  File "/usr/lib/python3/dist-packages/aptsources/distro.py", line 91, in get_sources
    raise NoDistroTemplateException(
aptsources.distro.NoDistroTemplateException: Error: could not find a distribution template for Pop/groovy
```

Actually RazerGenie tells me that daemon is installed and working (version 2.5.0), I think I installed it from the Pop shop. Despite that RazerGenie is unable to detect my Huntsman keyboard

TermoZour, 2020-12-25:
Try cloning the repo locally then perform the "build package" instructions for Ubuntu.
RazerGenie finds my devices and the daemon is working properly for me on Pop!_OS 20.04

Tommimon, 2020-12-25:
This worked for me, thanks.

Being able to use OpenRazer is grate but I'll like to add the ppa to receive automatic updates, any idea of why I can't add the ppa?

lah7, 2020-12-25:
According to [this reddit comment](https://www.reddit.com/r/pop_os/comments/bjnvk1/how_do_i_add_new_repositories/), you can add the PPA via the Pop Shop, but I would've thought `add-apt-repository` should still work in Pop!_OS

Try adding the PPA the "Debian" way:

```
echo "deb http://ppa.launchpad.net/openrazer/stable/ubuntu groovy main" | sudo tee -a /etc/apt/sources.list.d/openrazer.list
sudo apt-key adv --recv-key --keyserver keyserver.ubuntu.com 903936CAB6049E2E6C33D5D8073E051D7B2AEE37
sudo apt-get update
sudo apt-get dist-upgrade
```

TermoZour, 2020-12-26:
Adding PPAs from the terminal sure works for me on Pop!_OS 20.04 LTS. They then show up in Pop!_Shop.

Tommimon, 2020-12-26:
Thanks to everyone.

 Both adding ppa with `echo "deb http://ppa.launchpad.net/openrazer/stable/ubuntu groovy main" | sudo tee -a /etc/apt/sources.list.d/openrazer.list` or adding it from Pop!_shop leads to the same result:

Running `sudo apt-key adv --recv-key --keyserver keyserver.ubuntu.com 903936CAB6049E2E6C33D5D8073E051D7B2AEE37` gives this error:
```
Warning: apt-key is deprecated. Manage keyring files in trusted.gpg.d instead (see apt-key(8)).
Executing: /tmp/apt-key-gpghome.yPrPRVk0cp/gpg.1.sh --recv-key --keyserver keyserver.ubuntu.com 903936CAB6049E2E6C33D5D8073E051D7B2AEE37
gpg: keyserver receive failed: Server indicated a failure
```

And running `sudo apt-get update` gives this:
```
W: GPG error: http://ppa.launchpad.net/openrazer/stable/ubuntu groovy InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 073E051D7B2AEE37
E: The repository 'http://ppa.launchpad.net/openrazer/stable/ubuntu groovy InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

I never experienced errors like this when I was using Pop!_OS 20.04, maybe is their fault, they broke something in the 20.10 update  and I'm just asking in the wrong place.

closed 2021-06-20 by github-actions[bot]