openrazer/openrazer #1389: Support for [device name here]

YanDoroshenko, 2020-12-25:
### Device Info

* Name: Pro Click
* Product Number: RZ01-02990100
* Model Number: RZ01-0299

### Device Information

Please post the output of
```
    lsusb -d 1532: -v

Bus 001 Device 028: ID 1532:0077 Razer USA, Ltd Razer Pro Click
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0077
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Pro Click
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0
      bInterfaceProtocol      1 Keyboard
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0
      bInterfaceProtocol      1 Keyboard
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
```

```
Please post the output of
    ls -lah /dev/input/by-id/

total 0
drwxr-xr-x 2 root root 160 гру 25 15:49 .
drwxr-xr-x 4 root root 700 гру 25 15:49 ..
lrwxrwxrwx 1 root root  10 гру 25 15:45 usb-Razer_Razer_Pro_Click-event-if01 -> ../event13
lrwxrwxrwx 1 root root  10 гру 25 15:45 usb-Razer_Razer_Pro_Click-event-mouse -> ../event11
lrwxrwxrwx 1 root root  10 гру 25 15:45 usb-Razer_Razer_Pro_Click-if01-event-kbd -> ../event12
lrwxrwxrwx 1 root root  10 гру 25 15:45 usb-Razer_Razer_Pro_Click-if02-event-kbd -> ../event16
lrwxrwxrwx 1 root root   9 гру 25 15:45 usb-Razer_Razer_Pro_Click-mouse -> ../mouse0
lrwxrwxrwx 1 root root  10 гру 25 14:21 usb-SteelSeries_SteelSeries_KLC-event-if01 -> ../event19
```

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
[image]
[image]
[image]

YanDoroshenko, 2020-12-25:
Please let me know how can I provide packet captures in an optimal way. I was unable to find anything useful in the Wireshark captures, but I'm far from being an expert.

JulianGro, 2021-03-25:
@YanDoroshenko You followed https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol right?
I would be up for trying to help you out with wireshark. If you got USB passthrough going it *should* be pretty easy. At least that would be the bigger hurdle for me.

YanDoroshenko, 2021-03-26:
My mouse died a couple of days ago, so unfortunately I can't help :(

azhard, 2021-05-02:
I'd love to get support for this as a I have a pro click, @JulianGro if you're still able to help let me know what would be needed on my side

Hedronmx, 2021-06-28:
I can help out! I have both the Pro click and Pro type. What should I do?

z3ntu, 2021-06-30:
@Hedronmx If you follow https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol and provide the pcaps then I can look at them and figure out some of the details that are needed for implementing this.

bmarengo, 2021-09-07:
I pretend to buy the Pro Click and as soon as I do, I will certainly provide the pcaps.

I am just worried about it working on my HP notebook with Ubuntu 18.04. 
Without this driver, will I be able to just use it? I mean, is this driver only for configuring it? 
Is it possible to configure it from a Windows PC and then use it on my Ubuntu PC?

I hope you can help me so I make the decision!

Hedronmx, 2021-09-07:
I used pro click on eos 5 which is based on Ubuntu 18.04. It works correctly without windows  @bmarengo

bmarengo, 2021-09-07:
@Hedronmx and how did you customized the extra-buttons? or you just didn't?

Hedronmx, 2021-09-07:
I did it with shortcuts inside elementary os. Most extra buttons came mapped so I just assigned those to a shortcut

bmarengo, 2021-09-11:
@z3ntu @JulianGro 
Hey guys,
I got everything to work, but when I assign functions to the mouse on SYnapse (inside the VM) WIreshark on the host does not capture leftover data
[image]

any advice?

z3ntu, 2021-10-01:
@bmarengo Wireshark loves to put the actual data into a couple of different fields. Can you upload the pcap, then I'll take a look at them.

lukassup, 2022-04-06:
Hi, sharing some `.pcapng` files. Features used should be in the filenames. I can do more if you would like to:

https://drive.google.com/file/d/1KTmAZtfFQBJcLH6gesUQS_zBHGnRQtk1/view?usp=sharing

lukassup, 2022-04-06:
output for: `lsusb -d 1532: -v` :
https://gist.github.com/lukassup/61d323b08d3ae952e6b5a121708aea20

lukassup, 2022-04-06:
```
% ls -lah /dev/input/by-id/

total 0
drwxr-xr-x 2 root root 220 2022-04-06 19:32 .
drwxr-xr-x 4 root root 620 2022-04-06 19:32 ..
lrwxrwxrwx 1 root root  10 2022-04-06 19:32 usb-1532_Razer_Pro_Click_000000000000-event-if01 -> ../event10
lrwxrwxrwx 1 root root   9 2022-04-06 19:32 usb-1532_Razer_Pro_Click_000000000000-event-mouse -> ../event6
lrwxrwxrwx 1 root root   9 2022-04-06 19:32 usb-1532_Razer_Pro_Click_000000000000-if01-event-kbd -> ../event9
lrwxrwxrwx 1 root root   9 2022-04-06 19:32 usb-1532_Razer_Pro_Click_000000000000-if02-event-kbd -> ../event7
lrwxrwxrwx 1 root root   9 2022-04-06 19:32 usb-1532_Razer_Pro_Click_000000000000-mouse -> ../mouse1
```

lukassup, 2022-04-06:
Threw [a quick Python script to parse protocol](https://gist.github.com/lukassup/2a94f13a5d4b37c84c256062752224be) as [defined in your guide](https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol#phase-4---decoding-the-protocol), here are some results:

```
% tshark -r set-dpi-200.pcapng -T fields -e usb.data_fragment  | ./parse_razer_proto.py
start, id, reserved, num_params, command, sub_cmd, params
00, 1f, 04, 26, 06, 01, 01010000c800c800000000000000000000000
00, 1f, 04, 07, 05, 00, 00c800
00, 1f, 04, 07, 05, 00, 00c800
00, 1f, 04, 07, 05, 01, 00c800

% tshark -r set-dpi-1800.pcapng  -T fields -e usb.data_fragment  | ./parse_razer_proto.py
start, id, reserved, num_params, command, sub_cmd, params
00, 1f, 04, 26, 06, 01, 0101000708070800000000000000000000000
00, 1f, 04, 07, 05, 00, 070807
00, 1f, 04, 07, 05, 00, 070807
00, 1f, 04, 07, 05, 01, 070807

% tshark -r set-dpi-16000.pcapng   -T fields -e usb.data_fragment  | ./parse_razer_proto.py
start, id, reserved, num_params, command, sub_cmd, params
00, 1f, 04, 26, 06, 01, 0101003e803e8000000000000000000000000
00, 1f, 04, 07, 05, 00, 3e803e
00, 1f, 04, 07, 05, 00, 3e803e
00, 1f, 04, 07, 05, 01, 3e803e
```

lukassup, 2022-04-06:
```
% tshark -r set-polling-rate-125.pcapng -T fields -e usb.data_fragment  | ./parse_razer_proto.py 
start, id, reserved, num_params, command, sub_cmd, params
00, 1f, 00, 01, 05, 08,  # sets 125hz polling rate

% tshark -r set-polling-rate-500.pcapng -T fields -e usb.data_fragment  | ./parse_razer_proto.py 
start, id, reserved, num_params, command, sub_cmd, params
00, 1f, 00, 01, 05, 02,   # sets 500hz polling rate

% tshark -r set-polling-rate-1000.pcapng -T fields -e usb.data_fragment  | ./parse_razer_proto.py 
start, id, reserved, num_params, command, sub_cmd, params
00, 1f, 00, 01, 05, 01,  # sets 1000hz polling rate
```

Firestar-Reimu, 2022-04-07:
How does Pro Click support now? I am going to buy a Pro Click **Mini**, please add it to this Device Support issue, thanks!

https://www.razer.com/gaming-mice/razer-pro-click-mini/RZ01-03990100-R3U1#specs

[image]

z3ntu, 2022-04-07:
Hi everyone,

please test the PR #1788 that adds support for this device. You can find build instructions here: https://github.com/openrazer/openrazer/wiki/Building-a-package

Megaleo, 2022-04-07:
My Pro Click just arrived at the doorstep today and I can confirm that PR #1788 works as intended. Thanks @lukassup and @z3ntu!

[image]

Naive question: can I customize the actions of the mouse buttons?

lukassup, 2022-04-08:
Seems like no other mice implement such features yet:

https://github.com/openrazer/openrazer/blob/master/daemon/openrazer_daemon/hardware/mouse.py

My guess is that it will be left for future!

z3ntu, 2022-04-08:
@Megaleo Thanks a lot for testing!

Key rebinding you can do with e.g. https://github.com/sezanzeb/input-remapper

PR #1788 (2022-04-08): feature: implement razer pro click support
Implements some support for Razer Pro Click Wired & Receiver. Bluetooth does not create an event device file so maybe we can't control it over Bluetooth.

So far in `polychromatic` I'm able to control DPI (incl DPI stages), idle time, polling rate, low battery threshold.

Not sure which `razer_get_report` wait times to choose for the driver.

Let me know if you would like to make any amends.

Should close #1389
closed 2022-04-08 by z3ntu