openrazer/openrazer #1392: BlackWidow V3 support

miramar-labs, 2020-12-28:
I was having a go at adding support for the BlackWidow V3 today (without sniffing the USB comms) .... and I got it to show up in openrazer ui but looking at dmesg I saw errors:

razer driver: Invalid USB response. USB Report length: 

any idea how to fix that?

TheAlmightyZach, 2020-12-28:
Attempting to use Razer macOS (which uses these drivers) With the BlackWidow V3. Hoping for support soon as well!

miramar-labs, 2020-12-28:
also seeing:
WARNING  | No event files for KeyWatcher

miramar-labs, 2020-12-28:
I tried this to set the keyboard to static green .. but it didn't seem to work:
echo -n -e "\x00\xB3\x00" > /sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2.1/1-2.1.3/1-2.1.3:1.2/0003:1532:024E.0007/matrix_effect_static

which might explain why setting effect to static green in the openrazer UI also seems to have no effect

I'm guessing this keyboard's api is a little different but unsure how to reverse engineer further ... hopefully a guru will step in ....   :)

TermoZour, 2020-12-28:
I also have the same issue but with Nari Ultimate headphones. What packet size do you see reported in dmesg? I get -32 for some reason.

miramar-labs, 2020-12-28:
looking at the driver code where that message is written out I see a system call to to 'usb_control_msg()'
a negative return is an error code .. -ERRNO, (see /usr/include/asm/errno.h)
I think the next steps here might be to bust out wireshark as described in the reverse engineering notes, but not much time right now ..

TermoZour, 2020-12-28:
I think the error code means that our message either exceeds or is smaller than exactly 90 bytes, as specified in razerkbd_driver.h

nightsky30, 2020-12-29:
This seems to be a duplicate of #1319

z3ntu, 2021-01-02:
Yep, closing as duplicate of #1319

closed 2021-01-02 by z3ntu