openrazer/openrazer #1394: Sometimes, the Daemon does not start

mind-overflow, 2020-12-30:
### Distribution

Ubuntu 20.10 and Ubuntu 20.04 LTS

### Support

Every time I boot my PC, there is a chance that the openrazer daemon won't work.
I literally can't figure out why: I'm not changing anything, either hardware of software, between reboots.
I have a BlackWidow Elite and a Basilisk X HyperSpeed, and I'm using Polychromatic to set different things up.
Sometimes, after logging in, the daemon starts correctly and the polychromatic tray icon appears, allowing me to change settings. Some other times, after logging in, the daemon does not start. Polychromatic still loads up but it doesn't work. If I try to manually restart the daemon (either from Polychromatic's GUI or from Terminal) it just refuses to work. The Polychromatic GUI simply says that it was impossible to restart the daemon, while from terminal I get this output:

```bash
$ openrazer -Fv
Unable to lock on the pidfile.
# the daemon is active, but not working

$ pkill -e openrazer
openrazer-daemo killed (pid 28526)
$ pkill -e openrazer
openrazer-daemo killed (pid 28526)
# ^ it keeps saying it was killed, even if it actually doesn't get killed

$ sudo pkill -e openrazer
$ pkill -e openrazer
# ^ this actually kills it as I get no output after killing from root

$ openrazer -Fv
# now, the daemon starts and I get log messages. however, it still doesn't work.
```

This is the .local/share/openrazer/logs/razer.log file (only today's one; I also tried restarting the daemon):
```
2020-12-30 19:08:52 | razer                          | INFO     | Initialising Daemon (v2.9.0). Pid: 2798
2020-12-30 19:08:52 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2020-12-30 19:08:52 | razer                          | INFO     | Found device.0: 0003:1532:0228.0003
2020-12-30 19:08:52 | razer.device0                  | INFO     | Initialising device.0 RazerBlackWidowElite
2020-12-30 18:10:04 | razer.device0                  | INFO     | Putting device into driver mode. Daemon will handle special functionality
2020-12-30 18:10:14 | razer                          | INFO     | Found device.1: 0003:1532:0083.0006
2020-12-30 18:10:14 | razer.device1                  | INFO     | Initialising device.1 RazerBasiliskXHyperSpeed
2020-12-30 18:10:14 | razer                          | INFO     | Serving DBus
2020-12-30 18:50:22 | razer                          | INFO     | Stopping daemon on signal 15
2020-12-30 18:50:35 | razer                          | INFO     | Initialising Daemon (v2.9.0). Pid: 28320
2020-12-30 18:50:35 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2020-12-30 18:50:35 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2020-12-30 18:50:35 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus1800 ------------------------ (1532:0020)
2020-12-30 18:50:35 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus2000 ------------------------ (1532:005E)
2020-12-30 18:50:35 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusEliteDVaEdition ------------- (1532:006A)
2020-12-30 18:50:35 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusEssential ------------------- (1532:006B)

<OMITTED ALL THE OTHER DEVICES>

2020-12-30 18:50:35 | razer                          | DEBUG    | Loaded device specification: RazerViperUltimateWireless -------------- (1532:007B)
2020-12-30 18:50:35 | razer                          | INFO     | Found device.0: 0003:1532:0228.0003
2020-12-30 18:50:35 | razer.device0                  | INFO     | Initialising device.0 RazerBlackWidowElite
2020-12-30 18:50:45 | razer.device0                  | DEBUG    | getting serial:  count:1
2020-12-30 18:50:56 | razer.device0                  | DEBUG    | getting serial:  count:2
2020-12-30 18:51:02 | razer.device0                  | WARNING  | malformed serial: 'utf-8' codec can't decode byte 0xa1 in position 0: invalid start byte
2020-12-30 18:51:02 | razer.device0                  | DEBUG    | getting serial:  count:3
2020-12-30 18:51:12 | razer.device0                  | DEBUG    | getting serial:  count:4
2020-12-30 18:51:30 | razer                          | INFO     | Initialising Daemon (v2.9.0). Pid: 28526
2020-12-30 18:51:30 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2020-12-30 18:51:30 | razer                          | INFO     | Found device.0: 0003:1532:0228.0003
2020-12-30 18:51:30 | razer.device0                  | INFO     | Initialising device.0 RazerBlackWidowElite
2020-12-30 18:52:42 | razer.device0                  | INFO     | Putting device into driver mode. Daemon will handle special functionality
2020-12-30 18:52:52 | razer                          | INFO     | Found device.1: 0003:1532:0083.0006
2020-12-30 18:52:52 | razer.device1                  | INFO     | Initialising device.1 RazerBasiliskXHyperSpeed
2020-12-30 18:52:52 | razer                          | INFO     | Serving DBus

```

Polychromatic's GUI:
*(I know this is not the Polychromatic's issues page, but Polychromatic is relying on openrazer-daemon and the daemon itself is the issue, which can be seen from Polychromatic - I'm just using it as proof/explanation of the issue)*

[image]
[image]

lah7, 2020-12-31:
Something is causing D-Bus to hang - likely the driver, kernel or possibly some hardware related fault.

When it happens again, please check/post the output:

    dmesg

Try unplugging and replugging the devices, and see if the daemon restarts normally.

mind-overflow, 2021-06-07:
i'm closing this because i had no time to test and check this recently. might reopen if the issue is still there and when i have time to check better.
thank you!

closed 2021-06-07 by mind-overflow