openrazer/openrazer #1395: systemd[566]: Failed to start Daemon to manage razer devices in userspace

hanny00, 2020-12-31:
**Kernel: 5.10.2-2-MANJARO**

● openrazer-daemon.service - Daemon to manage razer devices in userspace
     Loaded: loaded (/usr/lib/systemd/user/openrazer-daemon.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Thu 2020-12-31 11:39:00 CET; 25s ago
       Docs: man:openrazer-daemon(8)
    Process: 1700 ExecStart=/usr/bin/openrazer-daemon -F (code=exited, status=1/FAILURE)
   Main PID: 1700 (code=exited, status=1/FAILURE)

Dez 31 11:39:00 manjaro-pc systemd[566]: Starting Daemon to manage razer devices in userspace...
Dez 31 11:39:00 manjaro-pc openrazer-daemon[1700]: Traceback (most recent call last):
Dez 31 11:39:00 manjaro-pc openrazer-daemon[1700]:   File "/usr/bin/openrazer-daemon", line 11, in <module>
Dez 31 11:39:00 manjaro-pc openrazer-daemon[1700]:     from openrazer_daemon.daemon import RazerDaemon, __version__
Dez 31 11:39:00 manjaro-pc openrazer-daemon[1700]: ModuleNotFoundError: No module named 'openrazer_daemon'
Dez 31 11:39:00 manjaro-pc systemd[566]: openrazer-daemon.service: Main process exited, code=exited, status=1/FAILURE
Dez 31 11:39:00 manjaro-pc systemd[566]: openrazer-daemon.service: Failed with result 'exit-code'.
Dez 31 11:39:00 manjaro-pc systemd[566]: Failed to start Daemon to manage razer devices in userspace.

lah7, 2020-12-31:
Manjaro just recently had a stable update - all your Python AUR packages need to be reinstalled against the new Python version:

    pacman -Qo /usr/lib/python3.8/

See also: https://github.com/openrazer/openrazer/issues/1362

hanny00, 2020-12-31:
pacman -Qo /usr/lib/python3.8/
/usr/lib/python3.8/ ist in openrazer-daemon 2.9.0-1 enthalten
/usr/lib/python3.8/ ist in polychromatic 0.3.12-3 enthalten
/usr/lib/python3.8/ ist in protontricks 1.4.3-1 enthalten
/usr/lib/python3.8/ ist in python-openrazer 2.9.0-1 enthalten
/usr/lib/python3.8/ ist in python-sip-pyqt5 4.19.19-3 enthalten
/usr/lib/python3.8/ ist in python-vdf 3.3-2 enthalten

Reinstall all Packages ?

lah7, 2020-12-31:
Yes, using an AUR helper. For example, `yay` or `pacaur`:

    yay -S openrazer-daemon polychromatic python-openrazer protontricks python-sip-pyqt5 python-vdf

closed 2021-06-20 by github-actions[bot]