openrazer/openrazer #1396: Support for Kraken X for console

Aslemammad, 2021-01-01:
### Device Info

* Name: Razer kraken X for console
* Product Number: <See under device>
* Model Number: RZ04-0289

z3ntu, 2021-01-02:
Please provide the output of `lsusb | grep 1532`

Aslemammad, 2021-01-02:
@z3ntu  nothing

z3ntu, 2021-01-02:
It seems the device only has a 3.5mm headphone jack, and no lighting so nothing we can control with openrazer. Please correct me if I'm wrong.

closed 2021-01-02 by z3ntu

Aslemammad, 2021-01-02:
@z3ntu  I wanted to access the 7.1 option.

TermoZour, 2021-01-02:
Are you talking about surround audio? If so, check this link out. It works for any headphones. Since you're not bound to Razer's audio driver and you can do whatever you want on Linux, make your own virtual surround. 

https://forum.endeavouros.com/t/howto-setting-up-virtual-surround-sound-for-headphones/6889