openrazer/openrazer #1398: Audio buttons - Razer Ornata Chroma V2

FIvanO, 2021-01-02:
### Device
Razer Ornata Chroma V2

### Problem
Audio buttons (switching songs and play/pause) and scroll (volume) don't work. Scroll works like a mouse scroll, but not a volume scroll.

Before installing drivers all buttons were working just fine.

Thanks!

lucallero, 2021-01-05:
I have exactly the same situation.

blastermaster77, 2021-01-12:
I installed the driver too and now the volume scroll and audio buttons do not work. And like @FIvanO comented it did work before the installation of the driver. keep the good work!

christophehenry, 2021-01-18:
Ok, so here are some inputs about this problem. Byt reading the code, it appears to me that the audio buttons are handled by the [`razer_raw_event` function](https://github.com/openrazer/openrazer/blob/master/driver/razerkbd_driver.c#L2137). By using a few `printk` functions around, I was able to determine this: the `size` argument is `22` when this function is triggered by media buttons and `fn + ` combination, not `16`, as filtered by [this line](https://github.com/openrazer/openrazer/blob/master/driver/razerkbd_driver.c#L2149]. So, this *should* explain why media button stop working when installing the OpenRazer driver.

**But**

After modifying the driver to debug it and installing it on my system, I noticed 2 things: 
1. the media buttons immediatly started working again (and the system was using the driver because I could see the `printk` statements on `dmesg`)
2. the Polychromatic controller immediately stopped working (crashing with core dumped)
3. the daemon is still working but I can see a critical log:

```
Could not access /sys/devices/pci0000:00/0000:00:14.0/usb1/1-5/1-5:1.2/0003:1532:025D.0004/device_type, file is not owned by plugdev
```

`1532:025D` corresponds to Ornata chroma V2's code. This may be related to the polychromatic crash (maybe Polychromatic is trying to set the light color and brightness on the keyboard via the daemon and the daemon returns an unexpected value)?

Anyway, the fact that the media button start working again when I remove the old driver and install my own version is disturbing. I'd like to dig deeper but I want the inputs of a more experienced developer here.

z3ntu, 2021-01-18:
This sounds like what's being fixed by #1363, could you try that if that changes anything?

christophehenry, 2021-01-19:
No. And I feel the problem is far worse on the Ornata V2. Here are the key when I debug:

- media buttons (play/pause, next, previous, volume up/down, mute): 2,
- fn button alone: 4 (enters the code block),
- fn + F11/F12: 5.

I may be wrong but this is totally new, right?

FIvanO, 2021-01-19:
By the way, fn + F11/F12 (backlight intensity), fn + F10 (game mode) doesn't work either

zekromisblack, 2021-01-20:
I also have been having the same issue but with the razer cynosa v2

CarrotNL, 2021-01-29:
I can confirm. Same problem here. (razer ornata v2)

christophehenry, 2021-01-29:
For thos who have the problem in the meantime, there is a wrokaround simply consisting of turning of the daemon (`systemctl stop --user openrazer-daemon.service`) when the system is up. You can bring it back up if you need to customise something using e.g. Polychromatic but, for now, using the daemon breaks the media keys so it's useless beyond keyboard color init on boot.

johnafogarty4, 2021-03-30:
Stopping the daemon seems to make my keyboard pretty angry in general.  It could just be a symptom of arch itself that is struggling.

In either case, hopefully we can get a fix soon, because it's pretty annoying to not have audio control.

christophehenry, 2021-03-31:
@johnafogarty4 Do you also have Polychromatic installed? It tries to restart the daemon as soon as it detects the daemon is stopped. So you should stop it too. I forgot to mention it in my explaination, I will add it.

johnafogarty4, 2021-03-31:
@christophehenry - I'm using razer genie, but I will check on closing it to see if that helps.

I 100% experienced the newlines until Ctrl + c.

Thanks.

AtlasC0R3, 2021-06-08:
Can confirm this is still happening over here too: Parrot OS (or debian), openrazer-daemon 3.0.1, Polychromatic 0.6.0, edge builds. Doesn't seem to be client-applet-controller-thing related.
On a Razer Ornata V2, previous, play/pause and skip input "bcd" in a text editor. Fn+Previous outputs fhjkl, Fn+Play/Pause outputs hjkl and Fn+Skip outputs efhjkl. And of course, volume scroll acts like a mouse scroll wheel.

Sometimes when launching openrazer, everything works just fine, other times it doesn't work and does these bugs. I dunno. I've also had this issue where pressing previous or skip would act like it's double pressing the key being pressed, when openrazer isn't launched.

wojciechkaleb, 2021-07-21:
I have similar issue with my setup. RazerGenie on Manjaro. I added workaround mentioned by @christophehenry (stoping `systemctl`) However, my keyboard getting wild :| Sometimes it lags and doesn't type character, then it lag and keeps typing next tttttttttttttttttttttttttttttttttttter (that's funny, I didn't want to post a example, but my keyboard wanted to express itself). Sometimes it works ok.

anoinoz, 2021-08-18:
thanks @christophehenry . 
"systemctl stop --user openrazer-daemon.service" gave me back my multimedia keys.

SrS2225a, 2021-10-17:
Is anyone still having this issue? I am having the same thing happen to me. Just did not know if there is some kind of patch now to fix this, since this github issue been open close to a year now. It is a bit annoying having to start/stop the service whenever I want to change something to the keyboard and then use the normal media keys.

On a sidenote, I also noticed this issue seems to affect the DeathAdder somewhat

Bart97, 2021-10-18:
Same here. The scroll wheel now sends the correct volume up/down commands, but pressing it will output "bcd". The same happens with other media keys.

SamantazFox, 2021-11-04:
I have an "Ornata v2" (apparently name changed in v3.1.0) and have the same problem, as posted here: https://github.com/openrazer/openrazer/issues/1552#issuecomment-961211873.

It seems that setting the keyboard back to "device mode" during init fixes the problem:
```diff
--- a/daemon/openrazer_daemon/hardware/keyboards.py
+++ b/daemon/openrazer_daemon/hardware/keyboards.py
@@ -652,6 +652,11 @@
                'set_ripple_effect', 'set_ripple_effect_random_colour']
     DEVICE_IMAGE = "https://assets.razerzone.com/eeimages/support/products/1672/ornata-v2.png"
 
+    def __init__(self, *args, **kwargs):
+        super().__init__(*args, **kwargs)
+
+        self.logger.info('Set back to device mode (Fix for #1398 and/or #1552)')
+        self.set_device_mode(0x00, 0x00)  # Device mode

 class RazerHuntsmanElite(_RippleKeyboard):
     """
```

SrS2225a, 2021-11-04:
> I have an "Ornata v2" (apparently name changed in v3.1.0) and have the same problem, as posted here: [#1552 (comment)](https://github.com/openrazer/openrazer/issues/1552#issuecomment-961211873).
> 
> It seems that setting the keyboard back to "device mode" during init fixes the problem:
> 
> ```diff
> --- a/daemon/openrazer_daemon/hardware/keyboards.py
> +++ b/daemon/openrazer_daemon/hardware/keyboards.py
> @@ -652,6 +652,11 @@
>                 'set_ripple_effect', 'set_ripple_effect_random_colour']
>      DEVICE_IMAGE = "https://assets.razerzone.com/eeimages/support/products/1672/ornata-v2.png"
>  
> +    def __init__(self, *args, **kwargs):
> +        super().__init__(*args, **kwargs)
> +
> +        self.logger.info('Set back to device mode (Fix for #1398 and/or #1552)')
> +        self.set_device_mode(0x00, 0x00)  # Device mode
> 
>  class RazerHuntsmanElite(_RippleKeyboard):
>      """
> ```

I actually have the Ornata v2 as well. Thank you, I will make those changes and recompile it and see if that helps.

On a side note, the numpad "5" key works as a play/pause button for some reason and does not enter "5" as it should, I have to use the keys on the top, and I am not entirely sure what could be causing it. Luckily all other keys work as you would expect. Maybe the recompile will help with that as well? Ill try it.

SamantazFox, 2021-11-09:
> I actually have the Ornata v2 as well. Thank you, I will make those changes and recompile it and see if that helps.

You don't need to recompile anything, btw. I've only modified the python daemon. You can directly edit `/usr/lib/python3.9/site-packages/openrazer_daemon/hardware/keyboards.py` (adapt path depending on your distro) with super-user privileges :)

AtlasC0R3, 2021-11-23:
Thanks @SamantazFox, works perfectly on my end!

t3nk3y, 2021-12-19:
That patch worked fine for me as well on the Ornata v2.  If I turn the patch in to a PR, is it likely to get accepted in it's current form?  This feels like it might be a bit hacky, but I don't know the code well enough to say if it's a good idea to commit it to the master or not.

DidoGamingYT, 2022-01-30:
> You don't need to recompile anything, btw. I've only modified the python daemon. You can directly edit `/usr/lib/python3.9/site-packages/openrazer_daemon/hardware/keyboards.py` (adapt path depending [...]

Editing the file like suggested thorws me an error 
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/daemonize.py", line 245, in start
    self.action(*privileged_action_result)
  File "/usr/bin/openrazer-daemon", line 140, in run_daemon
    daemon = RazerDaemon(verbose=args.verbose,
  File "/usr/lib/python3/dist-packages/openrazer_daemon/daemon.py", line 109, in __init__
    self._device_classes = openrazer_daemon.hardware.get_device_classes()
  File "/usr/lib/python3/dist-packages/openrazer_daemon/hardware/__init__.py", line 24, in get_device_classes
    imported_module = __import__(hw_module, globals=globals(), locals=locals(), fromlist=['*'], level=0)
  File "/usr/lib/python3/dist-packages/openrazer_daemon/hardware/keyboards.py", line 693
    super().__init__(*args, **kwargs)
    ^
IndentationError: expected an indented block

Stopping daemon.

Am I doing something wrong?

AtlasC0R3, 2022-01-30:
> Editing the file like suggested thorws me an error 
> ```Traceback (most recent call last):
> File "/usr/lib/python3/dist-packages/openrazer_daemon/hardware/keyboards.py", line 693
> super().**init**(*args, **kwargs)
> ^ IndentationError: expected an indented block
> ```
> 
> Stopping daemon.
> 
> Am I doing something wrong?

Python is very picky about how you indent code; it seems like you haven't put any spaces before that line of code.
8 spaces *should* be enough to keep Python happy, like so:
[image]

DidoGamingYT, 2022-01-30:
That works, thanks man, glad to have my media buttons with polychromatic running.

ItsRogueRen, 2022-04-06:
Wait, so I'm confused. There IS a solution to this? Do I copy/paste that code to line 693 and I'm done? Or do I have tp uninstall everything and recompile the daemon?

notdido, 2022-04-06:
Yes, you dont need to recompile the daemon

muesli, 2022-04-09:
Can confirm this also happens on the BlackWidow v3 Pro. Workaround applies for this device, too.

gardockt, 2022-04-09:
With today's v3.3.0 release the bug seems to be fixed.

z3ntu, 2022-04-09:
@gardockt You mean for the Ornata (Chroma) V2?

z3ntu, 2022-04-09:
Probably fixed by #1655 then, closing.

For people experiencing this issue on other devices, please search if for that device an issue is already open and if not, create a new one!

closed 2022-04-09 by z3ntu

ItsRogueRen, 2022-04-12:
Still having this issue on the new 3.3.0 version on the Ornata Chroma v2, and I also no longer have the files that was listed earlier to edit? I've completely reinstalled OpenRazer and it installed v3.3.0 so the fix should be there but my media keys still give the 3 letters in everything. 

OS: Pop!_OS 21.10
OpenRazer v3.3.0

xamindar, 2022-04-26:
This is still a problem on version 3.3.0 for the Blackwidow V3 pro.

I still had to manually add the two self. lines to the "Class for the Razer BlackWidow V3 Pro Wired" for this to resolve it.

z3ntu, 2023-11-22:
@ntorresalberto Please open a new issue for that device, and don't comment on year old issues that cover a different device. But please link to this one in the new issue you'll open.