openrazer/openrazer #1400: Razer Tartarus Chroma not found

forsternet, 2021-01-05:
i have tried everything but cant get my Razer Tartarus Chroma to be found by any razer software on Linux mint mate

lah7, 2021-01-06:
Please be more specific - [what have you tried?](https://github.com/openrazer/openrazer/wiki/Troubleshooting) Presumably, none of the frontends show any device, correct?

Let's start with the daemon, what are the last lines?

    tail ~/.local/share/openrazer/logs/razer.log

forsternet, 2021-01-06:
2020-12-03 10:37:19 | razer                          | INFO     | Serving DBus
2020-12-03 10:37:19 | razer                          | DEBUG    | DBus called get_serial_list
2020-12-03 10:37:19 | razer.device0                  | DEBUG    | DBus call get_device_type
2020-12-03 10:37:19 | razer.device0                  | DEBUG    | DBus call get_device_name
2020-12-03 10:37:19 | razer.device0                  | DEBUG    | DBus call get_device_type
2020-12-03 10:37:19 | razer.device0                  | DEBUG    | DBus call get_firmware
2020-12-03 10:37:19 | razer.device0                  | DEBUG    | DBus call version
2020-12-03 10:37:19 | razer.device0                  | DEBUG    | DBus call has_matrix
2020-12-03 10:37:19 | razer.device0                  | DEBUG    | DBus call has_matrix
2020-12-03 10:37:20 | razer.device0                  | DEBUG    | DBus call get_brightness

lah7, 2021-01-07:
From this output, it looks like the device is recognised by the OpenRazer daemon.

Please take a screenshot of the error?

forsternet, 2021-01-07:
its just doesnt matter which razer software i try they just say no device
found dnt know what to show you really

lah7, 2021-01-07:
So if you use [Polychromatic](https://polychromatic.app) (for example), it says "No Device Found" and not "Device Not Recognised"...?

Let's just check the groups on your user account:

    groups

If you don't see `plugdev` in the list, run this command and reboot:

    sudo gpasswd -a $USER plugdev

forsternet, 2021-01-07:
yeah sorry is saying device not recognised trying that reboot just now just need to save

forsternet, 2021-01-07:
ok that got me light control thanks but how about the but mapping

lah7, 2021-01-07:
OpenRazer doesn't _properly_ support key remapping right now (except M1-M5 keys on keyboards). 

You can try #1124 or an application like [key-mapper](https://github.com/sezanzeb/key-mapper) or [Keyboarding Master](https://sites.google.com/site/keyboardingmaster/).

closed 2021-01-07 by lah7

forsternet, 2021-01-07:
well got to say thanks bud got keyboarding master working