openrazer/openrazer #1401: OpenRazer can't find my devices

KeremArslan1, 2021-01-06:
I have Razer deathadder chroma and Razer blackwidow Chroma v2. Openrazer supports both of them. But the polychromatic can't recognise my devices.
Daemon log:
2021-01-06 19:42:50 | razer                          | INFO     | Initialising Daemon (v2.9.0). Pid: 121109
2021-01-06 19:42:50 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2021-01-06 19:42:50 | razer                          | INFO     | Serving DBus
2021-01-06 19:42:54 | razer                          | INFO     | Stopping daemon.
2021-01-06 19:42:56 | razer                          | INFO     | Initialising Daemon (v2.9.0). Pid: 121180
2021-01-06 19:42:56 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2021-01-06 19:42:57 | razer                          | INFO     | Serving DBus

lah7, 2021-01-06:
What distro are you using?

Please refer to the [troubleshooting guide](https://github.com/openrazer/openrazer/wiki/Troubleshooting) - what's the output of these commands (if any)?

    sudo modprobe razerkbd
    sudo modprobe razermouse

KeremArslan1, 2021-01-06:
Uhh nevermind I just rebooted the linux and it worked again magically. (I rebooted like 3-4 times)

closed 2021-01-06 by KeremArslan1