openrazer/openrazer #1402: Razer Deathadder v2 pro

larrryfisherthefisherman, 2021-01-06:
### Distribution

Arch

### Support

- [ ] Have you tried rebooting? Yes
- [ ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page? Yes

I installed Openrazer with openrazer-meta-git package, but it doesn't seem to have support for my device... I thought the git version would be the most up to date?

I need this: https://github.com/openrazer/openrazer/pull/1375/commits/8e78ea192f49761bd7048c82257024092f3fa24f

How can I add that myself or install the right version that has it?

z3ntu, 2021-01-06:
Can you run `pacman -Q openrazer-meta-git` and send the output?

larrryfisherthefisherman, 2021-01-06:
openrazer-meta-git 2.9.0.r14.g7c07ba0f-1

z3ntu, 2021-01-06:
Assuming openrazer-driver-dkms and openrazer-daemon are also on that version, I suspect you don't have your kernel headers installed. Please run 
`sudo dkms install openrazer-driver/2.9.0` and check if that works

larrryfisherthefisherman, 2021-01-06:
   ~  sudo dkms install openrazer-driver/2.9.0                                                                                              ✔  at 11:33:42 AM  
[sudo] password for larry: 
Module openrazer-driver/2.9.0 already installed on kernel 5.10.4-zen2-1-zen/x86_64
    ~ 

z3ntu, 2021-01-06:
Can you check the versions of the other packages? `pacman -Qs openrazer` should display those

larrryfisherthefisherman, 2021-01-06:
local/openrazer-daemon 2.9.0-1
    Userspace daemon that abstracts access to the kernel driver. Provides a DBus service for applications to use.
local/openrazer-driver-dkms 2.9.0-1
    Kernel driver for Razer devices (DKMS-variant)
local/openrazer-meta-git 2.9.0.r14.g7c07ba0f-1
    Meta package for installing all required openrazer packages.
local/polychromatic-git 0.6.0.r5.gf0e1e07-1
    Open source RGB lighting management front-end application for OpenRazer
local/python-openrazer 2.9.0-1
    Python library for accessing the Razer daemon from Python.

z3ntu, 2021-01-06:
Well you also need to install the -git versions of the rest of the openrazer packages, you shouldn't mix&match them.
openrazer-daemon-git & openrazer-driver-dkms-git & python-openrazer-git

larrryfisherthefisherman, 2021-01-06:
Well I didn't do anything but install openrazer-daemon-git, which said that these were dependencies and installed them it self.

z3ntu, 2021-01-06:
Then your AUR helper has something wrong. Just install the packages I said above (instead of the stable ones) and you should be good

larrryfisherthefisherman, 2021-01-06:
Thank you for your time. This is now solved.

closed 2021-01-06 by z3ntu