openrazer/openrazer #1404: Blackwidow X Chroma backlight goes partially off

sn0098, 2021-01-13:
Hi,
I'm using openrazer with razercommander. When I put it in static lighting mode, after a while half of my device's lighting goes off.
I haven't discovered a pattern for this. It just randomly goes off.
I've tried polychromatic too. had the same issue there.
Is this some kind of known bug or is something wrong with my installation?

Thanks a lot

lah7, 2021-01-13:
Hi, If some of the keyboard's lighting area turns off, it could be signs of hardware failure. Or perhaps a power issue if the brightness is above 75%. Is there a specific area of the keyboard that goes off?

The only thing that could turn **all** the lights off is the "devices off on screensaver" feature. You could try turning this off by editing `~/.config/openrazer/razer.conf` and changing the corresponding line to:

    devices_off_on_screensaver = False

Then, reload the daemon:

    openrazer-daemon -r

sn0098, 2021-01-13:
I tried setting that flag to False but the failure happened again. 
My brightness is almost always on max. There is a similar pattern of the leds that go off. They're always is same columns.
Sometimes some columns of one side of the device go off while some other times some columns on both sides go off and the center is okay or vice versa.
[image]

lah7, 2021-01-13:
Unfortunately, it does look like a hardware failure, similar to this:

* https://www.reddit.com/r/razer/comments/87cgcu/razer_blackwidow_v2_half_the_keyboards_leds_are/

If it's not under warranty, you could be out of luck. 😞  Alternately, there could be a firmware update if you plug it into a Windows PC / VM running Synapse 3.

---

Does it happen to other effects, like wave? What about at 50% brightness?

The difference with OpenRazer and Synapse (for Windows) is that this project utilities hardware effects built into the firmware whereas Synapse persistently sends frames over USB.

You could try a software-driven static effect with this script to rule out faulty firmware effects:

```
#!/usr/bin/python3
import time
import openrazer.client

device = openrazer.client.DeviceManager().devices[0]
for r in range(0, device.fx.advanced.rows):
    for c in range(0, device.fx.advanced.cols):
        device.fx.advanced.matrix[r,c] = [0,255,0]

while True:
    device.fx.advanced.draw()
    time.sleep(1)
```

sn0098, 2021-01-14:
Well I saw the post that you linked. He seems to have lost some of his leds permanently. Unlike him, mine work after each reboot. So I'm hopeful that it is not a hardware failure. 
This always happen when I'm in static mode. I will try the wave and other modes on different brightness levels.

Thank you for your time pal :)

closed 2021-06-20 by github-actions[bot]