openrazer/openrazer #1405: Manjaro can't find device

googleboy360, 2021-01-13:
im using cynosa razer keyboard and everytime i start it it says it can't find any devices 

I also did the Add to pluggroup command and restarted and got the daemon working but it now it wont let me use it cause it can't find any devices any help would be super awesome! <3

lah7, 2021-01-14:
We'll need some details - which kernel version?

    uname -r  

Have you looked at the [troubleshooting guide](https://github.com/openrazer/openrazer/wiki/Troubleshooting)? 

Does the daemon list the device?

    openrazer-daemon -F

Is there any error output when you run `sudo modprobe razerkbd`?

googleboy360, 2021-01-14:
nvm i guess i had o restart like 5 times until it just worked randomly

closed 2021-01-14 by googleboy360