openrazer/openrazer #1406: Razer Blade Stealth 13” Intel 7500U (2017) - Fedora - Freezed under some conditions

ctrl-alt-d, 2021-01-15:
### Distribution

* Razer Blade Stealth 13” Intel 7500U (2017) Product #: RZ09-01963G32
* Fedora 33 (Workstation Edition) with Fedora 33 (Workstation Edition)

```
dani@localhost ~ $ ls -lah /sys/bus/hid/drivers
total 0
drwxr-xr-x. 8 root root 0 15 gen.  2021 .
drwxr-xr-x. 4 root root 0 15 gen.  2021 ..
drwxr-xr-x. 2 root root 0 15 gen.  2021 hid-generic
drwxr-xr-x. 2 root root 0 15 gen.  2021 hid-multitouch
drwxr-xr-x. 2 root root 0 15 gen.  2021 magicmouse
drwxr-xr-x. 2 root root 0 15 gen.  2021 ntrig
drwxr-xr-x. 2 root root 0 15 gen. 08:57 razerkbd
drwxr-xr-x. 2 root root 0 15 gen. 08:57 razermouse

├── razerkbd
│   ├── 0003:1532:022D.0004 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:1532:022D.0004
│   ├── 0003:1532:022D.0006 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.1/0003:1532:022D.0006
│   ├── 0003:1532:022D.0007 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.2/0003:1532:022D.0007
│   ├── bind
│   ├── module -> ../../../../module/razerkbd
│   ├── new_id
│   ├── uevent
│   └── unbind
└── razermouse
    ├── 0003:1532:0084.0001 -> ../../../../devices/pci0000:00/0000:00:1c.4/0000:02:00.0/0000:03:02.0/0000:3a:00.0/usb3/3-1/3-1.1/3-1.1:1.0/0003:1532:0084.0001
    ├── 0003:1532:0084.0002 -> ../../../../devices/pci0000:00/0000:00:1c.4/0000:02:00.0/0000:03:02.0/0000:3a:00.0/usb3/3-1/3-1.1/3-1.1:1.1/0003:1532:0084.0002
    ├── 0003:1532:0084.0003 -> ../../../../devices/pci0000:00/0000:00:1c.4/0000:02:00.0/0000:03:02.0/0000:3a:00.0/usb3/3-1/3-1.1/3-1.1:1.2/0003:1532:0084.0003
    ├── 0003:1532:0084.0005 -> ../../../../devices/pci0000:00/0000:00:1c.4/0000:02:00.0/0000:03:02.0/0000:3a:00.0/usb3/3-1/3-1.1/3-1.1:1.3/0003:1532:0084.0005
    ├── bind
    ├── module -> ../../../../module/razermouse
    ├── new_id
    ├── uevent
    └── unbind
```
* I did this (and reboot):
```
dnf config-manager --add-repo https://download.opensuse.org/repositories/hardware:razer/Fedora_33/hardware:razer.repo
dnf install openrazer-meta
```

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

* When I shoutdown the laptop it freezes(*) about 90% of times.
* When I press Caps Lock key it freezes(*) about 5% of times (I use an external USB keyboard)

_(*) The screen becomes from a lot of random colors and the speakers emit random noises._

I miss something? Thanks people.

lah7, 2021-01-15:
Do the freezes still happen if you uninstall the driver? Did the freezes start as soon as you installed OpenRazer or recently updated the Linux kernel?

OpenRazer mainly interfaces with lighting effects and some hardware features. Unless this driver specifically causes the freezes, it could be a kernel or BIOS problem.

Which kernel version are you running?

   uname -r

ctrl-alt-d, 2021-01-15:
Hi Luke,

>Do the freezes still happen if you uninstall the driver? Did the freezes start as soon as you installed OpenRazer or recently updated the Linux kernel?

Since I reinstalled Fedora two moths ago (I sent the laptop to support because Swallen Battery and laptop comes back with Windows again) Before this I only have issues if driver was not present.

I remember, on first installation (three years ago) I changed init grub boot options and I added some flags (I don't remember exact options). This time I didn't do it (maybe this is the problem?)

>OpenRazer mainly interfaces with lighting effects and some hardware features. Unless this driver specifically causes the freezes, it could be a kernel or BIOS problem.

When I installed for first time (3 years ago) without the drivers freezes 100% of times I press Caps Lock.

>Which kernel version are you running?

Version: `5.9.16-200.fc33.x86_64`

```
dani@localhost ~ $ neofetch 
          /:-------------:\          dani@fedora 
       :-------------------::        ----------- 
     :-----------/shhOHbmp---:\      OS: Fedora 33 (Workstation Edition) x86_64 
   /-----------omMMMNNNMMD  ---:     Host: Blade Stealth 3.06 
  :-----------sMMMMNMNMP.    ---:    Kernel: 5.9.16-200.fc33.x86_64 
 :-----------:MMMdP-------    ---\   Uptime: 2 mins 
,------------:MMMd--------    ---:   Packages: 2513 (rpm), 9 (flatpak) 
:------------:MMMd-------    .---:   Shell: bash 5.0.17 
:----    oNMMMMMMMMMNho     .----:   Resolution: 3200x1800, 3840x2160 
:--     .+shhhMMMmhhy++   .------/   DE: GNOME 3.38.2 
:-    -------:MMMd--------------:    WM: Mutter 
:-   --------/MMMd-------------;     WM Theme: Adwaita 
:-    ------/hMMMy------------:      Theme: Adwaita [GTK2/3] 
:-- :dMNdhhdNMMNo------------;       Icons: Adwaita [GTK2/3] 
:---:sdNMMMMNds:------------:        Terminal: gnome-terminal 
:------:://:-------------::          CPU: Intel i7-7500U (4) @ 3.500GHz 
:---------------------://            GPU: Intel HD Graphics 620 
                                     Memory: 1722MiB / 15899MiB 
```

Thanks!

closed 2021-06-20 by github-actions[bot]