openrazer/openrazer #1408: Fedora 33, Openrazer claims I am not in plugdev, even though I am

jbaig77, 2021-01-22:
I'm on Fedora 33 64 bit, with a razer blackwidow chroma v2. 

This is similar to issue #761, but the solution there did not work for me. I do not have secure boot enabled, when i run 
`sudo dkms install openrazer-driver/2.9.0`,
 the messeage I get is:
`Module openrazer-driver/2.9.0 already installed on kernel 5.8.15-301.fc33.x86_64/x86_64`

When I run `sudo modprobe razerkbd`, I get no errors at all. I have added myself in the plugdev group several times using `sudo gpasswd -a $USER plugdev` and then rebooting my system, but openrazer still complains I am not in the plugdev group. Due to the two commands I mentioned before getting no errors, I believe that I installed the kernel headers correctly (I double checked I installed `kernel-devel`, and not `kernel-debug-devel`). I've included my `razer.log` [here](https://pastebin.com/sZgEG51U)

Thank you!

lah7, 2021-01-22:
From the logs, it looks like your user account is in the `plugdev` group. Check here to be certain:

    groups

Instead, it looks like the device is being bound with the wrong permissions at boot. I'm unfamiliar with Fedora but it could be a udev issue/bug. Try unplugging and replugging the hardware once you have logged in, and see if the permission of the driver files change to `plugdev` (not `root`):

    ls -l /sys/devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:06:00.1/usb1/1-1/1-1:1.2/0003:1532:0221.0004/

You can keep the daemon running in the foreground to check:

    openrazer-daemon -s
    openrazer-daemon -Fv

z3ntu, 2021-01-22:
Also you must look at the end of the file for the last daemon launch, this doesn't say that your user isn't in plugdev:
```
2021-01-22 01:36:37 | razer                          | INFO     | Initialising Daemon (v2.9.0). Pid: 25268
2021-01-22 01:36:37 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2021-01-22 01:36:37 | razer                          | INFO     | Found device.0: 0003:1532:0221.0004
2021-01-22 01:36:37 | razer                          | CRITICAL | Could not access /sys/devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:06:00.1/usb1/1-1/1-1:1.2/0003:1532:0221.0004/device_type, file is not owned by plugdev
2021-01-22 01:36:37 | razer                          | INFO     | Serving DBus
```
(and see @lah7 's answer)

jbaig77, 2021-01-23:
Plugging the keyboard back in again seemed to do the trick, however the daemon does not start up properly unless the keyboard is replugged in every time I reboot the computer. I do dual boot with windows 10 on this machine, and I have razer synapse installed on there. My guess is that synapse changes the permissons on the keyboard which interferes with OpenRazer's ability to read the keyboard.

nightsky30, 2021-01-24:
This sounds like #877

z3ntu, 2021-01-31:
Yeah closing as duplicate of #877

closed 2021-01-31 by z3ntu