openrazer/openrazer #1409: Support for [Razer Naga Epic]

SeanHubs, 2021-01-23:
### Device Info

* Name: Razer Naga Epic
* Product Number: RZ01-00510100
* Model Number: RC30-005101

### Device Information
```
Please post the output of
    lsusb -d 1532: -v
```
Bus 001 Device 002: ID 1532:001f Razer USA, Ltd Naga Epic (Wired)
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x001f Naga Epic (Wired)
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x003b
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      93
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      47
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0009  1x 9 bytes
        bInterval               1

```
Please post the output of
    ls -lah /dev/input/by-id/
```
total 0
drwxr-xr-x 2 root root 140 Jan 23 02:56 .
drwxr-xr-x 4 root root 600 Jan 23 02:56 ..
lrwxrwxrwx 1 root root   9 Jan 23 02:56 usb-Razer_Razer_Naga_Epic-event-mouse -> ../event2
lrwxrwxrwx 1 root root   9 Jan 23 02:56 usb-Razer_Razer_Naga_Epic-if01-event-kbd -> ../event3
lrwxrwxrwx 1 root root   9 Jan 23 02:56 usb-Razer_Razer_Naga_Epic-mouse -> ../mouse0
lrwxrwxrwx 1 root root   9 Jan 23 02:56 usb-SONiX_USB_DEVICE-event-if01 -> ../event7
lrwxrwxrwx 1 root root   9 Jan 23 02:56 usb-SONiX_USB_DEVICE-event-kbd -> ../event4
```
### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on Razer Synapse>
Link to the PDF on this mouse:
https://dl.razerzone.com/master-guides/Naga%20Epic%20OMG/NagaEpicOMG-ENG.pdf
Contains every image it needs

https://imgur.com/a/aY7mk4g
https://imgur.com/a/2zfiNkI

z3ntu, 2021-01-23:
Duplicate of #197 . You can test #1289 (see https://github.com/openrazer/openrazer/wiki/Building-a-package ) and test if that works. If yes this can be integrated into the main branch.

closed 2021-01-23 by z3ntu

SeanHubs, 2021-01-23:
Hey,

Yeah this worked fine, I am able to use open razer with Polychromatic GUI now.
Thanks!