openrazer/openrazer #1412: Cant get Ornata V2 to work

CarrotNL, 2021-01-26:
I just got a brand new Ornata V2 and tried to install it in Manjaro.
I tried to install it using the AUR openrazer-meta-git package.

After that I installed Polychromatic and OpenRGB.

My user is in the plugdev group.
I installed the kernel-headers for the running kernel, and modprobe razerkbd gives me no errors.

The Troubleshoot button in Polychromatic gives me nothing but passes, except for the last: that the device maybe is not supported, but my device is in the supported devices.

I don't know what to do else here. Is this a bug, or am I doing something wrong here? It should be supported, am I right?

CarrotNL, 2021-01-26:
BTW: In Polychromatic the usb-identifier shows up in the device-list on the left, with a question mark icon. So it recognizes it as a razer device, but doesn't seem to know which one it is.

z3ntu, 2021-01-26:
Can you send the output of `lsusb | grep 1532` ?

CarrotNL, 2021-01-26:
Output of `lsusb | greb 1532` is:
`Bus 001 Device 004: ID 1532:025d Razer USA, Ltd RZ03-0338, Gaming Keyboard [Ornata V2]`

lah7, 2021-01-26:
[Your device isn't in the stable packages yet](https://github.com/openrazer/openrazer/compare/stable...master) (currently 2.9.0). From the AUR, swap the `openrazer` packages with the `-git` equivalents and reboot - just until the next release.

CarrotNL, 2021-01-26:
SOLVED - Thank you very much!
I installed the openrazer-meta-git package, thinking it's dependencies would be git-packages too. I now installed the other dependencies as git-versions too, and now it works.

Thanks! :)

closed 2021-01-26 by lah7

MislavReversingLabs, 2021-02-13:
Hi. I'm using Ornata V2 Chroma on Mint 20.1 and I installed the current stable version of openrazer.
If Ornata V2 is not yet in the stable pacakages, can I install it on Mint from a dev build somehow until the next stable release? Thanks.