openrazer/openrazer #1413: Unrecognised Device - Chroma Cynosa

geeflaxx, 2021-01-26:
- [ yeah] Have you tried rebooting?
- [yeah ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

i want to start polychromatic-git, but i get the error that my device is unrecognised.
my chroma cynosa keyboard is in the list of compatibilty.

i use to install yay -S polychromatic-git

z3ntu, 2021-01-26:
Can you check the daemon logs? `journalctl --user-unit openrazer-daemon.service` and `sudo journalctl -b | grep -i razer_mount` ?

geeflaxx, 2021-01-26:
[image]
[image]

z3ntu, 2021-01-26:
`openrazer-daemon --version` ?

geeflaxx, 2021-01-26:
patrick@archlinux ~ $ openrazer-daemon --version
openrazer-daemon 2.9.0

z3ntu, 2021-01-26:
Also nothing happens when you unplug and replug the device? At least the command `sudo journalctl -b | grep -i razer_mount` from before should show something.

geeflaxx, 2021-01-26:
[image]

z3ntu, 2021-01-26:
That looks good now. The daemon should now recognize the device - if not please check `openrazer-daemon -Fv`

geeflaxx, 2021-01-26:
[image]

seemed good, i repluged the keyboard and start polychromatic again

z3ntu, 2021-01-26:
So it works now?

geeflaxx, 2021-01-26:
after i restart it works :D
i thank you very much !

closed 2021-01-26 by geeflaxx