openrazer/openrazer #1421: led matrix breaks wave and none effects on mouse

GM-Script-Writer-62850, 2021-01-30:
### Distribution

```
                          ./+o+-       me@X470GamingPlus
                  yyyyy- -yyyyyy+      OS: Ubuntu 20.04 focal
               ://+//////-yyyyyyo      Kernel: x86_64 Linux 5.8.0-38-generic
           .++ .:/++++++/-.+sss/`      Uptime: 6d 1h 50m
         .:++o:  /++++++++/:--:/-      Packages: 2906
        o:+o+:++.`..```.-/oo+++++/     Shell: bash 5.0.17
       .:+o:+o/.          `+sssoo+/    Resolution: 3840x1080
  .++/+:+oo+o:`             /sssooo.   DE: Xfce
 /+++//+:`oo+o               /::--:.   WM: Xfwm4
 \+/+o+++`o++o               ++////.   WM Theme: Greybird-dark-mod
  .++.o+++oo+:`             /dddhhh.   GTK Theme: Greybird-dark-mod [GTK2]
       .+.o+oo:.          `oddhhhh+    Icon Theme: elementary-xfce-darker
        \+.++o+o``-````.:ohdhhhhh+     Font: Noto Sans 9
         `:o+++ `ohhhhhhhhyo++os:      Disk: 876G / 2.0T (47%)
           .o:`.syhhhhhhh/.oo++o`      CPU: AMD Ryzen 5 3600 6-Core @ 12x 3.6GHz
               /osyyyyyyo++ooo+++/     GPU: Radeon RX 580 Series (POLARIS10, DRM 3.38.0, 5.8.0-38-generic, LLVM 11.0.0)
                   ````` +oo+++o\:     RAM: 6844MiB / 16019MiB
                          `oo++.      

```

### Support

- [ ] Have you tried rebooting?
- [ ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

- Hardware: Razer DeathAdder Elite:
- Driver Version: 2.9.0
This device has 2 leds, logo and scroll_wheel
After I set them via the led matrix (a 1x2 array) if I then set logo and or scroll_wheel to the none or wave effect it has no effect and no errors occur, furthermore setting the brightness/pwm duty cycle has no effect, at this point applying any other effect works as expected at which point i can set none or wave and it works

z3ntu, 2021-02-27:
Probably some mode switch back to the hardware effects is needed. Someone with the device will have to try that out.

GM-Script-Writer-62850, 2021-02-27:
I can test a patch

closed 2021-07-28 by github-actions[bot]