openrazer/openrazer #1423: Sub-process /usr/bin/dpkg returned an error code (1)

rusagent, 2021-01-30:
This is my system:

```
pi@raspberrypi:~ $ cat /etc/os-release 
PRETTY_NAME="Debian GNU/Linux 10 (buster)"
NAME="Debian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
pi@raspberrypi:~ $ lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 10 (buster)
Release:	10
Codename:	buster
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 5.10.11-v8+ #1399 SMP PREEMPT Thu Jan 28 12:14:03 GMT 2021 aarch64 GNU/Linux

```

Output of  dpkg -l | grep linux

```

pi@raspberrypi:~ $ dpkg -l | grep linux
ii  binutils-aarch64-linux-gnu           2.31.1-16                           arm64        GNU binary utilities, for aarch64-linux-gnu target
ii  console-setup-linux                  1.193~deb10u1                       all          Linux specific part of console-setup
ii  libselinux1:arm64                    2.8-1+b1                            arm64        SELinux runtime shared libraries
ii  libv4l-0:arm64                       1.16.3-3                            arm64        Collection of video4linux support libraries
ii  libv4lconvert0:arm64                 1.16.3-3                            arm64        Video4linux frame format conversion library
ii  linux-base                           4.6                                 all          Linux image base package
ii  linux-libc-dev:arm64                 4.19.160-2                          arm64        Linux support headers for userspace development
ii  util-linux                           2.33.1-0.1                          arm64        miscellaneous system utilities
ii  v4l-utils                            1.16.3-3                            arm64        Collection of command line video4linux utilities

```

I tried to install the openrazer-meta and i get these errors:
```

pi@raspberrypi:~ $ sudo apt --fix-broken install
Reading package lists... Done
Building dependency tree       
Reading state information... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
4 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up openrazer-driver-dkms (2.4.0+dfsg-1) ...
Removing old openrazer-driver-2.4.0 DKMS files...

------------------------------
Deleting module version: 2.4.0
completely from the DKMS tree.
------------------------------
Done.
Loading new openrazer-driver-2.4.0 DKMS files...
It is likely that 5.10.11-v8+ belongs to a chroot's host
Building for 5.10.11+, 5.10.11-v7+, 5.10.11-v7l+, 5.10.11-v8+ and 5.10.5-v8+
Module build for kernel 5.10.11+ was skipped since the
kernel headers for this kernel does not seem to be installed.
Module build for kernel 5.10.11-v7+ was skipped since the
kernel headers for this kernel does not seem to be installed.
Module build for kernel 5.10.11-v7l+ was skipped since the
kernel headers for this kernel does not seem to be installed.
Module build for kernel 5.10.11-v8+ was skipped since the
kernel headers for this kernel does not seem to be installed.
Building initial module for 5.10.5-v8+
Error! Bad return status for module build on kernel: 5.10.5-v8+ (aarch64)
Consult /var/lib/dkms/openrazer-driver/2.4.0/build/make.log for more information.
dpkg: error processing package openrazer-driver-dkms (--configure):
 installed openrazer-driver-dkms package post-installation script subprocess returned error exit status 10
dpkg: dependency problems prevent configuration of openrazer-meta:
 openrazer-meta depends on openrazer-driver-dkms (= 2.4.0+dfsg-1); however:
  Package openrazer-driver-dkms is not configured yet.

dpkg: error processing package openrazer-meta (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of openrazer-daemon:
 openrazer-daemon depends on openrazer-driver-dkms (= 2.4.0+dfsg-1); however:
  Package openrazer-driver-dkms is not configured yet.

dpkg: error processing package openrazer-daemon (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of python3-openrazer:
 python3-openrazer depends on openrazer-daemon (= 2.4.0+dfsg-1); however:
  Package openrazer-daemon is not configured yet.

dpkg: error processing package python3-openrazer (--configure):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 openrazer-driver-dkms
 openrazer-meta
 openrazer-daemon
 python3-openrazer
E: Sub-process /usr/bin/dpkg returned an error code (1)

``` 

I'm on a Raspberry PI 4b with 64 bit Raspberry OS. Any solutions?

z3ntu, 2021-01-30:
1. Check `/var/lib/dkms/openrazer-driver/2.4.0/build/make.log` as the dkms build has failed
2. We don't suport the version 2.4.0 (released October 2018), the newest is 2.9.0

lah7, 2021-01-31:
> Module build for kernel 5.10.11+ was skipped since the
> kernel headers for this kernel does not seem to be installed.

Be sure to install `raspberrypi-kernel-headers`. See https://github.com/openrazer/openrazer/issues/1336#issuecomment-731101076 for instructions.

rusagent, 2021-01-31:
I followed the instructions and now i get this error.
```

pi@securebox:~ $ sudo modprobe razerkbd
modprobe: FATAL: Module razerkbd not found in directory /lib/modules/5.10.11-v8+
```

lah7, 2021-01-31:
It doesn't look like the DKMS module was installed (or it did, but to the wrong kernel). Was there an error running `apt install`? Try manually:

    sudo dkms install openrazer-driver/2.9.0

Or specifically install for the running kernel:

    sudo dkms install openrazer-driver/2.9.0 -k 5.10.11-v8+

rusagent, 2021-01-31:
I installed it with the guide you posted. The driver can't be loaded with polychromatic or any gui application.

pi@securebox:~ $ sudo dkms install openrazer-driver/2.9.0 -k 5.10.11-v8+
Error! Your kernel headers for kernel 5.10.11-v8+ cannot be found.
Please install the linux-headers-5.10.11-v8+ package,
or use the --kernelsourcedir option to tell DKMS where it's located

rusagent, 2021-01-31:
```
pi@securebox:~ $ sudo apt update
OK:1 http://ppa.launchpad.net/polychromatic/stable/ubuntu bionic InRelease
OK:2 http://download.opensuse.org/repositories/hardware:/razer/Debian_10  InRelease
OK:3 http://deb.debian.org/debian buster InRelease
Holen:4 http://deb.debian.org/debian-security buster/updates InRelease [65,4 kB]
OK:5 http://archive.raspberrypi.org/debian buster InRelease
OK:6 http://deb.debian.org/debian buster-updates InRelease
OK:7 http://packages.microsoft.com/repos/code stable InRelease
OK:8 https://repo.nordvpn.com/deb/nordvpn/debian stable InRelease
Es wurden 65,4 kB in 2 s geholt (26,7 kB/s).
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.
Statusinformationen werden eingelesen.... Fertig
Alle Pakete sind aktuell.
pi@securebox:~ $ sudo apt install raspberrypi-kernel-headers
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.
Statusinformationen werden eingelesen.... Fertig
raspberrypi-kernel-headers ist schon die neueste Version (1.20210111.master-1).
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.
pi@securebox:~ $ sudo apt upgrade raspberrypi-kernel
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.
Statusinformationen werden eingelesen.... Fertig
raspberrypi-kernel ist schon die neueste Version (1.20210111.master-1).
Paketaktualisierung (Upgrade) wird berechnet... Fertig
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.

```

rusagent, 2021-01-31:
```
pi@securebox:~ $ sudo uname --all
Linux securebox 5.10.11-v8+ #1399 SMP PREEMPT Thu Jan 28 12:14:03 GMT 2021 aarch64 GNU/Linux

```

z3ntu, 2021-01-31:
It seems that you've somehow manually installed a not-yet-released version of the kernel (5.10.11-v8+)..? The newest version in the repos is `5.10.5-v8+`

rusagent, 2021-01-31:
```
pi@securebox:~ $ sudo apt install ./openrazer-meta_2.9.0-0_all.deb
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.
Statusinformationen werden eingelesen.... Fertig
Hinweis: »openrazer-meta« wird an Stelle von »./openrazer-meta_2.9.0-0_all.deb« gewählt.
Die folgenden Pakete wurden automatisch installiert und werden nicht mehr benötigt:
  gir1.2-appindicator3-0.1 gir1.2-javascriptcoregtk-4.0 gir1.2-soup-2.4 gir1.2-webkit2-4.0 libappindicator3-1 libdbusmenu-glib4 libdbusmenu-gtk3-4
Verwenden Sie »sudo apt autoremove«, um sie zu entfernen.
Die folgenden zusätzlichen Pakete werden installiert:
  openrazer-daemon openrazer-driver-dkms python3-openrazer
Vorgeschlagene Pakete:
  openrazer-doc
Die folgenden NEUEN Pakete werden installiert:
  openrazer-daemon openrazer-driver-dkms openrazer-meta python3-openrazer
0 aktualisiert, 4 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.
Es müssen noch 74,9 kB von 146 kB an Archiven heruntergeladen werden.
Nach dieser Operation werden 1.125 kB Plattenplatz zusätzlich benutzt.
Möchten Sie fortfahren? [J/n] y
Holen:1 http://download.opensuse.org/repositories/hardware:/razer/Debian_10  openrazer-driver-dkms 2.9.0-0 [50,3 kB]
Holen:2 http://download.opensuse.org/repositories/hardware:/razer/Debian_10  python3-openrazer 2.9.0-0 [24,7 kB]
Es wurden 74,9 kB in 0 s geholt (355 kB/s).
Vormals nicht ausgewähltes Paket openrazer-driver-dkms wird gewählt.
(Lese Datenbank ... 136695 Dateien und Verzeichnisse sind derzeit installiert.)
Vorbereitung zum Entpacken von .../openrazer-driver-dkms_2.9.0-0_all.deb ...
Entpacken von openrazer-driver-dkms (2.9.0-0) ...
Vormals nicht ausgewähltes Paket openrazer-daemon wird gewählt.
Vorbereitung zum Entpacken von .../openrazer-daemon_2.9.0-0_all.deb ...
Entpacken von openrazer-daemon (2.9.0-0) ...
Vormals nicht ausgewähltes Paket python3-openrazer wird gewählt.
Vorbereitung zum Entpacken von .../python3-openrazer_2.9.0-0_all.deb ...
Entpacken von python3-openrazer (2.9.0-0) ...
Vormals nicht ausgewähltes Paket openrazer-meta wird gewählt.
Vorbereitung zum Entpacken von .../openrazer-meta_2.9.0-0_all.deb ...
Entpacken von openrazer-meta (2.9.0-0) ...
openrazer-driver-dkms (2.9.0-0) wird eingerichtet ...
Loading new openrazer-driver-2.9.0 DKMS files...
It is likely that 5.10.11-v8+ belongs to a chroot's host
Building for 5.10.11+, 5.10.11-v7+, 5.10.11-v7l+, 5.10.11-v8+ and 5.10.5-v8+
Module build for kernel 5.10.11+ was skipped since the
kernel headers for this kernel does not seem to be installed.
Module build for kernel 5.10.11-v7+ was skipped since the
kernel headers for this kernel does not seem to be installed.
Module build for kernel 5.10.11-v7l+ was skipped since the
kernel headers for this kernel does not seem to be installed.
Module build for kernel 5.10.11-v8+ was skipped since the
kernel headers for this kernel does not seem to be installed.
Building initial module for 5.10.5-v8+
Done.

razerkbd.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.10.5-v8+/updates/dkms/

razermouse.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.10.5-v8+/updates/dkms/

razerkraken.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.10.5-v8+/updates/dkms/

razeraccessory.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.10.5-v8+/updates/dkms/

depmod...

DKMS: install completed.

ADD YOUR USER TO THE PLUGDEV GROUP: 'sudo gpasswd -a $USER plugdev'
openrazer-daemon (2.9.0-0) wird eingerichtet ...
python3-openrazer (2.9.0-0) wird eingerichtet ...
openrazer-meta (2.9.0-0) wird eingerichtet ...
Trigger für man-db (2.8.5-2) werden verarbeitet ...
pi@securebox:~ $ sudo dkms install openrazer-driver/2.9.0 -k 5.10.11-v8+
Error! Your kernel headers for kernel 5.10.11-v8+ cannot be found.
Please install the linux-headers-5.10.11-v8+ package,
or use the --kernelsourcedir option to tell DKMS where it's located

```

rusagent, 2021-01-31:
I'm using this one: https://downloads.raspberrypi.org/raspios_arm64/images/raspios_arm64-2020-08-24/
But i also did an rpi-update so i could install my wifi card with wifi-install

rusagent, 2021-01-31:
I'll try sudo apt-get install --reinstall raspberrypi-bootloader raspberrypi-kernel meanwhile

z3ntu, 2021-01-31:
Also please `dpkg -l | grep raspberrypi`

rusagent, 2021-01-31:
```
 it's located
pi@securebox:~ $ dpkg -l | grep raspberrypi
ii  libraspberrypi-bin                    2+git20201022~151804+e432bc3-1         arm64        Miscellaneous Raspberry Pi utilities
ii  libraspberrypi-dev                    2+git20201022~151804+e432bc3-1         arm64        EGL/GLES/OpenVG/etc. libraries for the Raspberry Pi's VideoCore IV (headers)
ii  libraspberrypi-doc                    2+git20200813~095940+f73fca0-1         all          EGL/GLES/OpenVG/etc. libraries for the Raspberry Pi's VideoCore IV (headers)
ii  libraspberrypi0                       2+git20201022~151804+e432bc3-1         arm64        EGL/GLES/OpenVG/etc. libraries for the Raspberry Pi's VideoCore IV
ii  raspberrypi-artwork                   20150921                               all          Copy of the Raspberry Pi logo
ii  raspberrypi-bootloader                1.20210111.master-1                    arm64        Raspberry Pi bootloader
ii  raspberrypi-kernel                    1.20210111.master-1                    arm64        Raspberry Pi bootloader
ii  raspberrypi-kernel-headers            1.20210111.master-1                    arm64        Header files for the Raspberry Pi Linux kernel
ii  raspberrypi-net-mods                  1.3.0                                  all          Network configuration for the Raspberry Pi UI
ii  raspberrypi-sys-mods                  20210125                               arm64        System tweaks for the Raspberry Pi
ii  raspberrypi-ui-mods                   1.20201210+nmu1                        arm64        Config to customise the LXDE UI for the Raspberry Pi
```

also reinstalling the kernel didn't help...

z3ntu, 2021-01-31:
You should somehow find out where the `5.10.11-v8+` kernel comes from and how to get back to the `5.10.5-v8+` one

rusagent, 2021-01-31:
It's the one from the masterbranch i guess? https://github.com/raspberrypi/firmware/commit/496e65477e06172ea20602e365d3790632c3cc06

z3ntu, 2021-01-31:
I can't imagine that you got the kernel version from the repos, but I don't know what you did with your rpi

rusagent, 2021-01-31:
I actually just used rpi-update.... 

But now i used sudo rpi-update 5467199574e4a6621ded42359dfd9aadfc4793ee to get back to the supported kernel version. 
Thanks for your help!

closed 2021-01-31 by rusagent

z3ntu, 2021-01-31:
If rpi-update fetches the latest kernel from git, then that's probably what happened. I have no idea what that tool does so :shrug:

lah7, 2021-01-31:
Makes sense; [`rpi-update` isn't recommended to update the kernel](https://www.raspberrypi.org/documentation/linux/kernel/updating.md) unless it's for testing or obtaining the latest kernel features. As long as the kernel and kernel header versions match, DKMS works. :+1: