openrazer/openrazer #1424: Using hasattr() can cause a error

GM-Script-Writer-62850, 2021-01-30:
### Distribution

```
$ screenfetch 
                          ./+o+-       me@X470GamingPlus
                  yyyyy- -yyyyyy+      OS: Ubuntu 20.04 focal
               ://+//////-yyyyyyo      Kernel: x86_64 Linux 5.8.0-41-generic
           .++ .:/++++++/-.+sss/`      Uptime: 6h 47m
         .:++o:  /++++++++/:--:/-      Packages: 2911
        o:+o+:++.`..```.-/oo+++++/     Shell: bash 5.0.17
       .:+o:+o/.          `+sssoo+/    Resolution: 3840x1080
  .++/+:+oo+o:`             /sssooo.   DE: Xfce
 /+++//+:`oo+o               /::--:.   WM: Xfwm4
 \+/+o+++`o++o               ++////.   WM Theme: Greybird-dark-mod
  .++.o+++oo+:`             /dddhhh.   GTK Theme: Greybird-dark-mod [GTK2]
       .+.o+oo:.          `oddhhhh+    Icon Theme: elementary-xfce-darker
        \+.++o+o``-````.:ohdhhhhh+     Font: Noto Sans 9
         `:o+++ `ohhhhhhhhyo++os:      Disk: 877G / 2.0T (47%)
           .o:`.syhhhhhhh/.oo++o`      CPU: AMD Ryzen 5 3600 6-Core @ 12x 3.6GHz
               /osyyyyyyo++ooo+++/     GPU: Radeon RX 580 Series (POLARIS10, DRM 3.38.0, 5.8.0-41-generic, LLVM 11.0.0)
                   ````` +oo+++o\:     RAM: 4448MiB / 16019MiB
                          `oo++. 
```

### Support

When testing if a attribute/function exist using `hasattr()` triggers a error
```
$ Desktop/untitled 
Testing device (Razer DeathAdder Elite):
	error at brightness
	error at dpi_stages
	Testing device fx:
		error at colors
		error at effect
		error at speed
	Testing zones:
		 logo
		 scroll_wheel
		 left
		 right
		 backlight
```
Test Script: [untitled.zip](https://github.com/openrazer/openrazer/files/5898950/untitled.zip)

lah7, 2021-01-31:
I have encountered this before, the workaround is to check the capability (using `device.capabilities`) and only try accessing that attribute if it exists. Exceptions are raised as it's being trying to call a non-existent method on D-Bus and `hasattr` is testing the output.

Here's an example with my **Razer Taipan** - which does not support any Chroma effects:
```
Traceback (most recent call last):
  File "/ramdisk/./untitled", line 145, in <module>
    print(hasattr(attr, x))
  File "/usr/lib/python3.9/site-packages/openrazer/client/fx.py", line 711, in colors
    return bytes(self._getattr('get#EffectColors')())
  File "/usr/lib/python3.9/site-packages/dbus/proxies.py", line 141, in __call__
    return self._connection.call_blocking(self._named_service,
  File "/usr/lib/python3.9/site-packages/dbus/connection.py", line 652, in call_blocking
    reply_message = self.send_message_with_reply_and_block(
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.UnknownMethod: Traceback (most recent call last):
  File "/usr/lib/python3.9/site-packages/dbus/service.py", line 658, in _message_cb
    (candidate_method, parent_method) = _method_lookup(self, method_name, interface_name)
  File "/usr/lib/python3.9/site-packages/dbus/service.py", line 248, in _method_lookup
    raise UnknownMethodException('%s is not a valid method of interface %s' % (method_name, dbus_interface))
dbus.exceptions.UnknownMethodException: org.freedesktop.DBus.Error.UnknownMethod: Unknown method: getLogoEffectColors is not a valid method of interface razer.device.lighting.logo
```

However, when looping through and catching all errors, it is only for the persistence feature (`getLogoEffect, getScrollEffect, getScrollEffectColors, getScrollEffectSpeed, etc`) which is in the `master` branch (added after 2.9.0), so it's possibly a duplicate of #1380.

GM-Script-Writer-62850, 2021-01-31:
looks like it is a dup to me, but it also affects `brightness` and `dpi_stages` which are older features, does the patch there fix that?

GM-Script-Writer-62850, 2021-01-31:
So as for the workaround how do i test if speed, colors, effects, and wave_dir are supported (mainly due to versions)

made a fn to test for rgb / zone / effect support
zones are [ 'logo', 'scroll_wheel', 'left', 'right', 'backlight' ]
where no zone is for device.fx (how do you refer to that as a zone
)
```
# example device = device_manager.devices[0]
# zone is any of these: logo, scroll_wheel, left, right, backlight
# scroll_wheel / scroll / wheel are all treated the same to avoid confusion from inconstant naming
# effect is the effect function/property name
def rgb_support(device,zone=False,effect=False):
    prop=["lighting"]
    if not device.capabilities[prop[0]]:
        return False
    if zone in ['scroll_wheel', 'wheel']:
        prop.append('scroll')
    elif zone and not zone == 'generic':
        prop.append(zone)
    if effect:
        prop.append(effect)
    prop='_'.join(prop)
    if prop in device.capabilities and device.capabilities[prop]:
	    return True
    return False
```

lah7, 2021-01-31:
> So as for the workaround how do i test if speed, colors, effects, and wave_dir are supported (mainly due to versions)

The simplest is wrapping `try:`, `except Exception:` around the lines trying to read `colors`, `effects`, `wave_dir`, etc.

By the way, you can also use the `device.has("")` function to get `True` or `False` when checking a device's capabilities.

GM-Script-Writer-62850, 2021-01-31:
i really do not like `.has` for reasons, I find it useless as seen below
```
   logo settings:
      brightness: 0.0
         p = brightness ; attr=getattr(device.fx.misc, 'logo' ); hasattr(attr,p) ==  True ; attr.has(p) ==  False
      colors: ['DACB6A', '63298D', '0000FF']
         p = colors ; attr=getattr(device.fx.misc, 'logo' ); hasattr(attr,p) ==  True ; attr.has(p) ==  False
      effect: none
         p = effect ; attr=getattr(device.fx.misc, 'logo' ); hasattr(attr,p) ==  True ; attr.has(p) ==  False
      speed: 1
         p = speed ; attr=getattr(device.fx.misc, 'logo' ); hasattr(attr,p) ==  True ; attr.has(p) ==  False
      wave_dir: 1
         p = wave_dir ; attr=getattr(device.fx.misc, 'logo' ); hasattr(attr,p) ==  True ; attr.has(p) ==  False
      effects available: none, brightness, breath_single, breath_dual, breath_random, reactive, spectrum, static
      effects unavailable: breath_triple, blinking, ripple, ripple_random, pulsate, starlight_single, starlight_dual, starlight_random, wave, multicolor
   scroll_wheel settings:
      brightness: 0.0
         p = brightness ; attr=getattr(device.fx.misc, 'scroll_wheel' ); hasattr(attr,p) ==  True ; attr.has(p) ==  False
      colors: ['BD3AA0', '097530', '0000FF']
         p = colors ; attr=getattr(device.fx.misc, 'scroll_wheel' ); hasattr(attr,p) ==  True ; attr.has(p) ==  False
      effect: none
         p = effect ; attr=getattr(device.fx.misc, 'scroll_wheel' ); hasattr(attr,p) ==  True ; attr.has(p) ==  False
      speed: 1
         p = speed ; attr=getattr(device.fx.misc, 'scroll_wheel' ); hasattr(attr,p) ==  True ; attr.has(p) ==  False
      wave_dir: 1
         p = wave_dir ; attr=getattr(device.fx.misc, 'scroll_wheel' ); hasattr(attr,p) ==  True ; attr.has(p) ==  False
      effects available: none, brightness, breath_single, breath_dual, breath_random, reactive, spectrum, static
      effects unavailable: breath_triple, blinking, ripple, ripple_random, pulsate, starlight_single, starlight_dual, starlight_random, wave, multicolor
```

lah7, 2021-06-01:
Closing as a duplicate of  #1380.

closed 2021-06-01 by lah7