openrazer/openrazer #1425: Nommo Chroma image URL no longer resolvable

brett-smith, 2021-02-03:
### Distribution

Linux Mint 20.1

### Support

- [X] Have you tried rebooting?
- [X] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

As title, the image URL for Razer Nommo Chroma https://d4kkpd69xt9l7.cloudfront.net/sys-master/root/he6/h1a/8939927175198/nommo-gallery-1500x1000-11.jpg no longer works. 

This - https://assets.razerzone.com/eeimages/support/products/1275/1275_nommo_chroma.png - looks like it may be a better choice, being on the 'assets' server like most of the other products, it's also a transparent PNG! 

[image]

z3ntu, 2021-02-03:
Thanks for reporting. Could you open a pull request changing this url?

brett-smith, 2021-02-03:
Done. I also fixed Razer Nommo Pro image while I was there, as I noticed it was also now broken in the same way.

z3ntu, 2021-02-27:
Fixed with #1427

closed 2021-02-27 by z3ntu