openrazer/openrazer #1426: Support for Razer Viper 8KHz

robertswiecki, 2021-02-03:
### Device Info

* Name: Viper 8KHz
* Product Number: RZ01-03580100
* Model Number: RZ01-0358

### Device Information
```
lsusb -d 1532: -v

Bus 001 Device 004: ID 1532:0091 Razer USA, Ltd Razer Viper 8KHz
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0091 
  bcdDevice            1.01
  iManufacturer           1 Razer
  iProduct                2 Razer Viper 8KHz
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
```

```
ls -lah /dev/input/by-id/ | grep Raze
lrwxrwxrwx 1 root root  10 02-03 19:28 usb-Razer_Razer_Viper_8KHz-event-if01 -> ../event10
lrwxrwxrwx 1 root root  10 02-03 19:31 usb-Razer_Razer_Viper_8KHz-event-mouse -> ../event11
lrwxrwxrwx 1 root root   9 02-03 19:28 usb-Razer_Razer_Viper_8KHz-if01-event-kbd -> ../event5
lrwxrwxrwx 1 root root  10 02-03 19:11 usb-Razer_Razer_Viper_8KHz-if02-event-kbd -> ../event12
lrwxrwxrwx 1 root root   9 02-03 19:31 usb-Razer_Razer_Viper_8KHz-mouse -> ../mouse0
```

### Packet Captures

This is a 8000Hz mouse. I don't know how Windows (via Synapse) is switching it to 8000Hz. I guess USB2.0 is used (USB1 polling seems to be limited to 1kHZ), but HWINFO.exe did show only USB1 enabled when using 8000Hz.

I saved some profiles on the mouse (with 8kHz polling), but these seem not to work in the absence of Windows software, even if I assigned 'Cycle Up Profile' to one of the buttons, the profiles are not switching (I'd see LED changing color in such case).

I'll be willing to buy this mouse to someone else, if that person promises to do some good effort on trying to reverse engineer the USB1->USB2 switching protocol and implement it here (or via some script, it doesn't need to be in openrazer).

robertswiecki, 2021-02-03:
Actually when connecting directly to my PC, w/o a hub it appears as USB2, with 125us interval. The evhz still shows that it's max 1000Hz, but I guess I can work from here.

I recorded some VM comms via usbmon, attaching it. It shows switching polling rate between 125 and 8000Hz. I hope it'll be useful for posterity, but please let me know if you'd like to see some more detailed usb packet dumps.
[a.usbmon.txt](https://github.com/openrazer/openrazer/files/5921833/a.usbmon.txt)

z3ntu, 2021-02-04:
This mouse also has lighting that can be configured (from what I can tell), so let's keep this open :)

Generally, especially for the polling rate feature, usb packet captures as described in the wiki would be useful, I can then check if Razer does anything special for the polling rate.

robertswiecki, 2021-02-10:
In any case, this might be useful for some. If the mouse is attached directly to PC all works fine, and it's possible to switch between profiles set with Windows Razer Synapse.

But, when using a USB KVM (I use some ATEN ones), the mouse downgrades itself to USB1 which gives max 1kHz polling (1ms USB polling interval).

Resetting the USB device doesn't work (with usbreset), the mouse doesn't seem to react to the USB reset cmd. It's necessary to power-cycle the switch with ```uhubctl```. I use the following udev rule to do that ondemand (only if the mouse attaches itself as USB1 - 12Mbps speed). This power-cycles the USB port on the hub, and the mouse attaches itself as USB2 / 480Mbps device.

```
$ cat /etc/udev/rules.d/92-razer-8khz-reset.rules 
SUBSYSTEM=="usb", ACTION=="add", ATTR{idVendor}=="1532", ATTR{idProduct}=="0091", ATTR{speed}=="12", RUN+="/usr/sbin/uhubctl -a 2 -s 'Razer Razer Viper 8KHz'"
```

Hope this helps

AleMax, 2021-03-16:
Hello!

Recently acquired a Viper 8k,
I also wanted to contribute to this.
I tried to capture the change in polling rate,
Having tried three different ways, 
switching between 125Hz and 500Hz, 1000Hz and 4000Hz and finally 2000Hz and 8000Hz,
I switched from the first number to the second, back, to the second and finally back to the first in every recording.
I recorded the three different ones each twice, as marked by the alt version every one has (just to be sure that there are no differences / noise).
I did not move the mouse, not click any button on it during the recordings,
the RGB Settings were turned completely off,
and, if it has any significance in sending the packets,
The mouse was configured to 5 DPI Settings (as seen in the screenshot). There was only a single profile stored on the mouse.

[image]

[Captures.zip](https://github.com/openrazer/openrazer/files/6145149/Captures.zip)

Hope this can help!

z3ntu, 2021-03-16:
Great captures, this seems to be the command for changing polling rate on this device:

command class 0x00, command 0x40
payload (2 bytes):
byte 1: 0x0 or 0x1 - not sure? all polling rate changes are sent once with 00, once with 01
byte 2:
* 0x40 = 125 polling rate
* 0x10 = 500 polling rate
* 0x08 = 1000 polling rate
* 0x04 = 2000 polling rate
* 0x02 = 4000 polling rate
* 0x01 = 8000 polling rate

and this device uses transaction id 0x1f

AleMax, 2021-03-16:
So i can try adding support for the mouse then,
At least as far as i can gather from recent additions / commits to the project i think i can try to get it running (at least the basic features),
But i have no idea how i would go implement such a specific thing like the polling rate change in a detailed manner,
as in, how to implement this specific command, is there an example i could cheat and copy from?

z3ntu, 2021-03-16:
@AleMax Here's a "template" you can use: https://github.com/openrazer/openrazer/compare/feature_viper8khz
I'm not sure if the new firmware commands have a "get" variant and not just a "set" variant. You can try adding a packet similar to the normal get poll rate command (in razerchromacommon.c) with class 0x00, command 0xC0 and see if the firmware returns anything.
Also frontends (polychromatic, RazerGenie etc) won't display a 8Khz option yet, though the other options (125, 500 & 1000) should work (also check dmesg, error responses from the firmware will be printed there). This will need a new api for giving frontends the info that 2000, 4000 & 8000Hz are available.
DPI could also work, if not try adding a block to the functions in the driver to set the transaction id to 0x1f for your device.
If you have any questions feel free to ask.

AleMax, 2021-03-17:
Okay,
I tried playing around a bit with it,
the mouse definitely gets detected, that appears to work fine
(polychromatic actually crashes when the mouse is detected though, i dont know if thats a fault on theirs or on here, razergenie doesnt)
The mouse got a really noticable input lag as soon as the driver detected it,
With evhz i was able to see that the actual polling rate of the mouse was stuck on 34Hz,

Further playing around with the Python API i then noticed following:
Setting the Poll Rate to 1000 results in a real poll rate of ~34Hz
Setting the Poll Rate to 500 results in a real poll rate of ~30Hz
Setting the Poll Rate to 125 results in a real poll rate of ~63Hz

With that i then dug deeper (I especially checked if thats the case because so far only 1 of the [apparently] 2 needed commands are sent).
I, again, captured the USB packets send from and to the mouse and noticed that the polling rates send to the mouse are wrong.

When i set the rate to 1000 the command sends a e8 (8000 / e8 = 34,48)
When i set the rate to 500 the command sends a f4 (8000 / f4 = 32,79)
When i set the rate to 125 the command sends a 7d (8000 / 7d = 64)

These results correlate with the observations.
That means the command parameter that is getting sent is somehow completely wrong, and i dont have a good enough understanding of the project to be able to debug or notice why that is the case,
I hope the info can help you maybe identify what could be wrong.

Two more things i noticed:

1) Before every command send it also sends a 81 command, which at least as i found it in the code has something to do with the firmware? Im not sure if thats wanted or not, but it definetly doesnt happen in the original Razer Software, so its something i noted and wanted to share.

2) The Mouse is apparently capable of receiving an arbitrary number as an argument for the polling number and then waits (1 / 8000 * x) seconds before polling next, it does not appear to be limited to 125-500-1000-2000-4000-8000, even though the official razer software only has these options.
Im not sure if it would be smart to expose the entire range of polling rates via the driver / API.
On one hand it appears to work completly fine for arbitrary polling rates,
on the other hand, as its not supported in the official software, there is no being for certain that it doesnt break anything,
But in the end, its probably something to think about.

robertswiecki, 2021-03-17:
PS. evhz doesn't detect >1000Hz polling rate. For that you need to do the following

```C
-                time = (unsigned long long)event.time.tv_sec * 1000ULL;
-                time += (unsigned long long)event.time.tv_usec / 1000ULL;
+                time = (unsigned long long)event.time.tv_sec * 1000000ULL;
+                time += (unsigned long long)event.time.tv_usec;
 
                 timediff = time - events[i].prev_time;
 
-                if(timediff != 0) hz = 1000ULL / timediff;
+                if (timediff != 0) hz = 1000000ULL / timediff;
```

```shell
Razer Razer Viper 8KHz: Latest  8000Hz, Average  4937Hz
Razer Razer Viper 8KHz: Latest  3831Hz, Average  4872Hz
Razer Razer Viper 8KHz: Latest  4184Hz, Average  4875Hz
Razer Razer Viper 8KHz: Latest  4000Hz, Average  4875Hz
Razer Razer Viper 8KHz: Latest  4000Hz, Average  4896Hz
Razer Razer Viper 8KHz: Latest  3984Hz, Average  4896Hz
Razer Razer Viper 8KHz: Latest  4016Hz, Average  4834Hz
Razer Razer Viper 8KHz: Latest  3984Hz, Average  4834Hz
Razer Razer Viper 8KHz: Latest  4016Hz, Average  4833Hz
Razer Razer Viper 8KHz: Latest  3968Hz, Average  4854Hz
``

SM9CC, 2021-05-15:
Any updates on this?

kodatarule, 2021-05-22:
I'm interested as well

grandprixgp, 2021-07-10:
Does anyone have any captures that involve LOD (lift off distance) changes? I've observed that the LOD setting is not stored in on-board flash, and is only active as long as Synapse is running, however if you close synapse, the LOD setting will remain until the mouse is reinitialized / replugged / etc - which indicates it is stored in runtime and set via a simple packet.

Unfortunately I do not have the ability to set up a virtualization environment right now, if anyone who has previously done captures (@AleMax @robertswiecki) would be willing to capture the LOD being modified, that would be fantastic. If not, I should be able to set up virtualization and capture this in the next week.

AleMax, 2021-07-10:
> Does anyone have any captures that involve LOD (lift off distance) changes? I've observed that the LOD setting is not stored in on-board flash, and is only active as long as Synapse is running, [...]

@grandprixgp 

I wasn't exactly sure what different settings you exactly wanted, so i put in a few different things.
The captures labelled with _smart_ are done from the smart calibration page,
The ones labelled _manual_ from the manual calibration.
The ones labelled _async-manual_ are from the async manual calibration and correspond to the Lift of setting distance.
I also added one _alt-mat_ because i wasnt sure if the mat you select makes a difference in the packets that are sent, so while i used to same mat for all captures, i used a different one for this one, so that in theory, one could check if it differs.

The numbers correspond to the values from start to finish that were tested in the capture (5-6-7 would mean, that its initial state was set on 5mm, then slided to 6mm lift off from the slider, slided to 7mm, and lift off again, a 5-7 would slide directly from 5mm to 7mm before lifting off from the slider)
I also added one 5-5 as i noticed it also sends a package even if i dont change the value (by just touching the slider and letting go)

As im not sure if wireshark saves all captures, if needed the filter should be

> usb.bus_id == 7 && usb.device_address == 3

I hope some of these can be of help!

[LOD-Captures.zip](https://github.com/openrazer/openrazer/files/6795440/LOD-Captures.zip)

grandprixgp, 2021-07-10:
> 
> 
> > Does anyone have any captures that involve LOD (lift off distance) changes? I've observed that the LOD setting is not stored in on-board flash, and is only active as long as Synapse is running, however if you close synapse, the LOD setting will remain until the mouse is reinitialized / replugged / etc - which indicates it is stored in runtime and set via a simple packet.
> > Unfortunately I do not have the ability to set up a virtualization environment right now, if anyone who has previously done captures (@AleMax @robertswiecki) would be willing to capture the LOD being modified, that would be fantastic. If not, I should be able to set up virtualization and capture this in the next week.
> 
> @grandprixgp
> 
> I wasn't exactly sure what different settings you exactly wanted, so i put in a few different things.
> The captures labelled with _smart_ are done from the smart calibration page,
> The ones labelled _manual_ from the manual calibration.
> The ones labelled _async-manual_ are from the async manual calibration and correspond to the Lift of setting distance.
> I also added one _alt-mat_ because i wasnt sure if the mat you select makes a difference in the packets that are sent, so while i used to same mat for all captures, i used a different one for this one, so that in theory, one could check if it differs.
> 
> The numbers correspond to the values from start to finish that were tested in the capture (5-6-7 would mean, that its initial state was set on 5mm, then slided to 6mm lift off from the slider, slided to 7mm, and lift off again, a 5-7 would slide directly from 5mm to 7mm before lifting off from the slider)
> I also added one 5-5 as i noticed it also sends a package even if i dont change the value (by just touching the slider and letting go)
> 
> As im not sure if wireshark saves all captures, if needed the filter should be
> 
> > usb.bus_id == 7 && usb.device_address == 3
> 
> I hope some of these can be of help!
> 
> [LOD-Captures.zip](https://github.com/openrazer/openrazer/files/6795440/LOD-Captures.zip)
> 
> Edit: I just noticed the smart capture apparently didnt save properly and is empty.
> I hope you dont need that, and if you do, tell me and i will capture it again later.

Thank you so much!

grandprixgp, 2021-07-10:
@AleMax If you're willing at some point to redo capture, there's a specific chain of captures that would be perfect for identifying the arguments.

Firstly add three surfaces: Razer Firefly Cloth Edition, Razer Firefly Hard Edition, Razer Firefly V2

For each surface, select LOD 3 (will be default but reselect), then select LOD 4, LOD 5. Switch surface, and repeat.

A separate capture for each surface would be perfect. Don't bother using async.

Thank you again regardless, you've saved me a bunch of time!

leimh, 2021-07-16:
About the calibration. With the exception of the custom/""manual"" calibration settings, all the others are predefined ones:
1) The razer mousepad predefined settings are usually stored in the following path: "C:\ProgramData/Razer/Synapse3/Service/Staging/Devices/Device_name/Bin/mmdatav*.ini"
U can check several of them, but u dont really need to create the full table via pcaps. It's already in mmdatav*.ini ;)
2) Generic settings (sync/async) are only 6 combinations (pcaps) that u really need. (+ 1 more the disable calib setting)
3) And for custom settings, the values are stored by synapse here:
"C:\ProgramData/Razer/Razer Central/Accounts/RZR_number_razer_id_account/Emily3/Devices/number_id_device/Calibrations/PixartCalibrationData.xml"
But for the focus+ sensor you will need more work to figure out how it really works to implement well that feature. (For previous sensors, 4g and 5g infrared, we probably got all the data to implement this feature fairly well...)

More info here: https://github.com/openrazer/openrazer/issues/783
The last 2 post are about "focus+" in the deathadder v2, but is the same sensor that got this device (viper 8khz) and should be similar.
Regards.

zischknall, 2021-09-01:
Hey, anything I can do to help with this? Otherwise I will probably try it myself later this week.

SM9CC, 2021-09-29:
Hello, I did a reinstall today so I figured I would check back on this, did a test with evhz and now I am able to reach 8khz

> Razer Razer Viper 8KHz: Latest  1000Hz, Average  4708Hz
Razer Razer Viper 8KHz: Latest  8000Hz, Average  4708Hz
Razer Razer Viper 8KHz: Latest   800Hz, Average  4595Hz
Razer Razer Viper 8KHz: Latest  1142Hz, Average  4551Hz
Razer Razer Viper 8KHz: Latest  1142Hz, Average  4444Hz
Razer Razer Viper 8KHz: Latest  4000Hz, Average  4381Hz
Razer Razer Viper 8KHz: Latest  1333Hz, Average  4277Hz
Razer Razer Viper 8KHz: Latest   999Hz, Average  4168Hz
Razer Razer Viper 8KHz: Latest   998Hz, Average  4121Hz
Razer Razer Viper 8KHz: Latest  2680Hz, Average  4039Hz
Razer Razer Viper 8KHz: Latest  2004Hz, Average  3945Hz
Razer Razer Viper 8KHz: Latest  1000Hz, Average  3836Hz
Razer Razer Viper 8KHz: Latest  1000Hz, Average  3725Hz
Razer Razer Viper 8KHz: Latest  1142Hz, Average  3619Hz
Razer Razer Viper 8KHz: Latest  7936Hz, Average  3618Hz
Razer Razer Viper 8KHz: Latest  1001Hz, Average  3571Hz
Razer Razer Viper 8KHz: Latest   888Hz, Average  3460Hz
Razer Razer Viper 8KHz: Latest  1000Hz, Average  3413Hz
Razer Razer Viper 8KHz: Latest  1333Hz, Average  3371Hz
Razer Razer Viper 8KHz: Latest   666Hz, Average  3319Hz
Razer Razer Viper 8KHz: Latest  1333Hz, Average  3277Hz
Razer Razer Viper 8KHz: Latest  4000Hz, Average  3277Hz
Razer Razer Viper 8KHz: Latest   997Hz, Average  3168Hz
Razer Razer Viper 8KHz: Latest   801Hz, Average  3056Hz
Razer Razer Viper 8KHz: Latest  2666Hz, Average  2973Hz
Razer Razer Viper 8KHz: Latest   666Hz, Average  2857Hz
Razer Razer Viper 8KHz: Latest   726Hz, Average  2806Hz
Razer Razer Viper 8KHz: Latest   399Hz, Average  2750Hz
Razer Razer Viper 8KHz: Latest   615Hz, Average  2636Hz
Razer Razer Viper 8KHz: Latest  2008Hz, Average  2604Hz
Razer Razer Viper 8KHz: Latest   242Hz, Average  2484Hz
Razer Razer Viper 8KHz: Latest  4000Hz, Average  2484Hz

I am running on Arch Linux with `openrazer-meta` `evhz-git` 
I enabled the openrazer daemon with `systemctl --user enable --now openrazer-daemon` and added myself to the `plugdev` user group using `sudo usermod -a -G plugdev <username>`

openglfreak, 2021-10-09:
I implemented almost everything in https://github.com/openglfreak/openrazer/commits/feature_viper8khz (based on the `feature_viper8khz` branch from this repo).
Everything that is implemented works.
The only missing thing is reading the polling rate from the device, which afaict isn't used by the daemon anyway.

closed 2021-10-30 by z3ntu

Evgenuch, 2023-12-01:
Guys, got a problem, plugged in 8khz in usb 3.2 port but still mouse rate checker showing max 2k hz, placed 8k in razer synapse and hidusbf (dont think that placed it right here) 
[image]

PR #1633 (2021-10-30): Add support for Razer Viper 8KHz
Fixes #1426