openrazer/openrazer #1428: Razer Book 2020

Niveuh, 2021-02-04:
Hi, 
The Razer Book isn't listed in the supported product, there is the infos, I didn't know if you need a package capture
Thanks for your work

### Device Info

* Name: Razer Book
* Product Number: RZ09-03571EM1
* Model Number: RZ09-0357

### Device Information
```
Please post the output of
    lsusb -d 1532: -v
```
Bus 003 Device 004: ID 1532:026a Razer USA, Ltd Razer Blade
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x026a 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Blade
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
```
Please post the output of
    ls -lah /dev/input/by-id/
```
total 0
drwxr-xr-x 2 root root 200 févr.  4 21:40 .
drwxr-xr-x 4 root root 680 févr.  4 21:40 ..
lrwxrwxrwx 1 root root  10 févr.  4 21:40 usb-Azurewave_Integrated_Camera_0000-event-if00 -> ../event18
lrwxrwxrwx 1 root root  10 févr.  4 21:40 usb-Azurewave_Integrated_Camera_0000-event-if02 -> ../event19
lrwxrwxrwx 1 root root   9 févr.  4 21:40 usb-ELAN_Touchscreen-event-if00 -> ../event9
lrwxrwxrwx 1 root root  10 févr.  4 21:40 usb-Razer_Razer_Blade-event-if01 -> ../event16
lrwxrwxrwx 1 root root  10 févr.  4 21:40 usb-Razer_Razer_Blade-event-kbd -> ../event12
lrwxrwxrwx 1 root root  10 févr.  4 21:40 usb-Razer_Razer_Blade-if01-event-kbd -> ../event13
lrwxrwxrwx 1 root root  10 févr.  4 21:40 usb-Razer_Razer_Blade-if02-event-mouse -> ../event17
lrwxrwxrwx 1 root root   9 févr.  4 21:40 usb-Razer_Razer_Blade-if02-mouse -> ../mouse3

z3ntu, 2021-02-08:
That's this thing, right? https://www.razer.com/productivity-laptops/razer-book-13

Niveuh, 2021-02-08:
> That's this thing, right? https://www.razer.com/productivity-laptops/razer-book-13

Yes

PR #1435 (2021-02-27): Support for the Razer Book 13 (2020)
Hi,

this adds support for the Keyboard of the Razer Book 13 (2020).

I basically took the [PR for the Blade Stealth Late 2020](https://github.com/openrazer/openrazer/pull/1371) as a template. But with reference to the Book 13 and its device ID (1532:026a). I tested the following via Polychromatic and all worked fine:

- Keyboard brightness
- Breath, Spectrum and Static modes
- Turning off keyboard backlight

So same as the other PR I've copy pasted the necessary bits of the Blade Stealth Late 2020.

Fixes #1428
closed 2021-02-27 by z3ntu