openrazer/openrazer #1429: Support for Razer BlackWidow V3 Pro 2020

EricGolovin, 2021-02-06:
### Device Info

* Name: Razer BlackWidow V3 Pro 2020 
* Product Number: RZ03-03530100
* Model Number: RZ03-0353

z3ntu, 2021-02-08:
Hi, please provide the output of `lsusb -d 1532: -v` and `ls -lah /dev/input/by-id/` and (if possible) Screenshots & Packet captures from Razer Synapse

z3ntu, 2021-02-08:
And probably a duplicate of #1279 ? If the usb ids from you and that issue match then you don't need to provide additional information.

EricGolovin, 2021-02-10:
> Hi, please provide the output of `lsusb -d 1532: -v` and `ls -lah /dev/input/by-id/` and (if possible) Screenshots & Packet captures from Razer Synapse

2021-02-10 23:51:51.417 system_profiler[4700:1490238] SPUSBDevice: IOCreatePlugInInterfaceForService failed 0xe00002be
USB:

    USB 3.1 Bus:

      Host Controller Driver: AppleUSBXHCIAR
      PCI Device ID: 0x15d4 
      PCI Revision ID: 0x0002 
      PCI Vendor ID: 0x8086 
      Bus Number: 0x01 

        PSSD T7:

          Product ID: 0x4001
          Vendor ID: 0x04e8  (Samsung Electronics Co., Ltd.)
          Version: 1.00
          Serial Number: S5SYNJ0N900129H
          Speed: Up to 10 Gb/s
          Manufacturer: Samsung
          Location ID: 0x01100000 / 1
          Current Available (mA): 900
          Current Required (mA): 896
          Extra Operating Current (mA): 0
          Media:
            PSSD T7:
              Capacity: 1 TB (1 000 204 886 016 bytes)
              Removable Media: No
              BSD Name: disk2
              Logical Unit: 0
              Partition Map Type: MBR (Master Boot Record)
              S.M.A.R.T. status: Verified
              USB Interface: 0
              Volumes:
                T7:
                  Capacity: 1 TB (1 000 202 273 280 bytes)
                  Free: 258,85 GB (258 850 029 568 bytes)
                  Writable: Yes
                  File System: ExFAT
                  BSD Name: disk2s1
                  Mount Point: /Volumes/T7
                  Content: Windows_NTFS
                  Volume UUID: C95DEF93-1CE4-347A-931C-DCF2F7904A80

    USB 3.1 Bus:

      Host Controller Driver: AppleUSBXHCIAR
      PCI Device ID: 0x15d4 
      PCI Revision ID: 0x0002 
      PCI Vendor ID: 0x8086 
      Bus Number: 0x00 

    USB 3.0 Bus:

      Host Controller Driver: AppleUSBXHCISPT
      PCI Device ID: 0xa12f 
      PCI Revision ID: 0x0031 
      PCI Vendor ID: 0x8086 

        Hub:

          Product ID: 0x8442
          Vendor ID: 0x0451  (Texas Instruments)
          Version: 1.00
          Serial Number: DF040841800D
          Speed: Up to 480 Mb/s
          Location ID: 0x14500000 / 4
          Current Available (mA): 500
          Current Required (mA): 0
          Extra Operating Current (mA): 0

            USB Receiver:

              Product ID: 0xc52b
              Vendor ID: 0x046d  (Logitech Inc.)
              Version: 12.11
              Speed: Up to 12 Mb/s
              Manufacturer: Logitech
              Location ID: 0x14530000 / 6
              Current Available (mA): 500
              Current Required (mA): 98
              Extra Operating Current (mA): 0

            Razer BlackWidow V3 Pro:

              Product ID: 0x025c
              Vendor ID: 0x1532
              Version: 1.00
              Serial Number: 000000000000
              Speed: Up to 12 Mb/s
              Manufacturer: Razer
              Location ID: 0x14520000 / 7
              Current Available (mA): 500
              Current Required (mA): 100
              Extra Operating Current (mA): 0

            Composite Device:

              Product ID: 0x82ff
              Vendor ID: 0x0451  (Texas Instruments)
              Version: 2.00
              Speed: Up to 480 Mb/s
              Location ID: 0x14550000 / 5
              Current Available (mA): 500
              Current Required (mA): 0
              Extra Operating Current (mA): 0

        Apple T1 Controller:

          Product ID: 0x8600
          Vendor ID: 0x05ac (Apple Inc.)
          Version: 1.01
          Manufacturer: Apple Inc.
          Location ID: 0x14200000

EricGolovin, 2021-02-12:
@z3ntu

z3ntu, 2021-02-12:
Yeah this is a duplicate of #1279 . Please also note that this repository is providing a **Linux** driver, nothing for macOS.

closed 2021-02-12 by z3ntu

EricGolovin, 2021-02-12:
> Yeah this is a duplicate of #1279 . Please also note that this repository is providing a **Linux** driver, nothing for macOS.

Thank you :) There is an open source project https://github.com/1kc/razer-macos and it is based on openrazer...