openrazer/openrazer #1432: Installation failed using Ubuntu 20.04

robertop96, 2021-02-09:
### Distribution

Ubuntu 20.04

### Support

- [X] Have you tried rebooting?
- [X] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I have Ubunto 20.04/ fresh install. followed steps as directed by the installation guide but ran with many issues.

```
➜  ~ sudo apt install openrazer-meta     
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  libdouble-conversion3 libpcre2-16-0 libqt5core5a libqt5dbus5 libqt5gui5 libqt5network5 libqt5svg5
  libqt5widgets5 libqt5xml5 libxcb-xinput0 qt5-gtk-platformtheme qttranslations5-l10n
Use 'sudo apt autoremove' to remove them.
The following additional packages will be installed:
  libblas3 libgfortran5 liblapack3 openrazer-daemon openrazer-driver-dkms python3-numpy
  python3-openrazer
Suggested packages:
  openrazer-doc gfortran python-numpy-doc python3-dev python3-pytest python3-numpy-dbg
The following NEW packages will be installed:
  libblas3 libgfortran5 liblapack3 openrazer-daemon openrazer-driver-dkms openrazer-meta
  python3-numpy python3-openrazer
0 upgraded, 8 newly installed, 0 to remove and 21 not upgraded.
Need to get 5,902 kB of archives.
After this operation, 26.2 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://us.archive.ubuntu.com/ubuntu focal/main amd64 libblas3 amd64 3.9.0-1build1 [142 kB]
Get:2 http://ppa.launchpad.net/openrazer/stable/ubuntu focal/main amd64 openrazer-driver-dkms all 2.9.0~ubuntu20.04.1 [50.3 kB]
Get:3 http://us.archive.ubuntu.com/ubuntu focal-updates/main amd64 libgfortran5 amd64 10.2.0-5ubuntu1~20.04 [736 kB]
Get:4 http://us.archive.ubuntu.com/ubuntu focal/main amd64 liblapack3 amd64 3.9.0-1build1 [2,154 kB]
Get:5 http://us.archive.ubuntu.com/ubuntu focal/main amd64 python3-numpy amd64 1:1.17.4-5ubuntu3 [2,725 kB]
Get:6 http://ppa.launchpad.net/openrazer/stable/ubuntu focal/main amd64 openrazer-daemon all 2.9.0~ubuntu20.04.1 [57.4 kB]
Get:7 http://ppa.launchpad.net/openrazer/stable/ubuntu focal/main amd64 python3-openrazer all 2.9.0~ubuntu20.04.1 [24.8 kB]
Get:8 http://ppa.launchpad.net/openrazer/stable/ubuntu focal/main amd64 openrazer-meta all 2.9.0~ubuntu20.04.1 [13.6 kB]
Fetched 5,902 kB in 2s (3,159 kB/s)       
Selecting previously unselected package libblas3:amd64.
dpkg: warning: files list file for package 'libpaper1:amd64' missing; assuming package has no files cu
rrently installed
(Reading database ... 228632 files and directories currently installed.)
Preparing to unpack .../0-libblas3_3.9.0-1build1_amd64.deb ...
Unpacking libblas3:amd64 (3.9.0-1build1) ...
Selecting previously unselected package libgfortran5:amd64.
Preparing to unpack .../1-libgfortran5_10.2.0-5ubuntu1~20.04_amd64.deb ...
Unpacking libgfortran5:amd64 (10.2.0-5ubuntu1~20.04) ...
Selecting previously unselected package liblapack3:amd64.
Preparing to unpack .../2-liblapack3_3.9.0-1build1_amd64.deb ...
Unpacking liblapack3:amd64 (3.9.0-1build1) ...
Selecting previously unselected package openrazer-driver-dkms.
Preparing to unpack .../3-openrazer-driver-dkms_2.9.0~ubuntu20.04.1_all.deb ...
Unpacking openrazer-driver-dkms (2.9.0~ubuntu20.04.1) ...
Selecting previously unselected package openrazer-daemon.
Preparing to unpack .../4-openrazer-daemon_2.9.0~ubuntu20.04.1_all.deb ...
Unpacking openrazer-daemon (2.9.0~ubuntu20.04.1) ...
Selecting previously unselected package python3-numpy.
Preparing to unpack .../5-python3-numpy_1%3a1.17.4-5ubuntu3_amd64.deb ...
Unpacking python3-numpy (1:1.17.4-5ubuntu3) ...
Selecting previously unselected package python3-openrazer.
Preparing to unpack .../6-python3-openrazer_2.9.0~ubuntu20.04.1_all.deb ...
Unpacking python3-openrazer (2.9.0~ubuntu20.04.1) ...
Selecting previously unselected package openrazer-meta.
Preparing to unpack .../7-openrazer-meta_2.9.0~ubuntu20.04.1_all.deb ...
Unpacking openrazer-meta (2.9.0~ubuntu20.04.1) ...
Setting up libblas3:amd64 (3.9.0-1build1) ...
update-alternatives: using /usr/lib/x86_64-linux-gnu/blas/libblas.so.3 to provide /usr/lib/x86_64-linu
x-gnu/libblas.so.3 (libblas.so.3-x86_64-linux-gnu) in auto mode
Setting up libgfortran5:amd64 (10.2.0-5ubuntu1~20.04) ...
Setting up openrazer-driver-dkms (2.9.0~ubuntu20.04.1) ...
Loading new openrazer-driver-2.9.0 DKMS files...
Building for 5.8.0-41-generic
Building initial module for 5.8.0-41-generic
ERROR: Cannot create report: [Errno 17] File exists: '/var/crash/openrazer-driver-dkms.0.crash'
Error! Bad return status for module build on kernel: 5.8.0-41-generic (x86_64)
Consult /var/lib/dkms/openrazer-driver/2.9.0/build/make.log for more information.
dpkg: error processing package openrazer-driver-dkms (--configure):
 installed openrazer-driver-dkms package post-installation script subprocess returned error exit statu
s 10
dpkg: dependency problems prevent configuration of openrazer-meta:
 openrazer-meta depends on openrazer-driver-dkms (= 2.9.0~ubuntu20.04.1); however:
  Package openrazer-driver-dkms is not configured yet.

dpkg: error processing package openrazer-meta (--configure):
 dependency problems - leaving unconfigured
Setting up liblapack3:amd64 (3.9.0-1build1) ...
No apport report written because the error message indicates its a followup error from a previous fail
ure.
    update-alternatives: using /usr/lib/x86_64-linux-gnu/lapack/liblapack.so.3 to provide /usr/lib/x86
_64-linux-gnu/liblapack.so.3 (liblapack.so.3-x86_64-linux-gnu) in auto mode
dpkg: dependency problems prevent configuration of openrazer-daemon:
 openrazer-daemon depends on openrazer-driver-dkms (= 2.9.0~ubuntu20.04.1); however:
  Package openrazer-driver-dkms is not configured yet.

dpkg: error processing package openrazer-daemon (--configure):No apport report written because the err
or message indicates its a followup error from a previous failure.

 dependency problems - leaving unconfigured
Setting up python3-numpy (1:1.17.4-5ubuntu3) ...
dpkg: dependency problems prevent configuration of python3-openrazer:
 python3-openrazer depends on openrazer-daemon (= 2.9.0~ubuntu20.04.1); however:
  Package openrazer-daemon is not configured yet.

dpkg: error processing package python3-openrazer (--configure):
 dependency problems - leaving unconfigured
Processing triggers for man-db (2.9.1-1) ...
No apport report written because MaxReports is reached already
                                                              Processing triggers for libc-bin (2.31-0
ubuntu9.2) ...
Errors were encountered while processing:
 openrazer-driver-dkms
 openrazer-meta
 openrazer-daemon
 python3-openrazer
E: Sub-process /usr/bin/dpkg returned an error code (1)
```
This is the content of /var/crash/openrazer-driver-dkms.0.crash
```
ProblemType: Package
DKMSBuildLog:
 DKMS make.log for openrazer-driver-2.9.0 for kernel 5.8.0-41-generic (x86_64)
 Mon 08 Feb 2021 08:18:13 PM EST
 -e
 ::^[[32m Compiling OpenRazer kernel modules^[[0m
 ========================================
 make -C /lib/modules/5.8.0-41-generic/build M=/var/lib/dkms/openrazer-driver/2.9.0/build/driver modules
 make[1]: Entering directory '/usr/src/linux-headers-5.8.0-41-generic'

   ERROR: Kernel configuration is invalid.
          include/generated/autoconf.h or include/config/auto.conf are missing.
          Run 'make oldconfig && make prepare' on kernel src to fix it.

 make[1]: *** [Makefile:746: include/config/auto.conf] Error 1
 make[1]: Leaving directory '/usr/src/linux-headers-5.8.0-41-generic'
 make: *** [Makefile:25: driver] Error 2
DKMSKernelVersion: 5.8.0-41-generic
Date: Mon Feb  8 20:18:16 2021
Package: openrazer-driver-dkms 2.9.0~ubuntu20.04.1
PackageVersion: 2.9.0~ubuntu20.04.1
SourcePackage: openrazer
Title: openrazer-driver-dkms 2.9.0~ubuntu20.04.1: openrazer-driver kernel module failed to build

```
* I have linux headers for my kernel.
* Universe repositories enabled.

lah7, 2021-02-09:
It could be due to a mismatching Linux kernel and headers version. The latest kernel version (at time of writing) is `5.8.0-43`, but testing a fresh live session here showed `5.8.0-41`.

Try upgrading/reinstalling the packages (or the entire system) and then reinstall the driver again:

    sudo apt install --reinstall linux-generic-hwe-20.04 linux-headers-generic-hwe-20.04
    sudo apt install --reinstall openrazer-driver-dkms

robertop96, 2021-02-09:
Thanks for the fast reply, I have done done as suggested and,

**After running**

```sudo apt install --reinstall linux-generic-hwe-20.04 linux-headers-generic-hwe-20.04```
```
 ~ sudo apt install --reinstall linux-generic-hwe-20.04 linux-headers-generic-hwe-20.04

[sudo] password for roberto: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  libdouble-conversion3 libpcre2-16-0 libqt5core5a libqt5dbus5 libqt5gui5
  libqt5network5 libqt5svg5 libqt5widgets5 libqt5xml5 libxcb-xinput0
  qt5-gtk-platformtheme qttranslations5-l10n
Use 'sudo apt autoremove' to remove them.
The following additional packages will be installed:
  linux-headers-5.8.0-43-generic linux-hwe-5.8-headers-5.8.0-43
  linux-image-5.8.0-43-generic linux-image-generic-hwe-20.04
  linux-modules-5.8.0-43-generic linux-modules-extra-5.8.0-43-generic
Suggested packages:
  fdutils linux-doc | linux-hwe-5.8-source-5.8.0 linux-hwe-5.8-tools
The following NEW packages will be installed:
  linux-headers-5.8.0-43-generic linux-hwe-5.8-headers-5.8.0-43
  linux-image-5.8.0-43-generic linux-modules-5.8.0-43-generic
  linux-modules-extra-5.8.0-43-generic
The following packages will be upgraded:
  linux-generic-hwe-20.04 linux-headers-generic-hwe-20.04
  linux-image-generic-hwe-20.04
3 upgraded, 5 newly installed, 0 to remove and 18 not upgraded.
4 not fully installed or removed.
Need to get 77.2 MB of archives.
After this operation, 372 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://us.archive.ubuntu.com/ubuntu focal-updates/main amd64 linux-modules-5.8.0-43-generic amd64 5.8.0-43.49~20.04.1 [15.0 MB]
Get:2 http://us.archive.ubuntu.com/ubuntu focal-updates/main amd64 linux-image-5.8.0-43-generic amd64 5.8.0-43.49~20.04.1 [9,499 kB]
Get:3 http://us.archive.ubuntu.com/ubuntu focal-updates/main amd64 linux-modules-extra-5.8.0-43-generic amd64 5.8.0-43.49~20.04.1 [40.1 MB]
Get:4 http://us.archive.ubuntu.com/ubuntu focal-updates/main amd64 linux-generic-hwe-20.04 amd64 5.8.0.43.49~20.04.29 [1,932 B]
Get:5 http://us.archive.ubuntu.com/ubuntu focal-updates/main amd64 linux-image-generic-hwe-20.04 amd64 5.8.0.43.49~20.04.29 [2,708 B]
Get:6 http://us.archive.ubuntu.com/ubuntu focal-updates/main amd64 linux-hwe-5.8-headers-5.8.0-43 all 5.8.0-43.49~20.04.1 [11.3 MB]
Get:7 http://us.archive.ubuntu.com/ubuntu focal-updates/main amd64 linux-headers-5.8.0-43-generic amd64 5.8.0-43.49~20.04.1 [1,218 kB]
Get:8 http://us.archive.ubuntu.com/ubuntu focal-updates/main amd64 linux-headers-generic-hwe-20.04 amd64 5.8.0.43.49~20.04.29 [2,596 B]
Fetched 77.2 MB in 3s (22.9 MB/s)                           
Selecting previously unselected package linux-modules-5.8.0-43-generic.
dpkg: warning: files list file for package 'libpaper1:amd64' missing; assuming p
ackage has no files currently installed
(Reading database ... 229214 files and directories currently installed.)
Preparing to unpack .../0-linux-modules-5.8.0-43-generic_5.8.0-43.49~20.04.1_amd
64.deb ...
Unpacking linux-modules-5.8.0-43-generic (5.8.0-43.49~20.04.1) ...
Selecting previously unselected package linux-image-5.8.0-43-generic.
Preparing to unpack .../1-linux-image-5.8.0-43-generic_5.8.0-43.49~20.04.1_amd64
.deb ...
Unpacking linux-image-5.8.0-43-generic (5.8.0-43.49~20.04.1) ...
Selecting previously unselected package linux-modules-extra-5.8.0-43-generic.
Preparing to unpack .../2-linux-modules-extra-5.8.0-43-generic_5.8.0-43.49~20.04
.1_amd64.deb ...
Unpacking linux-modules-extra-5.8.0-43-generic (5.8.0-43.49~20.04.1) ...
Preparing to unpack .../3-linux-generic-hwe-20.04_5.8.0.43.49~20.04.29_amd64.deb
 ...
Unpacking linux-generic-hwe-20.04 (5.8.0.43.49~20.04.29) over (5.8.0.41.46~20.04
.27) ...
Preparing to unpack .../4-linux-image-generic-hwe-20.04_5.8.0.43.49~20.04.29_amd
64.deb ...
Unpacking linux-image-generic-hwe-20.04 (5.8.0.43.49~20.04.29) over (5.8.0.41.46
~20.04.27) ...
Selecting previously unselected package linux-hwe-5.8-headers-5.8.0-43.
Preparing to unpack .../5-linux-hwe-5.8-headers-5.8.0-43_5.8.0-43.49~20.04.1_all
.deb ...
[... 146 lines cut ...]
```
**After running**

```sudo apt install --reinstall openrazer-driver-dkms  ```

```
sudo apt install --reinstall openrazer-driver-dkms

Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  libdouble-conversion3 libpcre2-16-0 libqt5core5a libqt5dbus5 libqt5gui5
  libqt5network5 libqt5svg5 libqt5widgets5 libqt5xml5 libxcb-xinput0
  linux-headers-5.8.0-36-generic linux-hwe-5.8-headers-5.8.0-36
  linux-image-5.8.0-36-generic linux-modules-5.8.0-36-generic
  linux-modules-extra-5.8.0-36-generic qt5-gtk-platformtheme
  qttranslations5-l10n
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 1 reinstalled, 0 to remove and 18 not upgraded.
4 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
E: Internal Error, No file name for openrazer-driver-dkms:amd64
➜  ~ 
```

lah7, 2021-02-09:
It looks like the kernel upgraded OK and the DKMS successfully installed the driver. Now just needs `apt`/`dpkg` to configure them.

Try:

    sudo dpkg --configure -a
    sudo apt-get -f install

If that didn't work, try uninstalling and installing all of OpenRazer's packages again:

    sudo apt remove openrazer-meta openrazer-driver-dkms openrazer-daemon python3-openrazer 
    sudo apt install openrazer-meta

robertop96, 2021-02-09:
**After Running**
 
 ```sudo dpkg --configure -a```
 
```
sudo dpkg --configure -a
[sudo] password for roberto: 
Setting up openrazer-driver-dkms (2.9.0~ubuntu20.04.1) ...
Removing old openrazer-driver-2.9.0 DKMS files...

-------- Uninstall Beginning --------
Module:  openrazer-driver
Version: 2.9.0
Kernel:  5.8.0-43-generic (x86_64)
-------------------------------------

Status: Before uninstall, this module version was ACTIVE on this kernel.

razerkbd.ko:
 - Uninstallation
   - Deleting from: /lib/modules/5.8.0-43-generic/updates/dkms/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

razermouse.ko:
 - Uninstallation
   - Deleting from: /lib/modules/5.8.0-43-generic/updates/dkms/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

razerkraken.ko:
 - Uninstallation
   - Deleting from: /lib/modules/5.8.0-43-generic/updates/dkms/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

razeraccessory.ko:
 - Uninstallation
   - Deleting from: /lib/modules/5.8.0-43-generic/updates/dkms/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

depmod...

DKMS: uninstall completed.

------------------------------
Deleting module version: 2.9.0
completely from the DKMS tree.
------------------------------
Done.
Loading new openrazer-driver-2.9.0 DKMS files...
Building for 5.8.0-41-generic 5.8.0-43-generic
Building initial module for 5.8.0-41-generic
ERROR: Cannot create report: [Errno 17] File exists: '/var/crash/openrazer-driver-dkms.0.crash'
Error! Bad return status for module build on kernel: 5.8.0-41-generic (x86_64)
Consult /var/lib/dkms/openrazer-driver/2.9.0/build/make.log for more information.
dpkg: error processing package openrazer-driver-dkms (--configure):
 installed openrazer-driver-dkms package post-installation script subprocess returned error exit status 10
dpkg: dependency problems prevent configuration of openrazer-meta:
 openrazer-meta depends on openrazer-driver-dkms (= 2.9.0~ubuntu20.04.1); however:
  Package openrazer-driver-dkms is not configured yet.

dpkg: error processing package openrazer-meta (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of openrazer-daemon:
[... 193 lines cut ...]

dpkg: error processing package openrazer-daemon (--configure):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of python3-openrazer:
 python3-openrazer depends on openrazer-daemon (= 2.9.0~ubuntu20.04.1); however:
  Package openrazer-daemon is not configured yet.

dpkg: error processing package python3-openrazer (--configure):
 dependency problems - leaving unconfigured
No apport report written because the error message indicates its a followup erro
r from a previous failure.
                          No apport report written because MaxReports is reached
 already
        Processing triggers for man-db (2.9.1-1) ...
Errors were encountered while processing:
 openrazer-driver-dkms
 openrazer-meta
 openrazer-daemon
 python3-openrazer
E: Sub-process /usr/bin/dpkg returned an error code (1)
➜  ~ 
 ```

***I just want to give thanks in advance for the time taken to review my issues.
It is greatly appreciated.***

lah7, 2021-02-09:
You're welcome. This is surprisingly unstable for a fresh install of Ubuntu 20.04. Something has _thrown a spanner at the works_. 🔧 

> ERROR: Cannot create report: [Errno 17] File exists: '/var/crash/openrazer-driver-dkms.0.crash'

That's apport making noise,  try cleaning the crash dumps:

    sudo rm /var/crash/*

> dpkg: warning: files list file for package 'libpaper1:amd64' missing; assuming package has no files currently installed

Has other package installations/upgrades failed since installing Ubuntu? Try reinstalling the affected package:

    sudo apt install --reinstall libpaper1

> Consult /var/lib/dkms/openrazer-driver/2.9.0/build/make.log for more information

Please paste the output. We're back to square one where DKMS isn't installing the driver, but it might work manually:

    sudo dkms install openrazer-driver/2.9.0

(This won't help with `dpkg`'s incomplete state as it is aborting due to an error)

> No apport report written because the error message indicates its a followup error from a previous failure.
> No apport report written because MaxReports is reached already

Some answers on the internet suggest running some apt commands to do some housecleaning. See if these help solve the issue: https://askubuntu.com/a/360036

> ➜  ~ 

Are you using a different shell to `bash`? If so, please try `/bin/bash`. If this just made no sense, scratch that, the interpreter usually reads `username@hostname:~$` and I'm not sure if this is the prompt for a different one.

CookedApps, 2021-04-28:
I have the same issue for some time now on Linux Mint 20.1 Cinnamon with Kernel version 4.15.0-58-generic.

lah7, 2021-04-28:
@CookedApps Please create a new issue, uninstall and let us know the output when following the install instructions.

@robertop96 Did you solve this issue in the end?

CookedApps, 2021-04-28:
@lah7 I updated my kernel to the newest version (5.8.0-50-generic) and I was able to reinstall openrazer-meta without a problem after that. Now it's working again. 👍

hensg, 2021-05-12:
I had the same issue, I was able to solve reinstalling linux kernel headers.

```
$ cat /var/lib/dkms/openrazer-driver/3.0.1/build/make.log
make -C /lib/modules/5.8.0-50-generic/build M=/var/lib/dkms/openrazer-driver/3.0.1/build/driver modules
make[1]: Entering directory '/usr/src/linux-headers-5.8.0-50-generic'

  ERROR: Kernel configuration is invalid.
         include/generated/autoconf.h or include/config/auto.conf are missing.
         Run 'make oldconfig && make prepare' on kernel src to fix it.
```

Then
```
$ sudo apt install --reinstall linux-headers-$(uname -r)
```

closed 2021-10-10 by github-actions[bot]