openrazer/openrazer #1433: Add more lighting effects for Naga Trinity

ohaessler, 2021-02-09:
### Distribution
RHEL 8
openrazer 2.9.0 (master branch)

### Support

- [ yes ] Have you tried rebooting?
- [ yes ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

In Polychromatic and RazerGenie I can only set a static lighting effect for my Naga Trinity - 1532:0067. I can also only set it for the whole mouse.

For my Naga Hex V2, I can set 4 lighting effects and different effects for the three sections (mouse wheel, logo, left side buttons).

I checked and the Naga Trinity supports the same settings as the Naga Hex V2 on Windows with Razer Synapse.

Please add the missing effects and separation for the three sections as well for the Naga Trinity.

Thanks
Oliver

lah7, 2021-02-09:
This is what I've gathered from this mouse:

* Naga Trinity supposedly supports a custom matrix ("advanced effects"), but it wasn't implemented properly in the driver - #1305

* Naga Trinity might not support "hardware effects" but actually emulates the static effect to achieve the same effect - https://github.com/openrazer/openrazer/issues/541#issuecomment-531107621 - OpenRazer does not handle this "static only" behaviour.

The first option would be a good starting point. I'm afraid others familiar with the driver will need to chime in here on how to modify the driver/daemon code to test custom frames to the hardware.

TermoZour, 2021-02-09:
This issue references #1323

ohaessler, 2021-06-16:
is there any progress on this?

rischiopoco, 2024-03-23:
Are there any updates regarding this feature?

z3ntu, 2024-04-04:
Doesn't look like it, somebody with the device will have to try various code paths and see if they work on the device, https://github.com/openrazer/openrazer/blob/master/DEVELOPMENT.md