openrazer/openrazer #1436: Elite blackwidow - freezes

purple-cmd, 2021-02-10:
I get this when running dmesg:

```
ms: 47 Reserved: 0f Command: 03 Params: 000001001500ff0000ff0000ff0000ff .
[  829.892911] razer driver: Device data transfer failed.
[  829.893105] razer driver: Invalid USB response. USB Report length: -32
[  829.893107] razerkbd: Invalid Report Length. Start Marker: 02 id: 1f Num Para
```

z3ntu, 2021-02-18:
Can you describe what you mean with "led freezes" in more detail?

closed 2021-07-28 by github-actions[bot]

OsoaGH, 2022-07-17:
I think this is also happening to me, I'm not sure how to check the log for this exact error (bit of a linux noob) but I am also using a Blackwidow Elite. What I think they mean is that the LED will work normally for a random amount of time (normally around 30mins) but then will randomly freeze.

z3ntu, 2022-07-23:
@OsoaGH Please open a new issue with any details you can find, logs, observations etc.