openrazer/openrazer #1437: Locking session in Cinnamon causes Chroma to turn off

HeyyItsBran, 2021-02-12:
- Locking session in Cinnamon causes brighness file to be set 49 on Deathadder Elite and Firefly, and 0 on 2014 Blackwidow Chroma, on all the leds turn completely off
- Mouse and Firefly will reliably restore previous setting when unlocked, keyboard will stay off unless file is changed manually
- If computer is shut down from the lock screen all devices will stay off at next boot until file is changed manually

System:
Kernel: 5.4.0-65-generic x86_64 
bits: 64 
compiler: gcc 
v: 9.3.0 
Desktop: Cinnamon 4.8.6 
wm: muffin 
dm: LightDM 
Distro: Linux Mint 20.1 Ulyssa 
base: Ubuntu 20.04 focal

z3ntu, 2021-02-12:
Generally the "devices off when the lockscreen is active" is a feature. If you want to disable that you can do that here: https://github.com/openrazer/openrazer/blob/master/daemon/resources/man/razer.conf.5.scd#L33-L34

HeyyItsBran, 2021-02-13:
Great find, thank you! I've updated OP to focus on the settings not restoring when unlocked/after restart

z3ntu, 2021-02-15:
What keyboard do you have? Please provide `lsusb | grep 1532` output

HeyyItsBran, 2021-02-18:
Hiya, sorry for the late reply, Its a Blackwidow Chroma Stealth Edition model RZ03-0122 (circa 2014/15 iirc)

`Bus 005 Device 002: ID 1532:0203 Razer USA, Ltd BlackWidow Chroma`
`Bus 003 Device 002: ID 1532:0c00 Razer USA, Ltd Firefly`
`Bus 001 Device 002: ID 1532:005c Razer USA, Ltd Razer DeathAdder Elite`

closed 2021-10-07 by github-actions[bot]