openrazer/openrazer #1438: Ornata V2 Chroma on Mint 20.1

MislavReversingLabs, 2021-02-13:
### Distribution

Linux Mint 20.1

### Support

- [yes] Have you tried rebooting?
- [yes] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Hi. I'm using Ornata V2 Chroma on Mint 20.1 and I installed the current stable version of openrazer. RazerGenie says that openrazer didn't recognize my device.
If support for Ornata V2 Chroma is not yet in the stable pacakages, can I install it on Mint 20.1 from a dev build somehow until the next stable release? Thanks.

z3ntu, 2021-02-13:
Please see the website https://openrazer.github.io/#ubuntu : 

> Alternately, if you wish to use the latest development builds, use daily instead -- use at your own risk!
> 
> $ sudo add-apt-repository ppa:openrazer/daily

MislavReversingLabs, 2021-02-13:
Ok, got it. Must have missed that sentence somehow. 
Works well enough, with ripple effect and custom configuration missing in RazerGenie at this point.
Thank you.

closed 2021-02-13 by MislavReversingLabs