openrazer/openrazer #1440: Razer Huntsman macro for non-macro keys (normal ones)l

gnapiorkowski, 2021-02-16:
### Distribution

Manjaro 20.2.1
openrazer 2.9.0-1

### Support

- [x ] Have you tried rebooting?
- [x ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Hi,
My Razer Keyboard:

* Name: Huntsman
* Product Number: RZ03-02520100-R3M1
* Model Number: RZ03-0252
```
1532:0227 Razer USA, Ltd Huntsman
```
doesn't allow me to use setting a macro for a normal key (not M1, M2,M3... keys, but the normal ones like QWERTY keys). I can easily do that in Synapse on Windows, but openrazer doesn't allow me to do that for some reason. The macro led blinks, but it shuts off as soon as I press any key.
[image]

z3ntu, 2021-02-18:
Should be handled by #1124

z3ntu, 2021-02-18:
Please also try e.g. https://github.com/sezanzeb/key-mapper if that works for your use case. I'd like to have feedback whether that works well. There are also some other applications listed at https://polychromatic.app/docs/faqs/#can-i-remap-keys

ArnoldDeRuiter, 2021-02-19:
Sezanzeb Key Mapper works wonders. :+1:

gnapiorkowski, 2021-02-19:
Wow, that seems perfect! Can't believe that I struggled with that for so long... I tried some remappings in X like setxkbmap -option caps:swapescape (because that's the most important one for me) but that was really prone to failure and broke when for example playing some games or sometimes randomly...  I also struggled with a custom keyboard map hoping that it would be a more robust solution, but I couldn't get that to work.

I have just started to use the key-mapper, didn't experience any issues for now, it seems beautifully simple and really powerful at the same time. I love it! Thank you for that suggestion!

closed 2021-07-21 by github-actions[bot]