openrazer/openrazer #1442: Support for [device name here]

dgmiller81, 2021-02-18:
### Device Info

* Name: Razer Blade 15 (2021)
* Product Number: RZ09-0369BEA2
* Model Number: RZ09-0369
### Device Information
```
Please post the output of
    lsusb -d 1532: -v
```
Bus 001 Device 004: ID 1532:026f Razer USA, Ltd Razer Blade
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x026f 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1

```
Please post the output of
    ls -lah /dev/input/by-id/
```
lrwxrwxrwx 1 root root   9 Feb 16 00:36 usb-Azurewave_USB_Camera_0x0001-event-if00 -> ../event6
lrwxrwxrwx 1 root root   9 Feb 16 00:36 usb-Logitech_USB_Receiver-if02-event-mouse -> ../event7
lrwxrwxrwx 1 root root   9 Feb 16 00:36 usb-Logitech_USB_Receiver-if02-mouse -> ../mouse2
lrwxrwxrwx 1 root root  10 Feb 16 00:36 usb-Razer_Razer_Blade-event-if01 -> ../event13
lrwxrwxrwx 1 root root  10 Feb 16 00:36 usb-Razer_Razer_Blade-event-kbd -> ../event10
lrwxrwxrwx 1 root root  10 Feb 16 00:36 usb-Razer_Razer_Blade-if01-event-kbd -> ../event11
lrwxrwxrwx 1 root root  10 Feb 16 00:36 usb-Razer_Razer_Blade-if02-event-mouse -> ../event17
lrwxrwxrwx 1 root root   9 Feb 16 00:36 usb-Razer_Razer_Blade-if02-mouse -> ../mouse3

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>
Will work on this.

### Synapse Screenshots
<Screenshots of all available options as seen on Razer Synapse>
Will work on this.

dgmiller81, 2021-02-20:
I'm new to github, so there might be some other places where I can ask some questions.    But I'm really interested in knowing where I can find some information regarding how the drivers work and how I can take an existing driver and add whatever to it to test my keyboard.    The biggest thing I'd like to see is getting the keyboard to a single color and not changing all the time.   I'd like to test that and get that, and then start to contribute once I know what I'm doing.

z3ntu, 2021-02-27:
@dgmiller81 It's probably the simplest to look at recent commits for similar devices (Blade laptops) and do similar changes for your device (with USB PID 026f), e.g. e5707574955a9b5512b76937c842b0bbc1e4f578 or b0b694e2d3119551fcf0f56ccf378c12ebf051fc or c744889e3cb304b5c4801d788ef777108af1b1f6 or 1d31ff5e1d47646497e25b6a822f79ec49e83dcf . Basically first add your pid to razerkbd_driver.h , add the thing with HID_USB_DEVICE to razerkbd_driver.c , the few lines with the `razer_attr_read_device_type` function & `is_blade_laptop` . Add the pid to the 99-razer.rules file, install the new udev file, install the new kernel modules to disk (see https://github.com/openrazer/openrazer/wiki/Building-a-package ) and the driver should be bound to the device. Also please make sure to have a spare keyboard on hand because it might happen that your keyboard will not work temporarily :) If you have any more specific questions feel free to ask (e.g. via https://github.com/openrazer/openrazer/wiki/Community-chat or here)

cachvico, 2021-03-05:
I have a 2021 Blade 15 arriving in a week and will too be attempting to get Ubuntu working on it within the 30 day returns period. Updates to follow.

cachvico, 2021-03-05:
I admit I am somewhat confused as to the status quo regarding Linux on Razers.

It is not clear to me if basic Ubuntu installations work - thinking about the keyboard, trackpad, wifi, webcam and so on. Is it the case that the basic OS does work but the drivers at openrazer are needed for custom control of the LEDs and so on?

dgmiller81, 2021-03-05:
I’d be happy to help you.   I run Ubuntu on mine and it works great!  Only a couple of tweaks.  Be happy to assist in anyway.

dgmiller81, 2021-03-07:
> I admit I am somewhat confused as to the status quo regarding Linux on Razers Blades.
> 
> It is not clear to me if basic Ubuntu installations work - thinking about the keyboard, trackpad, wifi, camera and so on. Is it the case that the built-in laptop features do work but the drivers at openrazer are needed for custom control of the keyboard LEDs and so on?
> 
> Or do the basic features like wifi and camera not work out of the box without openrazer.

All of the basic features work and run great.    I had to do one tweak with my power settings for sleep, that was really it.    Razer drivers just give you additional features like the lighting, maybe some macro record functions, etc.   But everything you need aside from those works out of the box.   It's REALLY well done.   I've been very impressed with it honestly.

dgmiller81, 2021-03-07:
> @dgmiller81 It's probably the simplest to look at recent commits for similar devices (Blade laptops) and do similar changes for your device (with USB PID 026f), e.g. [e570757](https://github.com/openrazer/openrazer/commit/e5707574955a9b5512b76937c842b0bbc1e4f578) or [b0b694e](https://github.com/openrazer/openrazer/commit/b0b694e2d3119551fcf0f56ccf378c12ebf051fc) or [c744889](https://github.com/openrazer/openrazer/commit/c744889e3cb304b5c4801d788ef777108af1b1f6) or [1d31ff5](https://github.com/openrazer/openrazer/commit/1d31ff5e1d47646497e25b6a822f79ec49e83dcf) . Basically first add your pid to razerkbd_driver.h , add the thing with HID_USB_DEVICE to razerkbd_driver.c , the few lines with the `razer_attr_read_device_type` function & `is_blade_laptop` . Add the pid to the 99-razer.rules file, install the new udev file, install the new kernel modules to disk (see https://github.com/openrazer/openrazer/wiki/Building-a-package ) and the driver should be bound to the device. Also please make sure to have a spare keyboard on hand because it might happen that your keyboard will not work temporarily :) If you have any more specific questions feel free to ask (e.g. via https://github.com/openrazer/openrazer/wiki/Community-chat or here)

Thanks for this information.   I'll check this out.  This is EXACTLY what I was interested in finding.   Much appreciated.

cachvico, 2021-03-10:
So I got Ubuntu installed... but it doesn't recognize the wifi adapter. Any ideas?

I added openrazer-meta package as per the download instructions for Ubuntu.

Only thing to report so far is that `iw dev | grep Interface` is empty. I'm not sure where to go from there..

cachvico, 2021-03-12:
Well I had to upgrade the kernel to 5.10 and install the driver for the Intel Wi-Fi 6 AX210 160MHz (from Intel's site) and now wifi works!

Camera works too ... just problems waking from sleep, the display blinks on and off sometimes... Bluetooth not working yet either..

But at least I have wifi!

dgmiller81, 2021-03-12:
> Well I had to upgrade the kernel to 5.10 and install the driver for the Intel Wi-Fi 6 AX210 160MHz (from Intel's site) and now wifi works!
> 
> Camera works too ... just problems waking from sleep, the display blinks on and off sometimes... Bluetooth not working yet either..
> 
> But at least I have wifi!

Sorry, I didn't see the first response.    Wifi worked on 5.8 for me, so I'm not sure if something is different between our models.   I have the Blade 15 (not advance or anything).   Model listed above.   Are you sure you have the same model?   Curious why you would have issues with Bluetooth and Wifi and I would not.   I am running Ubuntu 20.04 with 5.8.0-44 kernel.   

I did have the sleep issue, but I followed this to resolve my sleep.  Other than that, I have literally had no issues with Ubuntu 20.04.   Here is the link for the sleep issue:  https://help.ubuntu.com/community/RazerBlade#Suspend

cachvico, 2021-03-12:
I do have the Advanced, perhaps the wifi is a different module. https://www.intel.com/content/www/us/en/support/articles/000005511/wireless.html does list kernel 5.10+ support only for that one (AX210), so that makes sense.

I just made the change for the sleep issue that you pointed me to, works perfectly!

dgmiller81, 2021-03-12:
> I do have the Advanced, perhaps the wifi is a different module. https://www.intel.com/content/www/us/en/support/articles/000005511/wireless.html does list kernel 5.10+ support only for that one [...]

Yeah, that's probably likely the difference.   Just slightly different needs/drivers.    The advance wasn't in stock at the time when I purchased... From what I could tell, I got the advanced with a 3070 essentially vs. the 3080.    But looks like maybe there are a couple of other things that are different.   It said I got the (AX210) but maybe there is a variant of it with the advance.   Glad the other fix worked for you.   I really love linux on this thing.   It's great and so far, not a single regret buying this laptop.   I'm sold on it so far.

westleyfsm, 2021-04-29:
> Thanks for this information. I'll check this out. This is EXACTLY what I was interested in finding. Much appreciated.

Have you made any progress? I will take a shot at it if you haven't.

dgmiller81, 2021-04-29:
I haven’t...  been tied up at work with other focuses.   No urgency on my part but it’ll be a while before I get to it.

cachvico, 2021-04-29:
Here's my setup notes for Ubuntu 21 so far - https://gist.github.com/cachvico/03b6e37c8907e277dafc6c6183a732d5

It achieves the following:

- Gets wifi working
- Disables Bluetooth as I couldn't get it to work
- Fixes resume after lid close

Apart from Bluetooth, the only remaining problem I couldn't fix is that the entire display appears to jump up and down by a single pixel every few minutes or so. This happens even in console mode, no X11 involved. But not in Windows, so definitely not a hardware fault.

Any ideas about any of this would be greatly appreciated!

eirikaso, 2021-05-04:
Joining in on this thread. I have the Blade 15 Advanced 2021. 
Have you tried using an external hub. I have a Dell TB16 USB-C hub that works ok, but there are some issues. Sometimes when I boot only one of two external monitors display an image. All monitors seem to be connected though, since the login screen is on the one monitor that doesn't show any image, so the login is hidden behind a black screen. It works when I unplug and insert the USB-C, but then the mouse and keyboard that was connected to the hub stops working :/

Anyone else having similar issues?

cachvico, 2021-05-04:
@eirikaso to clarify, this the built-in display that is flickering, not external!

westleyfsm, 2021-05-04:
> Joining in on this thread. I have the Blade 15 Advanced 2021.
> Have you tried using an external hub. I have a Dell TB16 USB-C hub that works ok, but there are some issues. Sometimes when I boot only [...]

Have the same black screen issue, but my other devices still work after plugging it back in. Sorry!

eirikaso, 2021-05-05:
> Have the same black screen issue, but my other devices still work after plugging it back in. Sorry!

Didn't make more progress so I'm sending it back and will get a Tuxedo Xp15 instead that. It's my work computer so I can't spend days trying to fix linux issues on it

MagicGnome-UwU, 2021-05-08:
probably a dumb question but is it already supported because the daemon couldnt find the device

EtienneMueller, 2021-05-20:
I added support for the Razer Blade 15 Advanced 2021 and created a pull request: https://github.com/openrazer/openrazer/pull/1516

It works perfectly on my Blade, but apparently the check has failed, I will have to look into it.

EtienneMueller, 2021-05-21:
I just saw in your post that the PID of the Base Model is 026F, so I added it to my fork: https://github.com/EtienneMueller/openrazer/tree/razer_blade_base_early_2021

But because I have the Advance model, I can't test it. If any of you could try it and tell if it works, that would be great. Then I would do a pull request.

westleyfsm, 2021-06-12:
> I just saw in your post that the PID of the Base Model is 026F, so I added it to my fork: https://github.com/EtienneMueller/openrazer/tree/razer_blade_base_early_2021
>
> But because I have the Advance [...]

I tried it on mine and it did not work. Keyboard didn't work and device not recognized

EtienneMueller, 2021-06-13:
@westleyfsm okay, that is too bad, I thought it would have been an easy fix as well. Then I am unfortunately not able to help you guys

PR #1653 (2021-11-30): Adding Base 15 Early 2021
Made the changes I recommended a while back - they seem stable on my machine.

[image]

[image]

Any additional info needed just let me know.

Fixes #1442
closed 2021-11-30 by z3ntu