openrazer/openrazer #1443: Please change dbus-user-session dependency to recommends

GreenBeard, 2021-02-21:
I apologize for not submitting a pull request, but I have barely worked with Debian control files before. I believe that the dbus-user-session dependency should be a recommends as opposed to a hard dependency. It appears to work without it, and some Debian-based distributions don't provide dbus-user-session, additionally it seems to be a somewhat of a user preference (meaning if a user prefers they should be able to revert to the default user session behavior). I worked around this by just creating a fake dbus-user-session package.

Relevant line https://github.com/openrazer/openrazer/blob/25677427e9218f978749562a23129704500e55d6/debian/control#L73

z3ntu, 2021-02-21:
@daissi You're probably more familiar with this ^

z3ntu, 2021-02-27:
@GreenBeard Can you please also clarify what "some Debian-based distributions" means?

GreenBeard, 2021-02-27:
@z3ntu I know distributions without systemd at least don't have dbus. https://without-systemd.frama.wiki/linux_distributions_without_systemd (if you scroll down to the Debian, and Devuan based ones).

z3ntu, 2021-02-27:
Devuan does seem to have dbus-user-session though as far as I can tell: https://pkginfo.devuan.org/cgi-bin/policy-query.html?c=package&q=dbus-user*

GreenBeard, 2021-02-27:
@z3ntu I will double check in a few hours (when I'm back at my desktop), but the only reason I noticed is that it was the only unsatisfied dependency when I tried to install.

GreenBeard, 2021-02-27:
@z3ntu I don't think dbus-user-session is available currently for Devuan (it may be in a nightly version), for reference a version is available of dbus if you search https://pkginfo.devuan.org/cgi-bin/policy-query.html?c=package&q=dbus as you can see it shows which versions have it available, but if you search for dbus-user-session as you did no versions show up. I also checked on my computer, and cannot find any dbus-user-session packages (I can find dbus, and I haven't removed any of the default package lists).

z3ntu, 2021-02-27:
Ah right. It seems the search kind of recognizes dbus-user-session (and as such displays it) but it's not available in any Devuan version. Well anyways, openrazer is running on the session bus so we just need that. Debian's packaging is kind of a mess so I don't really know what "dbus-user-session" specifically does.

GreenBeard, 2021-02-27:
@z3ntu I think you want https://packages.debian.org/buster/dbus-session-bus as per my last moment edit before you posted.

z3ntu, 2021-02-27:
Hm sounds kind of right.. And it seems it's quite a recent addition in the openrazer timeframe ;) https://www.debian.org/doc/debian-policy/upgrading-checklist.html#version-4-3-0

z3ntu, 2021-02-27:
From looking through different Debian packages it looks like `Depends: default-dbus-session-bus | dbus-session-bus` is what should be used. I'll also drop Ubuntu 16.04 support while we're at it to simplify this.

GreenBeard, 2021-04-07:
@z3ntu I was curious if this was slated for a future update (not wanting to make breaking changes at the moment)? Or if it would be helpful at all if I submitted a pull request with these changes (obviously after validating them first)?

z3ntu, 2021-05-08:
@GreenBeard If you test it properly on Debian and Devuan and submit a PR that would be great

PR #1602 (2021-10-01): DRAFT: Changed dbus-user-session dependency to dbus-session-bus
Fixes #1443. Apologizes for the very late pull request, but I became busy and as they say: better late than never. I tested the tool on Debian 11.0, and Devuan Beowulf and it appears to still be working on both. I'm not really sure what to add as it only changes the debian package listed requirements to the newer version, and it seems to be fine. This will cause issues with distributions that don't have the `dbus-session-bus` virtual package. It appears that all currently supported versions of Debian (9/Stretch) and onward all hav [... cut]
closed 2021-10-01 by z3ntu