openrazer/openrazer #1448: Support for Tomahawk Gaming Desktop/Case

stellarhopper, 2021-03-05:
### Device Info

* Name: Razer Tomahawk Gaming Desktop
* Product Number: 0f27
* Model Number: <See under device>

### Device Information
```
Please post the output of
    lsusb -d 1532: -v

Bus 001 Device 004: ID 1532:0f27 Razer USA, Ltd Razer Tomahawk Gaming Desktop
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0f27 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Tomahawk Gaming Desktop
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0022
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      22
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)

```

```
Please post the output of
    ls -lah /dev/input/by-id/

total 0
drwxr-xr-x. 2 root root 100 Mar  5 02:09 .
drwxr-xr-x. 4 root root 460 Mar  5 02:09 ..
lrwxrwxrwx. 1 root root   9 Mar  5 02:09 usb-Logitech_USB_Receiver-if01-event-kbd -> ../event4
lrwxrwxrwx. 1 root root   9 Mar  5 02:09 usb-Logitech_USB_Receiver-if01-event-mouse -> ../event5
lrwxrwxrwx. 1 root root   9 Mar  5 02:09 usb-Logitech_USB_Receiver-if01-mouse -> ../mouse0

```

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
I'll grab these when I boot into windows next and update.

I just got this delivered today. Here is the product link:
https://www.razer.com/gaming-desktops-and-cases/razer-tomahawk-gaming-desktop/RZ80-03650200-B3U1

Just opening this issue to track the work - I'd actually love to try and implement support for this myself. Would definitely appreciate any pointers. I  plan to look into it more this weekend.

z3ntu, 2021-03-05:
With packet captures I could give you real pointers, but you could also just start adding the device to the razeraccessory driver (& udev rules) and then do trial & error and see what works.
You could use https://github.com/openrazer/openrazer/commit/9ea49c8d02af72fe065d2f805337da28b567fa48 as reference, start with the 99-razer.rules and razeraccessory_driver.h, add a `case` in razer_attr_read_device_type and add it to the HID_USB_DEVICE structure at the bottom of the file razeraccessory_driver.c. Then install that stuff and reboot, and the device should bind to the driver. From there you can start with the lighting effects etc (whatever this case can do)

mariocarbone, 2022-07-11:
Any news? I could be helpful for something?

stellarhopper, 2023-08-03:
I never got to this and I don't have this PC anymore.

LRipi, 2023-10-24:
Still no updates? How we could help?

z3ntu, 2023-10-24:
@LRipi Grab packet captures (see https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol) or just directly add support for the device: https://github.com/openrazer/openrazer/blob/master/DEVELOPMENT.md