openrazer/openrazer #1450: Support for [device name here]

Ahryman40k, 2021-03-05:
### Device Info

* Name: Razer Goliathus Extended Chroma
* Product Number: Not present under the device
* Model Number: RZ02-025003

### Device Information
```
Bus 003 Device 008: ID 1532:0c02 Razer USA, Ltd Goliathus Extended
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0c02 Goliathus Extended
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Goliathus Extended Chroma
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)
```

```
lrwxrwxrwx    10 root  5 Mar 17:33  usb-Razer_Razer_Goliathus_Extended_Chroma-event-mouse -> ../event17
lrwxrwxrwx    10 root  5 Mar 17:33  usb-Razer_Razer_Goliathus_Extended_Chroma-if01-event-kbd -> ../event27
lrwxrwxrwx    10 root  5 Mar 17:33  usb-Razer_Razer_Goliathus_Extended_Chroma-if02-event-kbd -> ../event28
lrwxrwxrwx     9 root  5 Mar 17:33  usb-Razer_Razer_Goliathus_Extended_Chroma-mouse -> ../mouse3

```

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
All required screenshots and options are presented in 
https://dl.razerzone.com/master-guides/RazerSynapse3/GoliathusChroma-00003073-en.pdf

z3ntu, 2021-03-05:
This device is already supported since v2.5.0 (March 2019)

closed 2021-03-05 by z3ntu

Ahryman40k, 2021-03-05:
Indeed, my fault sorry, I just need a new one for my mouse ... 
The second undetected device is my webcam (razer Kiyo), Is there a way to filter those devices ?

z3ntu, 2021-03-06:
`lsusb | grep 1532` should show the usb ids + device name