openrazer/openrazer #1451: Kraken Ultimate registers as a monochrome device.

hendy643, 2021-03-08:
### Distribution

Arch.
Ubuntu 18.04
Ubuntu 20.04

### Support

- [ x] Have you tried rebooting?
- [ x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

The Kraken Ultimate is a full RGB device, but it registers as a monochrome device.

[image]

lah7, 2021-03-08:
"Monochromatic" has nothing to do with OpenRazer. It's Polychromatic's way of _guessing_ (for the most part) whether a device uses the RGB APIs but only physically renders one colour (usually green). For example, the BlackWidow Ultimate 2016.

It's a bug in the application, [please raise the issue there](https://github.com/polychromatic/polychromatic/issues). But, seeing as I'm the author of the app, I've added it to my to-do list. 📝

closed 2021-03-08 by lah7

lah7, 2021-03-10:
Fixed in https://github.com/polychromatic/polychromatic/commit/9b70e7fe5e4c7c380e091565eb658467c04a8a3b. If you're on the edge PPA or `polychromatic-git` AUR, the fix will be heading your way. 🚚