openrazer/openrazer #1452: I keep getting a failed to fetch error

Dr4gon056, 2021-03-08:
Failed to fetch http://ppa.launchpad.net/openrazer/daily/ubuntu/pool/main/o/openrazer/openrazer-driver-dkms_2.9.0-1~610~ubuntu20.10.1_all.deb  404  Not Found [IP: 2001:67c:1560:8008::19 80]

z3ntu, 2021-03-08:
You might need to run `sudo apt update` and then try again.

Dr4gon056, 2021-03-08:
> You might need to run `sudo apt update` and then try again.

That was actually the first thing I tried and that did nothing. I then tried cleaning all the open razer files that did make it on the computer and tried again and that didn't work either

lah7, 2021-03-08:
Make sure there are no other errors when you run `sudo apt update`. Try disabling all third party sources in **Software & Updates → Other Software** tab and re-enable them one at a time.

Some cache might be outdated on your end since [the latest package in the PPA](https://launchpad.net/~openrazer/+archive/ubuntu/daily) is `2.9.0-1~612` (not `610`)

Dr4gon056, 2021-03-08:
> Make sure there are no other errors when you run `sudo apt update`. Try disabling all third party sources in **Software & Updates → Other Software** tab and re-enable them one at a time.
>
> Some cache [...]

ya somethings not in there I'm just going to say screw it and do a reinstall of pop os

closed 2021-03-08 by Dr4gon056