openrazer/openrazer #1453: Razer Ornata V2 detected but not found in one of the Applications

Somamint, 2021-03-08:
### Distribution

### Support

- [ ] Have you tried rebooting? yes 
- [ ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page? yes

Hello i try to use Razer Ornata V2 with Open Razer. I also use the Basilisk Mouse which works perfect. 
The Keyboard got detectet in the terminal :

Bus 001 Device 008: ID 1532:025d Razer USA, Ltd Razer Ornata V2
Bus 001 Device 009: ID 1532:0064 Razer USA, Ltd Basilisk

but it won't show up in Polychromatic or Razergenie.

z3ntu, 2021-03-08:
What's the output of the command `sudo dkms install openrazer-driver/2.9.0` ?

Somamint, 2021-03-08:
Module openrazer-driver/2.9.0 already installed on kernel 5.8.0-7642-generic/x86_64

z3ntu, 2021-03-08:
Thanks, what about `sudo modprobe razerkbd` and `openrazer-daemon -Fv` ?

z3ntu, 2021-03-08:
Actually the Ornata V2 isn't yet supported in the 2.9.0 release, you'll need to either wait for the next release or upgrade to a build from git (daily PPA on Ubuntu / openrazer-git from the AUR)

Somamint, 2021-03-08:
Its strange, because i reinstalled today my System and yesterday it worked perfect
sudo modprobe razerkbd
[sudo] Passwort für somavakien: 
somavakien@pop-os:~$ sudo modprobe razerkbd
somavakien@pop-os:~$ openrazer-daemon -Fv
Unable to lock on the pidfile.

Somamint, 2021-03-08:
ok after using daily it works again!

closed 2021-03-09 by Somamint

wojciechGaudnik, 2021-03-09:
I have the same issue but without a solution. 
 OS: Ubuntu 20.10 groovy
 Kernel: x86_64 Linux 5.8.0-44-generic
 Uptime: 29m
 Packages: 2702
 Shell: zsh 5.8
 Resolution: 6560x2981
 DE: KDE 5.74.0 / Plasma 5.19.5
 WM: KWin
 GTK Theme: Lim Green Dark [GTK2/3]
 Icon Theme: Breeze-openSUSE Dark Icons
 Disk: 1,1T / 1,4T (87%)
 CPU: Intel Core i9-10900K @ 20x 5,3GHz [48.0°C]
 GPU: Radeon RX 580 Series (POLARIS10, DRM 3.38.0, 5.8.0-44-generic, LLVM 11.0.0)
 RAM: 8866MiB / 64196MiB

2021-03-09 22:37:50 | razer                          | INFO     | Initialising Daemon (v2.9.0). Pid: 5622
2021-03-09 22:37:50 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2021-03-09 22:37:50 | razer                          | INFO     | Serving DBus

wojciechGaudnik, 2021-03-10:
@z3ntu 
ii  openrazer-daemon                              2.9.0~ubuntu20.10.1                                  all          OpenRazer peripheral drivers (daemon)
ii  openrazer-driver-dkms                         2.9.0~ubuntu20.10.1                                  all          OpenRazer peripheral drivers (DKMS)
ii  openrazer-meta                                2.9.0~ubuntu20.10.1                                  all          OpenRazer peripheral drivers (metapackage)
ii  python3-openrazer                             2.9.0~ubuntu20.10.1                                  all          OpenRazer peripheral drivers (Python 3)

z3ntu, 2021-03-12:
Read this comment https://github.com/openrazer/openrazer/issues/1453#issuecomment-793018670

wojciechGaudnik, 2021-03-13:
Yep, I have to learn how to read with understanding. Everything right now works great.